espressif/arduino-esp32 #9953: Can't set TX Power more than 15dBm

enjoyneering, 2024-06-28:
### Board

ESP32 Dev Module

### Device Description

plain module

### Hardware Configuration

plain module

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Can't set TX Power more than 15dBm. Arduino-esp32 v3.0.2

### Sketch

```cpp
WiFi.setTxPower(WIFI_POWER_17dBm);

WiFi.getTxPower(); //returns value not included in enun wifi_power_t

WiFi.setTxPower(WIFI_POWER_15dBm);

WiFi.getTxPower(); //returns WIFI_POWER_15dBm=60
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-06-29:
Looks like the [mappings have changed](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t). The [enum](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.h#L46-L62) needs to be fixed.
```
Mapping Table {Power, max_tx_power} = {{8, 2}, {20, 5}, {28, 7}, {34, 8}, {44, 11}, {52, 13}, {56, 14}, {60, 15}, {66, 16}, {72, 18}, {80, 20}}
```

TD-er, 2024-06-30:
Also keep in mind that there is also a limit per WiFi mode.
Typically 802.11b has the highest TX power.
So I wonder if you can actually set the TX power at such higher TX-power when connected using 802.11n (or 802.11ax on the ESP32C6)

VojtechBartoska, 2024-07-08:
Hello @enjoyneering, is this issue still valid or comments above helped?

enjoyneering, 2024-07-08:
Hi VojtechBartoska,

I haven't tried "esp_wifi_set_max_tx_power()". But transmit power more than 15dBm doesn't work in 3.0.2.

TD-er, 2024-07-08:
And what WiFi mode are you using?
802.11n? 802.11g? 802.11b?
Like I mentioned before, each mode has its own max. TX power and for 802.11n this is roughly 14 - 15 dBm.

enjoyneering, 2024-07-08:
Hi TD-er

This is exactly what I use 802.11n. I'm not complaining, there's just no information about this in the help. Is it possible to add a small note to the documentation?

TD-er, 2024-07-09:
Thh the ESP-IDF docs also do not mention this where one might expect it:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t

Even the datasheet is a bit unclear about this: https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf
Page 47 at the bottom does mention values I've also seen elsewhere when reading specs of the various ESP32's.
Also the table on page 48/49 does mention more plausible values.
However Page 32 at the top of the page mentions +18 dBm for 802.11n, which isn't correct as far as I know.  (maybe allowed by some country regulations?)

N.B. since the function to set the TX power does this in steps, you may need to also call the get function to see what actually has been set.

enjoyneering, 2024-07-09:
OK. Thank you.

closed 2024-07-09 by enjoyneering