espressif/arduino-esp32 #9954: possible memory leak

distrakt, 2024-06-29:
### Board

ESP32, ESP32-S2

### Device Description

I'm using "mini" style boards of ESP32 and ESP32-S2

### Hardware Configuration

none

### Version

v3.0.0

### IDE Name

ARDUINO

### Operating System

macos

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

With apologies -- this is a heads-up and I have not narrowed it down. I am using library "OmEspHelpers" which implements a web server.

On versions 2.0.17 and earlier, memory is solid, and any number of web requests continue working, free memory varies but does not trend down.

On versions 3.0.0 and 3.0.2, web requests appear to drain several hundred bytes each request.

this CAN be tested with the example web server sketch that is part of OmEspHelpers.
[The released version of OmEspHelpers may need manual editing of SYSTEM_EVENT_AP_STACONNECTED to ARDUINO_EVENT_WIFI_AP_STACONNECTED to compile for 2.0 and later]

### Sketch

```cpp
i will attempt to create minimal case... but so far I can only see it in a somewhat rich project based on OmEspHelpers example sketch.
```

### Debug Message

```plain
none
reading memory with system_get_free_heap_size()
eventually hangs
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2024-06-30:
Please also check if the used memory is later freed after +/- 2 minutes when you stop making the calls.

distrakt, 2024-06-30:
TD-er   with skepticism I checked... and you're right! Sure enough after a little while memory was freed. So that's both nice, and not-nice. For a REST-like service, calls still come in potentially quite fast (like being polled by another gadget), so a 2 minute memory cleanup turnaround is a step backwards from 2.x.

If this (garbage collector?) is configurable, I feel perhaps it ought to default to a short interval like before.

What is the name of this system/feature/api ... ?

TD-er, 2024-06-30:
See also: https://github.com/espressif/arduino-esp32/issues/8804

lbernstone, 2024-07-07:
I don't really have the ability to verify, but this kind of thing can often happen if you get a bunch of sockets in TIME_WAIT or CLOSE_WAIT status because sessions aren't closing cleanly.  Make sure you put appropriate headers on requests that should be closed.

TD-er, 2024-07-07:
I wonder how much you can do on the ESP side and how much is not closed properly by tools like the browser.

distrakt, 2024-07-09:
This feels plausible.
It is still, though, a regression from 2.x; if it’s a new “improvement”
with some compatibility issues it should be switchable perhaps?

VojtechBartoska, 2024-07-09:
@me-no-dev please help with triage of this issue,  thanks

TD-er, 2024-07-09:
Just serving some minimal HTML page and hitting it with a browser that does frequent refresh will cause this behavior.
Then periodically output the amount of free memory and a timestamp will show a decrease per pageview.
After stopping those calls you will gradually see the amount of free memory increase again, for each call that was 120 sec ago.

N.B. On nodes with PSRAM, make sure to also keep track of free PSRAM memory.

distrakt, 2024-07-09:
As per TD-er -- I attached a basic web server example i found, on the page it prints the free bytes, and indeed shows the issue.

me-no-dev, 2024-07-10:
Issue confirmed. Each request takes between 200 and 240 bytes. Those bytes are returned a minute or so later, which means that they are the connection in TIME_WAIT. It's interesting to me why the number of such sockets is not limited internally, as I was able to run the ESP almost out of memory by in effect DDoSing it with requests. The issue is somewhere below Arduino level though. I wonder if there is something we can do, like reuse client or something...

me-no-dev, 2024-07-10:
Patch added in the lib-builder. Will be part of next release

closed 2024-07-10 by me-no-dev

Jason2866, 2024-07-10:
Tested the fix. Does work :-)