espressif/arduino-esp32 #9955: The current mainline branch cannot use QSPI and will assert after it is turned on

lewisxhe, 2024-06-29:
### Board

ESP32S3 Dev (ESP32S3-N4R2)

### Device Description

No GPIO is connected

### Hardware Configuration

No GPIO is connected

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE2.3.2

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

468000

### Description

As long as QSPI RAM is turned on, the chip will always restart when it starts, and there is nothing on the sketch

[image]

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

```
```

### Debug Message

```plain
ELF file SHA256: 009567c960197a4b

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403775d5
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x438
load:0x403c9700,len:0x4
load:0x403c9704,len:0x9ac
load:0x403cc700,len:0x2764
entry 0x403c9868
E (554) mmap: esp_mmu_unmap(595): munmap target pointer is outside external memory regions

assert failed: spi_flash_munmap flash_mmap.c:241 (0 && "invalid handle, or handle already unmapped")

Backtrace: 0x40376fc2:0x3fcf3cd0 0x4037bd75:0x3fcf3cf0 0x403819d1:0x3fcf3d10 0x42008ca7:0x3fcf3e40 0x420087dd:0x3fcf3e60 0x420088e1:0x3fcf3e80 0x42008b7b:0x3fcf3eb0 0x42002b38:0x3fcf3f10 0x42001ca1:0x3fcf3f40 0x4202717d:0x3fcf3f70 0x4037e91a:0x3fcf3fb0

ELF file SHA256: 009567c960197a4b

Rebooting...
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-06-29:
Duplicate of #9936 Please close issue will be tracked there.

lewisxhe, 2024-06-29:
There is a similar issue here, tracked here

closed 2024-06-29 by lewisxhe