espressif/arduino-esp32 #9959: Unable to use uart0 RX pin in esp32-c6

adiiot-1316, 2024-06-29:
### Board

ESP32-C6

### Device Description

ESP32-C6-WROOM-1-N8 Board

### Hardware Configuration

GPIO 4,5 are connected to GSM Module TX RX respectively

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

9200

### Description

Hi , 

I am trying to connect my  Lora module with ESP32-c6 over UART Rx Tx line. UART 4,5 are already used in connection with GSM Module. I am trying to use ESP32-c6 UART Pin N.o. 16,17 with Lora Module , but Rx is not receiving the packets as GPIO17 is default  function active. How to enable the Rx function in Arduino  IDE?
As checked in c6 documentation, there are 3 functions associated with GPIO 17.
1.)Rx
2.)GPIO17
3.)Chipselect.
By default f1 is enabled. We need to change the function to f0. tried below code but it didnot work out. 
#define IO_MUX_GPIO17_MCU_SEL 0;

Same piece of code is working when i am using UART Pin# 4, 5.
I am quite stuck due to this issue in prototype phase and would really appreciate your support here.

### Sketch

```cpp
//Receiver Code
#include "Arduino.h"
#define E32_TTL_1W
#include "LoRa_E32.h"
/* For esp32
#define AUX 15
#define M0 2
#define M1 3
*/
#define AUX 23
#define M0 14
#define M1 15
//#define IO_MUX_GPIO17_MCU_SEL 0;

LoRa_E32 e32ttl1w(&Serial0, AUX, M0, M1);
ResponseContainer rc;
void setup() {
	Serial.begin(9600);
  Serial.println("In setup mode");
	delay(500);
// Startup all pins and UART
	e32ttl1w.begin();
	nodeConfig();
}

void loop() {
	if (e32ttl1w.available() > 0) {
		Serial.println("———-");
		rc = e32ttl1w.receiveMessageUntil();    
		if (rc.status.code != 1) {
			rc.status.getResponseDescription();      
		} else {
			// Print the data received
			Serial.println(rc.data);
		}
	}
	delay(100);
}

void nodeConfig() {
	ResponseStructContainer c;
	c = e32ttl1w.getConfiguration();
	Configuration configuration = *(Configuration*) c.data;
	configuration.ADDH = 0x0;
	configuration.ADDL = 0x3;
	configuration.CHAN = 0x4;
	configuration.SPED.airDataRate = AIR_DATA_RATE_010_24;
	configuration.SPED.uartBaudRate = UART_BPS_9600;
	configuration.SPED.uartParity = MODE_00_8N1;
	configuration.OPTION.fec = FEC_1_ON;
	configuration.OPTION.fixedTransmission = FT_FIXED_TRANSMISSION;
	configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;
	configuration.OPTION.transmissionPower = POWER_21;
	configuration.OPTION.wirelessWakeupTime = WAKE_UP_1250;
	// Set configuration changed and set to not hold the configuration
	c.close();
	Serial.println("Updtaed config is ");
  
	ResponseStatus rs = e32ttl1w.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
	Serial.print("Config: ");
	Serial.println(rs.getResponseDescription());
}
```

### Debug Message

```plain
No Error message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-01:
I don't see you using `Serial*.setPins(RX, TX);` anywhere? You need to set the pins prior to calling begin

adiiot-1316, 2024-07-01:
Thanks for your reply !! Though it should be covered in E32 Library, i'll cross verify and get back to you.

VojtechBartoska, 2024-07-08:
hello @adiiot-1316, can I mark this as answered?

adiiot-1316, 2024-07-12:
@VojtechBartoska  yes please. Some issue with uart pin. Work fine on other board.

adiiot-1316, 2024-08-12:
@VojtechBartoska  : I am facing this issue with all ESP32-C6-WROOM-1U. However it works fine on ESP32-C6-Mini. 
Tx is not working. I am using the Lora E32 Ebyte library. Is there a simple code to verify this?

adiiot-1316, 2024-08-25:
@VojtechBartoska @me-no-dev @SuGlider : Can you please help me urgently here as it is affecting the production please. Really appreciate your support in advance.

lbernstone, 2024-08-25:
Have you tried calling Serial0.end before you start the LORA?  It may still be in use by the console.

adiiot-1316, 2024-08-25:
Thanks @lbernstone  for your reply. Its not working. I added Serial0.end before lora begin. Attached below is the code. I am unable to use uart0 . Though lp_uart (4,5) are working fine. But that lp_uart is connected to my GSM Module. Hence i need another uart, which i can connect with LoRa

`//Receiver Code
#include "Arduino.h"
#define E32_TTL_1W
#include "LoRa_E32.h"
#define FREQUENCY_868

#define AUX 15
#define M0 22
#define M1 23

//#define IO_MUX_GPIO17_MCU_SEL 0;
LoRa_E32 e32ttl1w(&Serial0, AUX, M0, M1);
ResponseContainer rc;
void setup() {
  //uart_set_pin(UART_NUM_1, 17, 5, 18, 19)
//Inicio comunicacion serie con monitor
	Serial.begin(115200);
  Serial.println("In setup mode");
	delay(500);
// Startup all pins and UART
  Serial0.end();
	e32ttl1w.begin();
//Configuro el nodo LoRa
	nodeConfig();
  Serial.println("Setup Ends");
  //delay(4000);
  //fetchConfig();
}

void loop() {
  //Serial.println("In Loop");
	if (e32ttl1w.available() > 0) {
		Serial.println("———-");
		rc = e32ttl1w.receiveMessageUntil();
    //rc.close();
// If something goes wrong print error
		if (rc.status.code != 1) {
			rc.status.getResponseDescription();
      
		} else {
// Print the data received
			Serial.println(rc.data);
		}
    //rc.close();
	}
	delay(100);
}

void nodeConfig() {
	ResponseStructContainer c;
	c = e32ttl1w.getConfiguration();
	Configuration configuration = *(Configuration*) c.data;
	configuration.ADDH = 0x0;
	configuration.ADDL = 0x3;
	configuration.CHAN = 0x4;
	configuration.SPED.airDataRate = AIR_DATA_RATE_010_24;
	//configuration.SPED.uartBaudRate = UART_BPS_9600;
	configuration.SPED.uartParity = MODE_00_8N1;
	configuration.OPTION.fec = FEC_1_ON;
	configuration.OPTION.fixedTransmission = FT_FIXED_TRANSMISSION;
	configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;
	configuration.OPTION.transmissionPower = POWER_27;
	configuration.OPTION.wirelessWakeupTime = WAKE_UP_1250;
// Set configuration changed and set to not hold the configuration
  c.close();
  Serial.println("Updtaed config is ");
  
	ResponseStatus rs = e32ttl1w.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
	Serial.print("Config: ");
	Serial.println(rs.getResponseDescription());
}`

lbernstone, 2024-08-25:
The difference on pin 17 is very likely the console being attached.  Arduino is set up to "do the right thing", so it by default puts the console on both the usb jtag and uart0.  If you can set core debug level to verbose, and then post that log, maybe we can see an error there from the gpio mux.
In the worst case, you can try to use [arduino as an esp-idf component](https://github.com/espressif/arduino-esp32/tree/master/idf_component_examples/hello_world) and jtag to be the primary console output, with no secondary output. There will still be some logging sent out during boot, but it doesn't sound like that is preventing the lora from initializing.
[image]

adiiot-1316, 2024-08-26:
I want to use Arduino as component but not much familiar with espress-if ide. 
Below is the verbose log . 

`07:46:33.528 -> �ESP-ROM:esp32c6-20220919
07:46:33.528 -> Build:Sep 19 2022
07:46:33.528 -> rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
07:46:33.528 -> SPIWP:0xee
07:46:33.528 -> mode:DIO, clock div:2
07:46:33.528 -> load:0x4086c410,len:0xcf8
07:46:33.539 -> load:0x4086e610,len:0x2e44
07:46:33.539 -> load:0x40875728,len:0x113c
07:46:33.539 -> entry 0x4086c410
07:46:33.593 -> [    88][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42004c22
07:46:33.593 -> [    99][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42004be6
07:46:33.593 -> [   111][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42004baa
07:46:33.627 -> [   122][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42004b6e
07:46:33.627 -> [   133][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42004c22
07:46:33.627 -> [   145][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42004be6
07:46:33.661 -> [   156][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42004baa
07:46:33.661 -> [   167][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42004b6e
07:46:33.661 -> [   179][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (38) successfully set to 0x42000a1e
07:46:33.692 -> [   190][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (39) successfully set to 0x42000a1e
07:46:33.729 -> [   213][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type UART_RX (2) with bus 0x4080c110
07:46:33.729 -> [   223][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type UART_TX (3) with bus 0x4080c110
07:46:33.730 -> [   234][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 12 successfully set to type USB_DM (38) with bus 0x4080d774
07:46:33.730 -> [   245][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 13 successfully set to type USB_DP (39) with bus 0x4080d774`

MrFloppy70, 2024-09-19:
> ### Board
> ESP32-C6
> 
> ### Device Description
> ESP32-C6-WROOM-1-N8 Board
> 
> ### Hardware Configuration
> GPIO 4,5 are connected to GSM Module TX RX respectively
> 
> ### Version
> latest master (checkout manually)
> 
> ### IDE Name
> Arduino IDE
> 
> ### Operating System
> Windows 10
> 
> ### Flash frequency
> 80Mhz
> 
> ### PSRAM enabled
> yes
> 
> ### Upload speed
> 9200
> 
> ### Description
> Hi ,
> 
> I am trying to connect my Lora module with ESP32-c6 over UART Rx Tx line. UART 4,5 are already used in connection with GSM Module. I am trying to use ESP32-c6 UART Pin N.o. 16,17 with Lora Module , but Rx is not receiving the packets as GPIO17 is default function active. How to enable the Rx function in Arduino IDE? As checked in c6 documentation, there are 3 functions associated with GPIO 17. 1.)Rx 2.)GPIO17 3.)Chipselect. By default f1 is enabled. We need to change the function to f0. tried below code but it didnot work out. #define IO_MUX_GPIO17_MCU_SEL 0;
> 
> Same piece of code is working when i am using UART Pin# 4, 5. I am quite stuck due to this issue in prototype phase and would really appreciate your support here.
> 
> ### Sketch
> ```c++
> //Receiver Code
> #include "Arduino.h"
> #define E32_TTL_1W
> #include "LoRa_E32.h"
> /* For esp32
> #define AUX 15
> #define M0 2
> #define M1 3
> */
> #define AUX 23
> #define M0 14
> #define M1 15
> //#define IO_MUX_GPIO17_MCU_SEL 0;
> 
> LoRa_E32 e32ttl1w(&Serial0, AUX, M0, M1);
> ResponseContainer rc;
> void setup() {
> 	Serial.begin(9600);
>   Serial.println("In setup mode");
> 	delay(500);
> // Startup all pins and UART
> 	e32ttl1w.begin();
> 	nodeConfig();
> }
> 
> void loop() {
> 	if (e32ttl1w.available() > 0) {
> 		Serial.println("———-");
> 		rc = e32ttl1w.receiveMessageUntil();    
> 		if (rc.status.code != 1) {
> 			rc.status.getResponseDescription();      
> 		} else {
> 			// Print the data received
> 			Serial.println(rc.data);
> 		}
> 	}
> 	delay(100);
> }
> 
> void nodeConfig() {
> 	ResponseStructContainer c;
> 	c = e32ttl1w.getConfiguration();
> 	Configuration configuration = *(Configuration*) c.data;
> 	configuration.ADDH = 0x0;
> 	configuration.ADDL = 0x3;
> 	configuration.CHAN = 0x4;
> 	configuration.SPED.airDataRate = AIR_DATA_RATE_010_24;
> 	configuration.SPED.uartBaudRate = UART_BPS_9600;
> 	configuration.SPED.uartParity = MODE_00_8N1;
> 	configuration.OPTION.fec = FEC_1_ON;
> 	configuration.OPTION.fixedTransmission = FT_FIXED_TRANSMISSION;
> 	configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;
> 	configuration.OPTION.transmissionPower = POWER_21;
> 	configuration.OPTION.wirelessWakeupTime = WAKE_UP_1250;
> 	// Set configuration changed and set to not hold the configuration
> 	c.close();
> 	Serial.println("Updtaed config is ");
>   
> 	ResponseStatus rs = e32ttl1w.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
> 	Serial.print("Config: ");
> 	Serial.println(rs.getResponseDescription());
> }
> ```
> 
> ### Debug Message
> ```
> No Error message
> ```
> 
> ### Other Steps to Reproduce
> _No response_
> 
> ### I have checked existing issues, online documentation and the Troubleshooting Guide
> * [x]  I confirm I have checked existing issues, online documentation and Troubleshooting guide.
Hi [adiiot-1316](https://github.com/adiiot-1316).

¿Can you show me your code to comunicate with the GSM module using the LPUART Serial1 (gipios 4 y 5)?

I am trying to comunicate with an A7672E LTE Modem, with de same sample code that works in an ESP32S3 and it's imposible to me. My board is an original ESP32-C6-DevKitC-1.

Thanks a lot.

SuGlider, 2025-01-15:
>07:46:33.729 -> [ 213][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type UART_RX (2) with bus 0x4080c110
07:46:33.729 -> [ 223][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type UART_TX (3) with bus 0x4080c110
07:46:33.730 -> [ 234][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 12 successfully set to type USB_DM (38) with bus 0x4080d774
07:46:33.730 -> [ 245][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 13 successfully set to type USB_DP (39) with bus 0x4080d774`

This indicates that UART0 is using the default pins RX0 pin 17, TX0, pin 16 and also USB Hardware Serial and JTAG default pins 12 and 13 (most probably for Serial interface with CDC on Boot Enabled).

SuGlider, 2025-01-15:
> @VojtechBartoska @me-no-dev @SuGlider : Can you please help me urgently here as it is affecting the production please. Really appreciate your support in advance. Serial1 pins 4,5 (LPUART) are working fine. But when using Serial 0 (16,17) , its not working. https://docs.espressif.com/projects/esp-dev-kits/en/latest/_images/esp32-c6-devkitc-1-pin-layout.png

Serial0 seems to be fine and correctly set from the logs.
But Serial1 has no log messages saying that it was set.
CDC on Boot seems Enabled because Pins 12 and 13 are being assigned to D+/D- USB Port. In that case, `Serial` will be lied to the USB Serial port.