espressif/arduino-esp32 #9960: V3 ESP32 Arudino IDE library missing opensll (not supported https for C6)

virco, 2024-06-29:
### Board

esp32 C6 Dev

### Device Description

DevKit C6 N8

### Hardware Configuration

N/A

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

C6 module are supported from V3 onwards, where there is an issue with missing  "openssl/ssl.h"

### Sketch

```cpp
https://github.com/khoih-prog/HTTPS_Server_Generic/blob/main/examples/ESP32_WiFi/REST-API/REST-API.ino
```

### Debug Message

```plain
C:\Users\virco\AppData\Local\Temp\.arduinoIDE-unsaved2024529-27392-1upclbq.1rf7\REST-API\REST-API.ino:57:4: error: #error To be use with ARDUINOJSON_VERSION v5.13.5-
   57 |   #error To be use with ARDUINOJSON_VERSION v5.13.5-
      |    ^~~~~
In file included from C:\Users\virco\AppData\Local\Temp\.arduinoIDE-unsaved2024529-27392-1upclbq.1rf7\REST-API\REST-API.ino:79:
C:\Users\virco\OneDrive\Documents\Arduino\libraries\HTTPS_Server_Generic\src/HTTPS_Server_Generic.h:42:14: fatal error: openssl/ssl.h: No such file or directory
   42 |     #include "openssl/ssl.h"
      |              ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: #error To be use with ARDUINOJSON_VERSION v5.13.5-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-06-29:
The standard library supported from espressif Arduino is mbedtls.

virco, 2024-06-30:
OK, is there an Arduino HTTPS WEbServer library that utilizes mbedtls?
Looks like all available packages at the moment rely on open openssl/ssl.h which was available prior Version 3 framework.

virco, 2024-06-30:
I looked all "https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples" but didn't find https there, can you point out some examples with TLS 1.2 or TLS 1.3?

Thanks in advance!

al-TideLight, 2024-07-09:
I have this same issue with V3 and C6. I have not been able to find any HTTPS examples or working alternatives.

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-15 by Parsaabasi

r57zone, 2025-05-23:
Has anyone found this openssl library? Looks like some old version is needed