espressif/arduino-esp32 #9962: Zigbee Sleepy End Device Example

felipejfc, 2024-06-29:
### Related area

Zigbee

### Hardware specification

ESP32-C6

### Is your feature request related to a problem?

We currently lack an example of working sleepy end device.

### Describe the solution you'd like

We should have a working example, I did try to make it by myself, however I don't ever receive "ESP_ZB_COMMON_SIGNAL_CAN_SLEEP", also, I can't configure PM cause it causes a crash. Reference project: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2024-11-14:
Hi @felipejfc, I have added a sleepy example #10551. Deep sleep is used so there is no need to set up PM.
I am closing this issue as solved. If you have any questions feel free to reopen or text here :)

closed 2024-11-14 by P-R-O-C-H-Y

felipejfc, 2024-11-14:
Thx for that!! Super cool.
Is light sleep also supported though? For some applications, deep sleeping will be less power efficient because it takes much longer to bootstrap radio signal etc after one than if light sleeping

P-R-O-C-H-Y, 2024-11-14:
The light sleep is currently not possible, as the Power Management would need to be enabled in sdk config which is not currently. Also what is used for Zigbee in the official light sleep example (what you have linked in the description) is the FreeRTOS tickles idle, which may bring some unexpected behavior (`CONFIG_FREERTOS_USE_TICKLESS_IDLE`).

We may add the PM in the future releases, when we do I will test the light sleep with Zigbee and if its doable, I will implement it and add another example :)

tomasz90, 2025-04-13:
Hi!
@P-R-O-C-H-Y is there another option to reduce power consumption than light sleep or deep sleep? Unfortunately deep sleep is only suitable for completly passive elements like sensors, cause reconnection takes couple of seconds.. I am stuck at 60mA which is really unacceptable, specially for a zigbee device which should be very power efficient.

P-R-O-C-H-Y, 2025-04-14:
Hi @tomasz90 please open a discussion or join our discord channel for those questions to not spam already closed issue.
Also the reconnection time is now about 320 mS if the device is in good range, as we are saving the network channel. So the scan is fast. I don't see a different way to reduce power consumption than using deep sleep. Maybe you can lower the clock if its possible for your device.