espressif/arduino-esp32 #9963: 3.0.0. and above SerialBT.connect(deviceName) ist not succesfully

BlackWolfDEsign, 2024-06-30:
### Board

ESP32-WROOM-DA Module

### Device Description

ESP32-WROOM-32

### Hardware Configuration

nothing

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

Win 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Since version 3.0.0, SerialBT.connect(DeviceName) (DeviceName as String) is no longer successful.
With version 2.0.17, the connection between master and slaves works.

### Sketch

```cpp
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

void setup() {
    Serial.begin(115200);
    SerialBT.begin("ESP32Master", true); // Bluetooth Serial initialisieren

    Serial.println("Starting Bluetooth...");
    delay(1000);

    // Verbindung zum Slave herstellen
    if (SerialBT.connect("ESP32Slave")) {
        Serial.println("Connected to ESP32Slave");
    } else {
        Serial.println("Failed to connect to ESP32Slave");
    }
}

void loop() {
    if (SerialBT.connected()) {
        Serial.println("Bluetooth connected...");

        delay(5000); // Beispiel: Warte 5 Sekunden zwischen den Nachrichten

    } else {
        Serial.println("Disconnected from ESP32Slave. Reconnecting...");
        if (SerialBT.connect("ESP32Slave")) {
            Serial.println("Reconnected to ESP32Slave");
        } else {
            Serial.println("Reconnection failed");
        }
        delay(2000);
    }
}

#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

void setup() {
    Serial.begin(115200);
    SerialBT.begin("ESP32Slave"); // Bluetooth Serial initialisieren

    Serial.println("Starting Bluetooth...");
    delay(1000);
}

void loop() {
    if (SerialBT.connected()) {
        Serial.println("Bluetooth connected...");

        delay(5000); // Beispiel: Warte 5 Sekunden zwischen den Nachrichten

    } else {
        Serial.println("Waiting for connection...");
        delay(2000);
    }
}

#created by openai
```

### Debug Message

```plain
no output
```

### Other Steps to Reproduce

try both version of arduino core

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-01:
Please enable core debug to Verbose and post the output of both sides

BlackWolfDEsign, 2024-07-01:
master v3.0.0

`09:35:13.082 -> Rebooting...
09:35:13.082 -> �%��� U�8��6i�(���*�9�e�iM��w���*�9/�j��5�h���Am�������<Y��Z�)�LW�'�p�OԅlC�(�W�E�<\�D�^a�\-s��&��U	@�ش������Ok��F� ���������8�����$���.�$����0���+�d�$�=�������-�6�+�T�����}�����S������!5�����!P��J������`������
��������������Vh�L����~����.����c�<�K����(� 4��A'��[   737][V][esp32-hal-uart.c:544] uartBegin(): UART0 initialization done.
09:35:14.417 -> [   917][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_SCK (37) successfully set to 0x400e0ab4
09:35:14.549 -> [  1060][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_MISO (38) successfully set to 0x400e09d4
09:35:14.706 -> [  1204][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_MOSI (39) successfully set to 0x400e08ec
09:35:14.834 -> [  1348][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_SS (40) successfully set to 0x400e07b4
09:35:14.993 -> [  1490][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x401aa6c4
09:35:15.122 -> [  1621][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 4 successfully set to type GPIO (1) with bus 0x5
09:35:15.217 -> [  1737][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 4 successfully set to type SPI_MASTER_SCK (37) with bus 0x3
09:35:15.346 -> [  1864][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x401aa6c4
09:35:15.505 -> [  1995][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type GPIO (1) with bus 0x11
09:35:15.601 -> [  2113][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type SPI_MASTER_MISO (38) with bus 0x3
09:35:15.729 -> [  2243][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x401aa6c4
09:35:15.858 -> [  2374][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type GPIO (1) with bus 0x12
09:35:15.986 -> [  2492][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type SPI_MASTER_MOSI (39) with bus 0x3
09:35:16.114 -> [  2621][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x401aa6c4
09:35:16.242 -> [  2752][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 0 successfully set to type GPIO (1) with bus 0x1
09:35:16.370 -> [  2868][V][esp32-hal-periman.c:174] perimanSetPinBusExtraType(): Successfully set extra_type SD_SS for pin 0
09:35:16.548 -> [  3044][W][sd_diskio.cpp:489] ff_sd_initialize(): GO_IDLE_STATE failed
09:35:16.611 -> [  3121][E][sd_diskio.cpp:759] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
09:35:16.708 -> [  3224][E][vfs_api.cpp:23] open(): File system is not mounted
09:35:17.273 -> [  3767][I][BluetoothSerial.cp[p : 537756]7 ]e[sIp]_[bBtl_ugeatpo_octbh(S)e:r iEaSlP._cBpTp_:G7A0P7_]C O_NiFnIiGt__EbItR(_)D:A Tdevice nTmessat
0 num:4
09:35:17.434 -> [  3927][I][BluetoothSerial.cpp:1154] discover(): discover::disconnect
09:35:17.497 -> [  3928][I][BluetoothSerial.cpp:248] esp_spp_cb(): ESP_SPP_INIT_EVT
09:35:17.561 -> [  4074][I][BluetoothSerial.cpp:1156] discover(): discovering
09:35:17.626 -> [  4074][I][BluetoothSerial.cpp:254] esp_spp_cb(): ESP_SPP_INIT_EVT: master: start
09:35:17.721 -> [  4227][I][BluetoothSerial.cpp:575] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
09:35:17.817 -> [  4335][I][BluetoothSerial.cpp:329] esp_spp_cb(): ESP_SPP_START_EVT
09:35:17.913 -> [  4408][I][BluetoothSerial.cpp:504] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT started
09:35:18.009 -> [  4509][I][BluetoothSerial.cpp:422] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
09:35:18.106 -> [  4605][I][BluetoothSerial.cpp:425] esp_bt_gap_cb(): Scanned device: f4:4e:fd:6a:c1:96
09:35:18.202 -> [  4698][D][BluetoothSerial.cpp:451] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x240404
09:35:18.298 -> [  4790][D][BluetoothSerial.cpp:461] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 203
09:35:18.362 -> [  4877][I][BluetoothSerial.cpp:470] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : Fosi Audio BT30D PRO : 20
09:35:18.491 -> [  4996][I][BluetoothSerial.cpp:422] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
09:35:18.588 -> [  5092][I][BluetoothSerial.cpp:425] esp_bt_gap_cb(): Scanned device: f4:4e:fd:6a:c1:96
09:35:18.684 -> [  5185][D][BluetoothSerial.cpp:451] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x240404
09:35:18.780 -> [  5276][D][BluetoothSerial.cpp:461] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 204
09:35:18.876 -> [  5364][I][BluetoothSerial.cpp:470] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : Fosi Audio BT30D PRO : 20
09:35:18.971 -> [  5482][I][BluetoothSerial.cpp:422] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
09:35:19.067 -> [  5578][I][BluetoothSerial.cpp:425] esp_bt_gap_cb(): Scanned device: f4:4e:fd:6a:c1:96
09:35:19.164 -> [  5671][D][BluetoothSerial.cpp:451] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x240404
09:35:19.261 -> [  5763][D][BluetoothSerial.cpp:461] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 200
09:35:19.357 -> [  5850][I][BluetoothSerial.cpp:470] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : Fosi Audio BT30D PRO : 20
09:35:19.453 -> [  5969][I][BluetoothSerial.cpp:422] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[... cut ...]
09:35:22.918 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
09:35:22.982 -> 
09:35:22.982 -> 
09:35:22.982 -> 
09:35:22.982 -> 
09:35:22.982 -> 
09:35:22.982 -> Backtrace: 0x3ffc500d:0x000007d0 |<-CORRUPTED
09:35:23.046 -> 
09:35:23.046 -> 
09:35:23.046 -> 
09:35:23.046 -> 
09:35:23.046 -> ELF file SHA256: 06ba690b10a53635
09:35:23.078 -> 
09:35:23.078 -> Re-entered core dump! Exception happened during core dump! `

-> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception).

closed 2024-07-01 by BlackWolfDEsign