espressif/arduino-esp32 #9969: String data = USBCOM.readString();

Red-Owl, 2024-06-30:
### Board

ESP32 wt32-s3-wrover (wt32sc01 plus)

### Device Description

ESP32 wt32-s3-wrover (wt32sc01 plus)

### Hardware Configuration

ESP32 wt32-s3-wrover (wt32sc01 plus)

### Version

v2.0.16

### IDE Name

Platformio

### Operating System

win11

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

***ERROR*** A stack overflow in task **usbd** has been detected. 

### Sketch

```cpp
void usb_as_serial_read(){
    String data = USBCOM.readString();
}
```

### Debug Message

```plain
***ERROR*** A stack overflow in task usbd has been detected.

Backtrace: 0x40378296:0x3fcb8c90 0x4037d7e9:0x3fcb8cb0 0x40380c9a:0x3fcb8cd0 0x4037f49c:0x3fcb8d50 0x4037d8a0:0x3fcb8d80 0x4037d896:0x00000000 |<-CORRUPTED
```

### Other Steps to Reproduce

send via serial monitor a INFO string in repeat mode repeat 10
arrow up, INFO enter
arrow up,INFO enter
arrow up,INFO enter
....
panic !!
or  send \r\n
panic !!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-06-30:
Once again, programming error.  Don't pass Strings (or any large object) around, use pointers. USBCOM is not part of the code here.  For help with 3rd party libraries, you need to contact the author.

SuGlider, 2024-08-02:
@Red-Owl - please provide more details about how to reproduce this issue.
What is `USBCOM` obejct? Is it an external library as commented by @lbernstone?

SuGlider, 2024-08-02:
ESP32-S3 has two possible USB CDC objects: from HW Serial JTAG and from USB OTG. Both implement `Stream` which allows to execute `Serial.readString()` whenever `Serial` is configured as the USB port.

Is that the issue?

SuGlider, 2024-08-02:
Arduino Core 3.0.3 works fine.

closed 2025-01-15 by Parsaabasi