espressif/arduino-esp32 #9970: AsyncUDP ipv6 Multicast not working

friedrichengel, 2024-06-30:
### Board

ESP-32-Wrover-E

### Device Description

Own Hardware, uses Etehrnet (Rj45)

### Hardware Configuration

nothing i would think that it has something to do with multicast Ethernet. Ehernet + Webserver in general is working fine.

### Version

v3.0.1

### IDE Name

VSCode with Platformio

### Operating System

Windows 11

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

922190

### Description

Settingup an UDP Listener to use IPv6 Multicast dont seem to work.

if you setup your listener with:

IPAddress udp_ip_v6;     
udp_ip_v6.fromString("ff02::1"); //"ff02::1"
listen_multicast_v6 = udp.listenMulticast(udp_ip_v6, udp_port, 0, TCPIP_ADAPTER_IF_ETH);

It seems to setup this listener and it seems fine. Return Value says it is ok.

If you then wants to send something to this multicast Adress it returns an error. I have tried different ways to send:

1) 
udp_ip_2.fromString("ff02::1");
udp_send_rv = udp.writeTo((uint8_t*)(temp_casambi_command.c_str()),temp_casambi_command.length(), udp_ip_2, udp_port, TCPIP_ADAPTER_IF_ETH);

2)
udp_send_rv = udp.print(temp_casambi_command.c_str());

3)
udp_message.write((uint8_t*)(temp_casambi_command.c_str()), temp_casambi_command.length());

But it always returned an error code in most cases it is:
ERR_VAL        = -6,
or 
ERR_RTE        = -4,

If i do it on an non multicast way like
listen_v4 = udp.listen(udp_port);
listen_v6 = udp.listenIPv6();

and sending it to an known working ip it seems to be fine and i can receive the Packet.

If i send commands directed in ipv4 it also works and if i send it in broadcast in ipv4 it works too.

### Sketch

```cpp
See Description
```

### Debug Message

```plain
See Description
```

### Other Steps to Reproduce

I have tried it with different adresses and also changing between TCPIP_ADAPTER_IF_ETH and TCPIP_ADAPTER_IF_MAX or letting it empty. In Multicast it seems not to send something.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-01:
- Have you tried printing back `udp_ip_v6` and `udp_ip_2 `, in case it did not convert them properly?
- `listen_multicast_v6` is checked to be true, correct?
- Can you enable core debug to Verbose and post the minimal sketch and serial output? I would like to try and reproduce it exactly.

friedrichengel, 2024-07-01:
Yes i have printed all these informations like IP-Adress and return values from listenMulticast and writeTo.

listen_multicast_v6 is true, yes i have checked this. I also added some more debug output to listenMulticast but it does not throw any error.

if changed also the return values from writeTo so i can extract the error codes from inside the async udp library.
ERR_VAL = -6,
or
ERR_RTE = -4,

I have set "-DCORE_DEBUG_LEVEL=5" for my build flags but it does not give me many informations. Is there another way to enable debug output?

I sadly dont have an minimalized example. I could try to change the multicast example from the framework to use ipv6 instead of ipv4.

Strange is that normal ipv6 with directed adresses are working fine.

I would also suppose that i could send to the multicast address directy without joining the multicast group. Because send and receive should be seperated. But i can also not sent to an multicast adress only with listening to udp and enable listenv6.

friedrichengel, 2024-07-01:
Ok, i have added a simple sketch basically with all part i think are needed. I hope i didnt forget anything.

https://github.com/friedrichengel/esp32_multicast_sample

I hope it is ok that i put it in an extra repo for easier access. Quick and Dirty.

Debug Output:

```
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1288
load:0x40078000,len:13856
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3048
entry 0x40080590
[     2][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400da0f0
[    14][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400da0c0
[    27][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400da090
[    41][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400da060
[    54][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400da0f0
[    67][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400da0c0
[    81][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400da090
[    94][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400da060
[   108][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400da0f0
[   121][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400da0c0
[   135][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400da090
[   148][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400da060
��␙ɕ�Օ���5��J�␂���J␂�␂z␂�␂�␂���␂j��␂"յ�����ѥѥ���2�չ��jRT␅       �␓S�%HY.\5�KW�E���}���͡�␂r��␚�ɕ�"յ�����ѥѥ���2�չ��jR�␅           R��S�E��Uk�͒jB���␚�չ␚钲��␅���
␅%J␂������␂B�5R�[   627][I][esp32-hal-psram.c:90] psramInit(): PSRAM enabled
[  1973][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb6c
[  1984][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb6c
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 1.44
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   373508 B ( 364.8 KB)
  Free Bytes        :   329248 B ( 321.5 KB)
  Allocated Bytes   :    35480 B (  34.6 KB)
  Minimum Free Bytes:   323948 B ( 316.4 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  4194304 B (4096.0 KB)
  Free Bytes        :  4161192 B (4063.7 KB)
  Allocated Bytes   :    30916 B (  30.2 KB)
  Minimum Free Bytes:  4161192 B (4063.7 KB)
  Largest Free Block:  4128756 B (4032.0 KB)
[... 92 lines cut ...]
  Minimum Free Bytes:  4161192 B (4063.7 KB)
  Largest Free Block:  4128756 B (4032.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------
     0 : ETHERNET_CLK
     1 : UART_TX[0]
     3 : UART_RX[0]
    12 : ETHERNET_PWR
    18 : ETHERNET_MDIO
    19 : ETHERNET_RMII
    21 : ETHERNET_RMII
    22 : ETHERNET_RMII
    23 : ETHERNET_MCD
    25 : ETHERNET_RMII
    26 : ETHERNET_RMII
    27 : ETHERNET_RMII
============ After Setup End =============
UDP_SEND [0] to (ff02::1) test
UDP_SEND [0] to (ff02::1) test
UDP_SEND [0] to (ff02::1) test
UDP_SEND [0] to (ff02::1) test
```

closed 2025-01-15 by Parsaabasi