espressif/arduino-esp32 #9972: WPS is not working in version 3.0.2.

hugoviolin, 2024-07-01:
### Board

ESP32S3

### Device Description

ESP32S2 Dev Module

### Hardware Configuration

ESP32S2 Dev Module

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 10

### Flash frequency

240MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When using the WPS example from the WiFi library with esp32 version 3.0.2, it compiles perfectly, but when it starts the WPS function returns with the message "WPS Enable Failed" in the serial. This does not occur in version 2.0.17 or earlier. It works perfectly. What to do? Any suggestion? I await your response, thank you very much for your attention.

### Sketch

```cpp
/*
Example Code To Get ESP32 To Connect To A Router Using WPS
===========================================================
This example code provides both Push Button method and Pin
based WPS entry to get your ESP connected to your WiFi router.

Hardware Requirements
========================
ESP32 and a Router having WPS functionality

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#include "WiFi.h"
#include "esp_wps.h"
/*
Change the definition of the WPS mode
from WPS_TYPE_PBC to WPS_TYPE_PIN in
the case that you are using pin type
WPS
*/
#define ESP_WPS_MODE     WPS_TYPE_PBC
#define ESP_MANUFACTURER "ESPRESSIF"
#define ESP_MODEL_NUMBER "ESP32-S3"
#define ESP_MODEL_NAME   "ESP32S3 Dev Module"
#define ESP_DEVICE_NAME  "ESP STATION"

static esp_wps_config_t config;

void wpsInitConfig() {
  config.wps_type = ESP_WPS_MODE;
  strcpy(config.factory_info.manufacturer, ESP_MANUFACTURER);
  strcpy(config.factory_info.model_number, ESP_MODEL_NUMBER);
  strcpy(config.factory_info.model_name, ESP_MODEL_NAME);
  strcpy(config.factory_info.device_name, ESP_DEVICE_NAME);
}

void wpsStart() {
  if (esp_wifi_wps_enable(&config)) {
    Serial.println("WPS Enable Failed");
  } else if (esp_wifi_wps_start(0)) {
    Serial.println("WPS Start Failed");
  }
}

void wpsStop() {
  if (esp_wifi_wps_disable()) {
    Serial.println("WPS Disable Failed");
  }
}

String wpspin2string(uint8_t a[]) {
  char wps_pin[9];
  for (int i = 0; i < 8; i++) {
    wps_pin[i] = a[i];
  }
  wps_pin[8] = '\0';
  return (String)wps_pin;
}

// WARNING: WiFiEvent is called from a separate FreeRTOS task (thread)!
void WiFiEvent(WiFiEvent_t event, arduino_event_info_t info) {
  switch (event) {
    case ARDUINO_EVENT_WIFI_STA_START: Serial.println("Station Mode Started"); break;
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
      Serial.println("Connected to: " + String(WiFi.SSID()));
      Serial.println("Password: " + String(WiFi.psk()));
      Serial.print("Got IP: ");
      Serial.println(WiFi.localIP());

      break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
      Serial.println("Disconnected from station, attempting reconnection");
      WiFi.reconnect();
      break;
    case ARDUINO_EVENT_WPS_ER_SUCCESS:
      Serial.println("WPS Successful, stopping WPS and connecting to: " + String(WiFi.SSID()));
      wpsStop();
      delay(10);
      WiFi.begin();
      break;
    case ARDUINO_EVENT_WPS_ER_FAILED:
      Serial.println("WPS Failed, retrying");
      wpsStop();
      wpsStart();
      break;
    case ARDUINO_EVENT_WPS_ER_TIMEOUT:
      Serial.println("WPS Timedout, retrying");
      wpsStop();
      wpsStart();
      break;
    case ARDUINO_EVENT_WPS_ER_PIN: Serial.println("WPS_PIN = " + wpspin2string(info.wps_er_pin.pin_code)); break;
    default:                       break;
  }
}

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println();
  WiFi.onEvent(WiFiEvent);  // Will call WiFiEvent() from another thread.
  WiFi.mode(WIFI_MODE_STA);
  Serial.println("Starting WPS");
  wpsInitConfig();
  delay(1000);
  wpsStart();
}

void loop() {
  //nothing to do here
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880

Starting WPS
WPS Enable Failed
Station Mode Started
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-01:
This issue is triggered by an unintentional bug in the wifi layer. Because Arduino libs are compiled with PSRAM enabled, it tries to allocate it's memory in PSRAM and it fails if it is not enabled in Arduino. If you enable PSRAM, it will work, but... for ESP32-S3 PSRAM is currently broken in master and in 3.0.2. We have a fix, but it's not out yet. You could use 3.0.1 to test my theory, given that you have PSRAM in the module (R2 for QSPI and R8 for OPI)

everslick, 2024-08-22:
Does that mean that for the time being LwIP and WiFi will NOT allocate from PSRAM on all ESP32 models? If this is true is there a estimation for when this will be enabled again?

me-no-dev, 2024-08-22:
IDF WIFI team is working on it @everslick . I can not give estimates on their work. Currently WPS will not work on Xtensa ESP32 chips that do not have PSRAM or is not enabled. Other chips that do not support PSRAM are not affected.

everslick, 2024-08-22:
I'm asking because I see degrading stability when WiFi reception is bad and this led me to believe this is due to RAM pressure, because TCP cannot free its packet buffers as long as there are ACKs missing or still in flight.

Thus I further assume that disabling `CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP` will reduce the available RAM to the WiFi stack which, according to my theory, might impact stability negatively regardless of if WPS is used or which ESP32 chip.

But I'm aware this is mostly guesswork on my side and that I may be just worrying too much! ;-)

If you tell me the cavalry is on it's way and I will be fine - I'm happy. ;-)

me-no-dev, 2024-08-22:
I tried that and it did not help. So we need the fix in IDF

gbconnet, 2024-09-02:
Even version 3.0.4 does not solve the problem

me-no-dev, 2024-09-02:
there is a fix coming from IDF. We will close this issue once that get's to Arduino

closed 2025-01-15 by Parsaabasi