espressif/arduino-esp32 #9974: Chip Report Does Not Recognize Embedded Flash or Embedded PSRAM

Xylopyrographer, 2024-07-01:
### Board

seeed studio XIAO ESP32S3, Waveshare ESP32-S3-Matrix

### Device Description

Nothing else attached.

### Hardware Configuration

Nothing else attached.

### Version

v3.0.1

### IDE Name

Arduino IDE v2.3.2

### Operating System

Debian-GNU 12

### Flash frequency

240

### PSRAM enabled

yes

### Upload speed

921600

### Description

When running the sketch below with "Core Debug Level: Debug" on boards using ESP32-S3 chips with in-package (Embedded) flash and/or with in-package (Embedded) PSRAM, the Chip Debug Report shows "No" for Embedded Flash and/or Embedded PSRAM.

Tested using:
seeed studio XIAO ESP32S3 which uses an:
ESP32-S3R8 (8MB Octal SPI In-Package PSRAM)

Waveshare ESP32-S3-Matrix which uses an:
ESP32-S3FH4R2 (4 MB Quad SPI In-Package Flash and 2MB Quad SPI In-Package SPI PSRAM)
Note: As there is no board definition for this device, tested with the following boards selected:
ESP32S3 Dev Module
Lolin S3 Mini

Other boards using the ESP32-S3FH4R2 include:
WeAct Studio ESP32S3-Mini
Waveshare ESP32-S3 Mini
Waveshare ESP32-S3-Zero

### Sketch

```cpp
//
// Q&D sketch to see if embedded flash and/or PSRAM is reported.
//  - see '../esp32-3.0.1/cores/esp32/chip-debug-report.h'

#include "chip-debug-report.h"
#include <esp_psram.h>
#define PS_BUFFER 255

void setup() {

  printBeforeSetupInfo();   // comment out this line if Core Debug Level is set to 'Debug' or 'Verbose'

  Serial.begin(115200);
  Serial.print("\n\nPSRAM Found: ");
  if (psramFound()) {
    Serial.println("Yes");
    psramInit();
    Serial.print("On init, avaialble PSRAM: ");
    Serial.println(ESP.getFreePsram());
    Serial.println("Allocating and filling PSRAM data array.");
    uint8_t* data = (uint8_t*)ps_malloc(PS_BUFFER);
    for (int i = 0; i < PS_BUFFER; i++) {
      data[i] = i;
    }
    Serial.print("After write, avaialble PSRAM: ");
    Serial.println(ESP.getFreePsram());
    Serial.println("Dumping PSRAM data array:");
    for (int i = PS_BUFFER - 1; i >= 0; i--) {
      Serial.print(data[i]);
      Serial.print(", ");
      if ((PS_BUFFER - i) % 20 == 0) Serial.println();
    }
  }
  else {
    Serial.println("No");
  }
}

void loop() {
}
```

### Debug Message

```plain
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 1
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   405316 B ( 395.8 KB)
  Free Bytes        :   376892 B ( 368.1 KB)
  Allocated Bytes   :    23520 B (  23.0 KB)
  Minimum Free Bytes:   371844 B ( 363.1 KB)
  Largest Free Block:   335860 B ( 328.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8385672 B (8189.1 KB)
  Allocated Bytes   :      576 B (   0.6 KB)
  Minimum Free Bytes:  8385672 B (8189.1 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
  Bus Mode          : OPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  8388608 B (8 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  3264.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00340000, size:  3264.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00670000, size:  1536.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x007F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jul  1 2024 11:33:40
  Compile Host OS   : linux
  ESP-IDF Version   : v5.1.4-51-g442a798083-dirty
  Arduino Version   : 3.0.1
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : XIAO_ESP32S3
  Arduino Variant   : XIAO_ESP32S3
  Arduino FQBN      : esp32:esp32:XIAO_ESP32S3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=8M,PartitionScheme=default_8MB,DebugLevel=none,PSRAM=opi,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default
============ Before Setup End ============

PSRAM Found: Yes
On init, avaialble PSRAM: 8385672
Allocating and filling PSRAM data array.
After write, avaialble PSRAM: 8385400
Dumping PSRAM data array:
254, 253, 252, 251, 250, 249, 248, 247, 246, 245, 244, 243, 242, 241, 240, 239, 238, 237, 236, 235, 
234, 233, 232, 231, 230, 229, 228, 227, 226, 225, 224, 223, 222, 221, 220, 219, 218, 217, 216, 215, 
214, 213, 212, 211, 210, 209, 208, 207, 206, 205, 204, 203, 202, 201, 200, 199, 198, 197, 196, 195, 
194, 193, 192, 191, 190, 189, 188, 187, 186, 185, 184, 183, 182, 181, 180, 179, 178, 177, 176, 175, 
174, 173, 172, 171, 170, 169, 168, 167, 166, 165, 164, 163, 162, 161, 160, 159, 158, 157, 156, 155, 
154, 153, 152, 151, 150, 149, 148, 147, 146, 145, 144, 143, 142, 141, 140, 139, 138, 137, 136, 135, 
134, 133, 132, 131, 130, 129, 128, 127, 126, 125, 124, 123, 122, 121, 120, 119, 118, 117, 116, 115, 
114, 113, 112, 111, 110, 109, 108, 107, 106, 105, 104, 103, 102, 101, 100, 99, 98, 97, 96, 95, 
94, 93, 92, 91, 90, 89, 88, 87, 86, 85, 84, 83, 82, 81, 80, 79, 78, 77, 76, 75, 
74, 73, 72, 71, 70, 69, 68, 67, 66, 65, 64, 63, 62, 61, 60, 59, 58, 57, 56, 55, 
54, 53, 52, 51, 50, 49, 48, 47, 46, 45, 44, 43, 42, 41, 40, 39, 38, 37, 36, 35, 
34, 33, 32, 31, 30, 29, 28, 27, 26, 25, 24, 23, 22, 21, 20, 19, 18, 17, 16, 15, 
14, 13, 12, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 0,
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-01:
Is the flash not being automatically initialized if you don't call psramInit?

Xylopyrographer, 2024-07-01:
Same result with:
`    psramInit();`
commented out.

So, no, that line is not required.

The chip report shows "No" for Embedded Flash and for Embedded PSRAM.

me-no-dev, 2024-07-01:
We just print what the chip says. We have no control over that and has nothing to do with actual config. It should show what is in the package itself, not what is under the module can.

Xylopyrographer, 2024-07-01:
That is how I thought it should be.

In the boards tested one, the  XIAO ESP32S3, uses an ESP32-S3R8 which has 8MB Octal SPI In-Package (Embedded) PSRAM.

The other board, the Waveshare ESP32-S3-Matrix, uses an ESP32-S3FH4R2 which has 4 MB Quad SPI In-Package (Embedded) Flash and 2MB Quad SPI In-Package (Embedded) SPI PSRAM.

Should these not be reflected in the "Chip Info" section of the chip debug report?

me-no-dev, 2024-07-01:
They should, but I guess their fuses were not burned or not marked in production. Again, we just print what the chip tells us

Xylopyrographer, 2024-07-01:
OK. Gotcha. QA problem at Espressif I guess.

Thanks.

closed 2024-07-01 by Xylopyrographer