espressif/arduino-esp32 #9976: i2c_param_config call causes change in Wire.cpp private variable "is_slave"

spivey, 2024-07-02:
### Board

ESP32 DEVKIT V1

### Device Description

ESP32 DEVKIT V1

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino 2.3.2

### Operating System

Windows 11

### Flash frequency

40 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

In Wire.cpp, there is a call is made to i2cInit() in esp32-hal-i2c.c - which in turn has a call to the i2c_param_config() function in the esp_idf/components/driver/i2c/i2c.c file. There is a private variable in Wire.cpp (is_slave) - whose value is being changed if the i2c_param_confi() is executed. This should not be happening (as it is a private variable in Wire.cpp). The value of is_slave is 0 before this call, and after this call, the value has been changed to 144

### Sketch

```cpp
#ifdef CORE_DEBUG_LEVEL
#undef CORE_DEBUG_LEVEL
#endif

#define CORE_DEBUG_LEVEL ARDUHAL_LOG_LEVEL_VERBOSE
#define LOG_LOCAL_LEVEL ESP_LOG_DEBUG

#include "esp32-hal-log.h"
#include <Arduino.h>
#include <Wire.h>

// Create TwoWire instances for I2C0 and I2C1
TwoWire I2C0 = TwoWire(0);
// TwoWire I2C1 = TwoWire(1);

void setup() {
  // Start serial communication for debugging
  Serial.begin(115200);
  delay(1000); // Give time for the serial monitor to initialize

  // test the logs to make sure they are working
  log_v("This is a verbose log message");
  log_d("This is a debug log message");
  log_i("This is an info log message");
  log_w("This is a warning log message");
  log_e("This is an error log message");

  if (I2C0.begin()) {
    Serial.println("I2C0 Initialized");
  } else {
    Serial.println("Failed to initialize I2C0");
  }

  I2C0.setClock(400000);

}

void loop() {
  // Perform a transmission on I2C0
  Serial.println("Starting I2C0 Transmission to address 0x50");
  //I2C0.beginTransmission(0x50);
  //I2C0.write(0);

  // Check for errors in the transmission
  uint8_t error0 = I2C0.endTransmission(true);
  if (error0 == 0) {
    Serial.println("Transmission on I2C0 successful.");
  } else {
    Serial.print("Error in transmission on I2C0: ");
    Serial.println(error0);
  }

  delay(1000); // Delay to avoid flooding the bus
}
```

### Debug Message

```plain
(after modifying Wire.cpp to add in the log_i for the value of is_slave before and after the i2cInit() call (in TwoWire::Begin - the master call)

[  1556][I][Wire.cpp:311] begin(): is_slave is = 0
[  1561][I][esp32-hal-i2c.c:109] i2cInit(): Initializing I2C Master: sda=21 scl=22 freq=100000
[  1570][I][Wire.cpp:316] begin(): AFTER i2cInit
[  1574][I][Wire.cpp:317] begin(): is_slave is = 144
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

spivey, 2024-07-05:
Turns out the problem is in naming the variable I2C0 - apparently that is colliding with something else.

me-no-dev, 2024-07-05:
Yes. that is the name of the peripheral structure itself. Glad you found it. Can we close this now?

spivey, 2024-07-05:
Yes - it would have been nice if somewhere in the process there was some notification of a name collision - especially when I recast I2C0 as a TwoWire ... not sure why that didn't error out at some point.

closed 2024-07-05 by spivey