espressif/arduino-esp32 #9978: ESP32-C6 doesn't compile with zigbee and WiFi.h

Nanowires, 2024-07-02:
### Board

ESP32-C6

### Device Description

-

### Hardware Configuration

-

### Version

v3.0.1

### IDE Name

Arduino IDE and VSCode

### Operating System

Fedora 39

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

-

### Description

When try to compile a Zigbee application together with a WiFi.h header, I get the following error:

```
/home/<user>/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_transmit_done':
(.iram1.0+0x0): multiple definition of `esp_ieee802154_transmit_done'; /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.1+0x0): first defined here
/home/<user>/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_receive_done':
(.iram1.4+0x0): multiple definition of `esp_ieee802154_receive_done'; /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.2+0x0): first defined here
/home/<user>/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_transmit_failed':
(.iram1.5+0x0): multiple definition of `esp_ieee802154_transmit_failed'; /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.4+0x0): first defined here
/home/<user>/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_receive_sfd_done':
(.iram1.6+0x0): multiple definition of `esp_ieee802154_receive_sfd_done'; /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.5+0x0): first defined here
/home/<user>/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_transmit_sfd_done':
(.iram1.7+0x0): multiple definition of `esp_ieee802154_transmit_sfd_done'; /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.6+0x0): first defined here
/home/<user>/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_energy_detect_done':
(.iram1.8+0x0): multiple definition of `esp_ieee802154_energy_detect_done'; /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.7+0x0): first defined here
/home/<user>/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_cca_done':
(.iram1.9+0x0): multiple definition of `esp_ieee802154_cca_done'; /home/<user>/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.8+0x0): first defined here
collect2: error: ld returned 1 exit status

Error during build: exit status 1
```

When I remove the WiFi.h header, everything compiles as expected.

### Sketch

```cpp
e.g. libraries/ESP32/examples/Zigbee/Zigbee_Light_Switch/Zigbee_Light_Switch.ino

once without #include <WiFi.h> and once with.
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-02:
WiFi and Zigbee do not coexist at this time.  Pick one or the other.

Nanowires, 2024-07-02:
Do you have some proof for this?

As they share the same medium, I guess performance is not the best, but theoretically it should work.
Also according to this page, they can coexists:
https://github.com/espressif/esp-idf/issues/10423

> Coexistence of Wi-Fi and 802.15.4 - Supported

lbernstone, 2024-07-02:
Nope. Guess I was working on old information

P-R-O-C-H-Y, 2024-07-03:
From what I see from the debug log, there is a conflict between Zigbee and OpenThread libs.
I will investigate this. Thank you for reporting.

P-R-O-C-H-Y, 2024-07-29:
Hi @Nanowires, I finally get to test/investigate this issue. However, on latest master I was not able to reproduce this.
I was testing on C6 running both Zigbee as end device or coordinator. Including WiFi did not cause any compilation error.
I also added a wifi connection to the sketch with successful connect to wifi with also successful Zigbee network formation.
See attached log:
```
12:29:08.378 -> Connecting to MyWiFI
12:29:08.940 -> ...
12:29:09.947 -> WiFi connected
12:29:09.947 -> IP address: 
12:29:09.947 -> 192.168.33.161
12:29:10.128 -> [  2035][I][Zigbee_Light_Switch.ino:208] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
12:29:10.128 -> [  2046][I][Zigbee_Light_Switch.ino:152] esp_zb_app_signal_handler(): Zigbee stack initialized
12:29:10.128 -> [  2056][I][Zigbee_Light_Switch.ino:158] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
12:29:10.128 -> [  2066][I][Zigbee_Light_Switch.ino:160] esp_zb_app_signal_handler(): Start network formation
12:29:12.515 -> [  4441][W][Zigbee_Light_Switch.ino:204] esp_zb_app_signal_handler(): Network(0x32a8) closed, devices joining not allowed.
12:29:12.515 -> [  4453][I][Zigbee_Light_Switch.ino:175] esp_zb_app_signal_handler(): Formed network successfully (Extended PAN ID: 60:55:f9:00:00:f7:52:d0, PAN ID: 0x32a8, Channel:25, Short Address: 0x0000)
12:29:13.028 -> [  4937][I][Zigbee_Light_Switch.ino:202] esp_zb_app_signal_handler(): Network(0x32a8) is open for 180 seconds
12:29:13.028 -> [  4947][I][Zigbee_Light_Switch.ino:188] esp_zb_app_signal_handler(): Network steering started
```

Can you please retest on latest version 3.0.3 or on latest master? Thank you

Nanowires, 2024-07-29:
Hi @P-R-O-C-H-Y,
Thanks for the response.
Indeed, with v3.0.3 I can not reproduce the issue, too.
Seems like something from 3.0.2 to 3.0.2 fixed this issue :-)

closed 2024-07-29 by Nanowires

lsroka76, 2024-12-23:
IT seems that problem is back with 3.1.0 :-(