espressif/arduino-esp32 #9981: Unable to setPin to connect to Bluetooth ELM327

JayKapse, 2024-07-03:
### Board

ESP32-D0WDQ5

### Device Description

No hardware, directly connected to Mac for power and flashing. 

### Hardware Configuration

No hardware, directly connected to Mac for power and flashing. 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS Sonoma v14.5 (23F79)

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm not able to use the setPin parameter. I have tried connecting the same ELM327 adapter using an android tablet and it works with the pin code "1234". I have tried running the code without the setPin parameter and it is not able to connect to the adapter. 

I am trying to use the ELMduino library for the project, but the error comes from the ESP32 board installation according to the path displayed in the console. 

### Sketch

```cpp
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  ELM_PORT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
```

### Debug Message

```plain
/private/var/folders/fm/2sf3tvf92vxcwskbwrwpjpyr0000gn/T/.arduinoIDE-unsaved202463-67124-5xxjv0.o4u6/sketch_jul3a/sketch_jul3a.ino: In function 'void setup()':
/private/var/folders/fm/2sf3tvf92vxcwskbwrwpjpyr0000gn/T/.arduinoIDE-unsaved202463-67124-5xxjv0.o4u6/sketch_jul3a/sketch_jul3a.ino:24:12: error: 'class BluetoothSerial' has no member named 'setPin'
   24 |   ELM_PORT.setPin("1234");
      |            ^~~~~~

exit status 1

Compilation error: 'class BluetoothSerial' has no member named 'setPin'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-03:
[SSP is enabled in newer releases](https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial#pairing-options). To use a fixed pin, you would need to compile your own libraries.

brugomes1407, 2024-09-04:
Hi, I have the same problem but I found a solution below:
Insert 

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif

#define BT_DISCOVER_TIME 10000
esp_spp_sec_t sec_mask = ESP_SPP_SEC_NONE;  // or ESP_SPP_SEC_ENCRYPT|ESP_SPP_SEC_AUTHENTICATE to request pincode confirmation
esp_spp_role_t role = ESP_SPP_ROLE_SLAVE;   // or ESP_SPP_ROLE_MASTER
uint8_t address[6]  = {0x00, 0x10, 0xCC, 0x4F, 0x36, 0x03};

Insert in **void setup()**
   if (!ELM_PORT.connect(address, **sec_mask, role**))

closed 2025-01-15 by Parsaabasi