espressif/arduino-esp32 #9983: Switch between network interfaces

MehdiSadeghii, 2024-07-03:
### Board

ESP32 Dev Module

### Device Description

When using Ethernet and WiFi and PPP together, we can't switch between interfaces easily all interfaces, there's jut printTo method which we can print them. now my idea is to have a function which returns all NetworkInterface and when we want to set default interface we can choose interface from the list we extracted.

This feature is needed when we have more than one interfaces but our default interface doesn't have access to the internet. In this case we need to switch to another interface. 

### Hardware Configuration

gpio 16, 17 used for SIM800,
VSPI is used for ETH

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

when my WiFi doesn't have access to the internet I need to switch to another interface. 

### Sketch

```cpp
/*
    This sketch shows the Ethernet event usage

*/
#include <PPP.h>
#include <WiFi.h>
#include <ETH.h>
#include <SPI.h>

#include <Ticker.h>

#include <PubSubClient.h>

#define PPP_MODEM_APN "mtnirancell"
#define PPP_MODEM_PIN "0000" // or NULL

NetworkClient net;
PubSubClient mqtt("xxx", 1883, net);

Ticker GSMTimer;

#ifndef ETH_PHY_TYPE
#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
[... 278 lines cut ...]
Failed to connect to internet!
Failed to connect to internet!
Failed to connect to internet!
MQTT connection failed, rc= -2
******************************************************
Changing Interface....
*sta: <UP,BehamadModem,CH:11,RSSI:-34,N,WPA2_PSK> (DHCPC,GARP,IP_MOD,V6_REP)
      ether FC:B4:67:D7:07:00
      inet 192.168.199.229 netmask 255.255.255.0 broadcast 192.168.199.255
      gateway 192.168.199.1 dns 192.168.208.1

ppp: <DOWN,RSSI:20,BER:0> (,PPP)
      ether 00:00:00:00:00:00
      inet 0.0.0.0 netmask 255.255.255.255 broadcast 0.0.0.0
      gateway 0.0.0.0 dns 192.168.208.1

eth0: <UP,100M,FULL_DUPLEX,AUTO,ADDR:0x1> (DHCPC,GARP,IP_MOD)
      ether FE:B4:67:D7:07:00
      inet 192.168.208.194 netmask 255.255.255.0 broadcast 192.168.208.255
      gateway 192.168.208.1 dns 192.168.208.1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-03:
it's up to you to manage your interfaces. You have methods to check which one is connected and to set it as default outgoing interface. You have events, you can implement that there. I'm not sure that we need to implement something more than that.

BluesBilly, 2024-07-05:
if (NetInterface == "WIFI") {
  WiFi.STA.isDefault();
} else if (NetInterface == "PPP") {
  PPP.isDefault();
} else if (NetInterface == "ETH") {
  ETH.isDefault();
}

JAndrassy, 2024-07-05:
@BluesBilly you meant to use setDefault()?

MehdiSadeghii, 2024-07-06:
> if (NetInterface == "WIFI") { WiFi.STA.setDefault(); } else if (NetInterface == "PPP") { PPP.setDefault(); } else if (NetInterface == "ETH") { ETH.setDefault(); }

Thanks Problem Solved..

MehdiSadeghii, 2024-07-06:
> it's up to you to manage your interfaces. You have methods to check which one is connected and to set it as default outgoing interface. You have events, you can implement that there. I'm not sure that we need to implement something more than that.

Yeah You're correct. it was m confusion about managing interfaces. I thought that I have to manage it using mid level APIs but now I see it's easier than I thought. Thanks a lot *_* .

closed 2024-07-06 by MehdiSadeghii