espressif/arduino-esp32 #9985: printing >= 4 times with the esp_log.h library conflicts with Serial.print() (IDFGH-13196)

shyney7, 2024-07-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-arduinoespressif32 @ 3.20016.0 (2.0.16)

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

WEMOS LOLIN S3 PRO

### Power Supply used.

USB

### What is the expected behavior?

Using the esp_log.h library sould not conflict with Serial.print() function.

### What is the actual behavior?

When printing e.g. with the ESP_LOGD() function >= 4 times and then using Serial.print() to output something to the serial monitor no output is visible. The same problem was already reported 4 years ago here: https://github.com/espressif/arduino-esp32/issues/4261 . This issue was closed although the problem seems to still exist. It doesnt matter wich print function you use (Serial.printf() etc.) the output is not visible besides the output from esp_log.h

### Steps to reproduce.

1. Include esp_log.h
2. Use any print function e.g. ESP_LOGD() equal or more than 4 times subsequently
3. Use any Serial print function like Serial.print() or Serial.printf() right after
4. Observe that there is no output to the serial monitor besides the ESP_LOGD() output.

me-no-dev, 2024-07-04:
Please post minimal example code that will reproduce the problem. there is no reason IDF's LOG to interfere with Serial output.

shyney7, 2024-07-04:
I've uploaded my code with the Arduino IDE instead of platformio and its working now. I dont know what caused this issue in the first place. It was so far that no matter if I used the esp_log library or not the Serial.print() statements were not output to the serial monitor. I don't know what caused this to happen but it doesn't seem to be the esp_log library. After I uploaded the sketch with the Arduino IDE and uploaded my code via full clean back in platformio it worked.

closed 2024-07-04 by shyney7