espressif/arduino-esp32 #9986: Error on POST request "send header failed" -2

JesusLizAlles, 2024-07-04:
### Board

ESP32 Dev Board

### Device Description

I have an ESP32 with a QR code reader

### Hardware Configuration

I2C devices

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I get this error from time to time when i'm sending the request to the server, it works a few times then sometimes sends this error message "send header failed"

I dont know if it can be memory issues, it started doing it when updating to V3, with V2.0,0 works perfect

### Sketch

```cpp
#include <WiFi.h>
#include <HTTPClient.h>
#include <SPI.h>
#include <Wire.h>
#include <ArduinoJson.h>

/* Constants */
const char* ssid = "xxx";
const char* pass = "xxx";

#define RXD2 16
#define TXD2 17
#define PUERTA 19

HTTPClient http;

void setup() {
  pinMode(PUERTA, OUTPUT);
  digitalWrite(PUERTA, HIGH);

  Serial.begin(9600);
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);

  connectToWiFi();
  Serial2.setTimeout(200);
}

void loop() {
  String lectura = Serial2.readString();
  if(lectura != ""){ 
    handleRequest(lectura);
  }
}

void connectToWiFi() {
  WiFi.begin(ssid, pass);
  WiFi.setSleep(false);

  Serial.println("Connecting to WiFi...");
  while(WiFi.status() != WL_CONNECTED){
    delay(500);
    Serial.print(".");
  }

  Serial.println("\nWiFi Connected");
  Serial.println("IP Address: " + WiFi.localIP().toString());
  Serial.println("MAC Address: " + WiFi.macAddress());

  delay(1000);
  togglePuerta();
}

int statusCode = 0;
int httpCode = 0;

void handleRequest(const String& lectura) {
  const String url = "my_app_url_here";
  
  DynamicJsonDocument doc(256);
  doc["code"] = lectura;

  String data;
  serializeJsonPretty(doc, data);
  Serial.println(data);

  http.begin(url);
  http.addHeader("Content-Type", "application/json; charset=UTF-8");       
  
  int httpCode = http.POST(data);
  
  if (httpCode > 0) { // Check for the returning code
    String payload = http.getString();
    Serial.println(payload);

    DynamicJsonDocument responseDoc(512);
    DeserializationError error = deserializeJson(responseDoc, payload);

    if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.f_str());
    } else {         
      statusCode = responseDoc["status_code"];
      if(statusCode == 200){
        togglePuerta();
      }
    }
  } else {
    Serial.println("Error on HTTP request");
    Serial.println(http.errorToString(httpCode).c_str());
  }

  http.end();
}

void togglePuerta() {
  digitalWrite(PUERTA, LOW);
  delay(1000);
  digitalWrite(PUERTA, HIGH);
}
```

### Debug Message

```plain
send header failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-10:
can you try with 3.0.2 or latest master? I think we fixed some of those

P-R-O-C-H-Y, 2024-07-29:
Hi @JesusLizAlles, I have tested HTTP post (your sketch) on latest master by running it 1000 times in a loop and I got 0 errors. 
Please retest on the last release 3.0.3 or latest master and let us know if you are still seeing errors. If yes, feel free to reopen this issue.

closed 2024-07-29 by P-R-O-C-H-Y