espressif/arduino-esp32 #9987: WiFi disconnects after 48h-20min but reconnects again after 48h-20min

AndyKoder, 2024-07-04:
### Board

ESP32-Wroom32 (ESP32 Dev Module)

### Device Description

Module on breadboard

### Hardware Configuration

DHT20

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

ubuntu linux

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

WiFi connects, but after 48h - ~20mins it disconnects and could not reconnect.
But after another 48h - ~20mins (without connection) it suddenly is able to reconnect again.
It is otherwise fully working, serial console shows normal output of wifi status, temp every 60s
(this does not happen with 1.0.6)
I don't see a reason anywhere, in the WiFi AP/router log i see only lots of "esp32-xxxx connects/disconnects"

### Sketch

```cpp
[Basically this example](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientConnect/WiFiClientConnect.ino)
(changed only ssid+pass, some debug msgs every 60s, dht20 temp, everything working, but wifi is disconnected)

void printWifiStatus() {
   Serial.printf("WiFi.getHostname[%s]\n", WiFi.getHostname());
   Serial.printf("MAC WiFi.macAddress() [%s]\n", WiFi.macAddress().c_str());
   Serial.printf("SSID[%s] IP[%s] RSSI[%ld dBm]\n", WiFi.SSID().c_str(), WiFi.localIP().toString().c_str(), (long)WiFi.RSSI());
Serial.printf("Status[%d]\n", WiFi.status());
}
```

### Debug Message

```plain
WiFi.status() gets 0 that is WL_IDLE_STATUS
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-04:
Please enable core debug level to verbose and let's see what reason the WiFi gives for disconnecting. 2900 minutes is oddly specific and in no way around some 32 bit overflow (or similar)

AndyKoder, 2024-07-06:
```
[168428788][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 14 - STA_DISCONNECTED
[168428797][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 14 - STA_DISCONNECTED
[168428805][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
```

me-no-dev, 2024-07-09:
And what reason does it give when it fails to reconnect after that?

AndyKoder, 2024-07-13:
Compiled the same code with 1.0.6, runs for a week now.
In the router log i see 3 disconnects after 42h, 47h and 53h,
but it ALWAYS reconnected right after the disconnect.
So maybe it's normal that it disconnects from time to time, but the reconnect part in the 3.0.x doesn't work for some reason...
btw: how would i get the `Reason 8 ASSOC_LEAVE` error from inside the esp32, so i could log that without using serial console?

TD-er, 2024-07-13:
Do you have "band steering" enabled on your access point?
If so, try disabling it.
Of try to connect you ESP with forced 802.11g mode (not 802.11n)

AndyKoder, 2024-08-19:
> And what reason does it give when it fails to reconnect after that?
```
[486788][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 13 - STA_CONNECTED
[486795][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 13 - STA_CONNECTED
```
and then after a couple of seconds again
```
[546732][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 14 - STA_DISCONNECTED
[546740][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 14 - STA_DISCONNECTED
[546748][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
```

AndyKoder, 2024-08-19:
> Do you have "band steering" enabled on your access point? If so, try disabling it. Of try to connect you ESP with forced 802.11g mode (not 802.11n)
I do not see any "band steering" on the router.
How would i tell the ESP32 to use 802.11g?

TD-er, 2024-08-19:
```c++
uint8_t protocol = WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G; // Default to BG
esp_wifi_set_protocol(WIFI_IF_STA, protocol);
```

And you could also set the STA interface to use "Long GI" making it more resilliant to reflections
See: https://www.tp-link.com/us/configuration-guides/q_a_basic_wireless_concepts/?configurationId=2958#_idTextAnchor038
```c++
esp_wifi_config_80211_tx_rate(WIFI_IF_STA, WIFI_PHY_RATE_MCS3_LGI);
```

N.B. you need to set these before you make the connection.

lbernstone, 2024-08-19:
This sounds a lot like the router is refusing to allocate a DHCP address.  Can you try a manual IP assignment and see if it has the same issue?  This will help identify what the router is having a problem with.

closed 2025-01-15 by Parsaabasi