espressif/arduino-esp32 #9989: WDT init compilation error

Burdi60, 2024-07-05:
### Board

ESP32-WROOM

### Device Description

ESP32-WROOM module on a specific PCB, to format information received on a serial port and transmit it to a WEB server

### Hardware Configuration

Nothing specific

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

In the latest versions of the ESP32 card manager (version 3.0.0 to 3.0.2), compiling the esp_task_wdt_init() function generates an error.
There are no compilation errors with earlier 2.0.X versions (2.0.17 for example).

### Sketch

```cpp
#include <Arduino.h>
#include <esp_task_wdt.h>

#define WDT_DELAI 54

void setup() {
  //
  // initialisation du watchdog
  //esp_task_wdt_init(WDT_DELAI, true);
  esp_task_wdt_init(WDT_DELAI, true);
  esp_task_wdt_add(NULL);

}

void loop() {

}
```

### Debug Message

```plain
C:\Users\Pierre\AppData\Local\Temp\.arduinoIDE-unsaved202465-23444-mhiyh.ip69lj\sketch_jul5a\sketch_jul5a.ino: In function 'void setup()':
C:\Users\Pierre\AppData\Local\Temp\.arduinoIDE-unsaved202465-23444-mhiyh.ip69lj\sketch_jul5a\sketch_jul5a.ino:4:19: error: invalid conversion from 'int' to 'const esp_task_wdt_config_t*' [-fpermissive]
    4 | #define WDT_DELAI 54
      |                   ^~
      |                   |
      |                   int
C:\Users\Pierre\AppData\Local\Temp\.arduinoIDE-unsaved202465-23444-mhiyh.ip69lj\sketch_jul5a\sketch_jul5a.ino:10:21: note: in expansion of macro 'WDT_DELAI'
   10 |   esp_task_wdt_init(WDT_DELAI, true);
      |                     ^~~~~~~~~
C:\Users\Pierre\AppData\Local\Temp\.arduinoIDE-unsaved202465-23444-mhiyh.ip69lj\sketch_jul5a\sketch_jul5a.ino:10:20: error: too many arguments to function 'esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t*)'
   10 |   esp_task_wdt_init(WDT_DELAI, true);
      |   ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
In file included from C:\Users\Pierre\AppData\Local\Temp\.arduinoIDE-unsaved202465-23444-mhiyh.ip69lj\sketch_jul5a\sketch_jul5a.ino:2:
C:\Users\Pierre\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/esp_system/include/esp_task_wdt.h:47:11: note: declared here
   47 | esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t *config);
      |           ^~~~~~~~~~~~~~~~~

exit status 1

Compilation error: invalid conversion from 'int' to 'const esp_task_wdt_config_t*' [-fpermissive]
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-05:
The [API](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/system/wdts.html#_CPPv421esp_task_wdt_config_t) has changed. This is part of ESP-IDF, not arduino.

Burdi60, 2024-07-06:
I use Arduino IDE and not ESP-IDF.
The ESP32 board driver is installed in the Arduino IDE application.
The compilation error is obtained in the Arduino IDE application and not ESP-IDF which I do not use.

me-no-dev, 2024-07-06:
But you are using ESP_IDF APIs. What you posted is NOT part of Arduino API, but from the underlying ESP-IDF that is included with Arduino

me-no-dev, 2024-07-06:
Closing as answered

closed 2024-07-06 by me-no-dev

Burdi60, 2024-07-07:
So who should I report this bug to?
The compilation error is indeed generated by the Arduino-IDE compiler!!!
I was just asking for help, and you say I need to close the post. For what ?
Be a little more forthcoming in your answers!

me-no-dev, 2024-07-09:
This is not a bug. Arduino has changed to new MAJOR version, which also changes the ESP-IDF's MAJOR version, which has changes in it's API. You need to implement the new API if you want your ESP-IDF code to work. @lbernstone has highlighted the updated header where you can read the new API.

Burdi60, 2024-07-09:
Thank you for your reply.
Indeed, watchdog management on the ESP32 has been very significantly modified by ESPRESSIF.
I just modified my code and I no longer have any compilation errors.