espressif/arduino-esp32 #9994: Random UART Bug, hard to replicate

terrafirma2021, 2024-07-08:
### Board

YD ESP32-S3 devkit (clone)

### Device Description

devkitc1 clone
n16r8

### Hardware Configuration

Another devkit clone linked via gpio 1,2

### Version

other

### IDE Name

Platformio Latest 6.7.0

### Operating System

windows 10

### Flash frequency

40 mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Random UART bug slows down the UART RX, causing the bytes to fill up untill buffer overflow occurs

I am able to reproduce this same behaviour if it upload code to the esp32 s3 and as soon as the esp starts send uart data to the esp

23:19:35.156 -> [  1402][E][USBHID.cpp:346] SendReport(): not ready

will be printed and the uart will slow down, and overflow until uart data stops, then the esp will function normally.

However, this is the closest way to produce the bug, however the bug occurs whilst the esp is running normally.

If this does not make sense then please forgive me. 

### Sketch

```cpp
To fix, I have to either:

Stop sending UART data to allow the ESP to resume normal UART COM speed.

Or 

void handleUnknown(const Command &cmd) {
    Serial0.println("ESP UART bug occurred");
    
        // Disable and re-enable UART to reset
    uart_disable_intr_mask(UART_NUM_1, 0xFFFFFFFF);

    // Quick UART reset
    uart_flush_input(UART_NUM_1);

    uart_enable_intr_mask(UART_NUM_1, 0xFFFFFFFF);
}

also, using serial.end and serial.begin fixes this issue,

I am using the hid device lib (Mouse)

I have tried to debug the issue, but there is no ability to read from the serial hardware, nor is there any known reason for this in my code, this only seems to occur whilst using the USB HID MOUSE

To reproduce expected behavoiur
```

### Debug Message

```plain
No debug, no crash
```

### Other Steps to Reproduce

My repo is on my github ESP_KMBOX if anybody wants to load and test.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-07-09:
@terrafirma2021 - Arduino is about `HardwareSerial` class only.

There is nothing that we can do if the application changes IDF ISR driver directly.

This issue is not Arduino related.

SuGlider, 2024-07-09:
@terrafirma2021 - please provide a clear step by step procedure, in addition to a Arduino Sketch, that could be used to replicate the issue.

I also need a description about what was expected as output and what has gone wrong.

Thanks.

terrafirma2021, 2024-07-13:
@SuGlider Sorry for not reaching back sooner, I was updating my code, I will create a super lightweight code for you asap.

terrafirma2021, 2024-07-13:
@SuGlider i have made a test code, i will ensure that the bug still happens then upload for you

terrafirma2021, 2024-07-16:
Going to close this error for now, because it may be due to a bad usbc port. 

When testing further there was no issue, so the USB-C may of been the issue

closed 2024-10-30 by terrafirma2021