espressif/arduino-esp32 #10001: W5500 crash : Stack canary watchpoint triggered (w5500_tsk)

tresler, 2024-07-09:
### Board

ESP32-S3 

### Device Description

own ESP32s3-wroom LED output schema

### Hardware Configuration

#define ETH_PHY_TYPE        ETH_PHY_W5500
#define ETH_PHY_ADDR         1
#define ETH_PHY_CS          10
#define ETH_PHY_IRQ          9
#define ETH_PHY_RST          14
#define ETH_PHY_SPI_HOST    SPI2_HOST
#define ETH_PHY_SPI_SCK     12
#define ETH_PHY_SPI_MISO    13
#define ETH_PHY_SPI_MOSI    11

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Ubuntu 24.04

### Flash frequency

240MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have problem with <ETH.h> and W5500. I use [W5500-io](https://docs.wiznet.io/Product/ioModule/W5500-io) modul and everything running fine (elegantOTA, async webserver, artnet, sacn) until I send more package from lighting console. No matter if it by sACN or ArtNet. If I use WiFi and send everything running without crash, but after I switch to Ethernet it send me this and restart ESP.

### Sketch

```cpp
#include <WiFi.h>
#include <ETH.h>
#include <ArtnetETH.h>
#include <sACN.h>
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (w5500_tsk) 
Core  1 register dump:
PC      : 0x40380712  PS      : 0x00060236  A0      : 0x8037de72  A1      : 0x3fcad920  
A2      : 0x3fcad298  A3      : 0xffffffff  A4      : 0x00060223  A5      : 0x00060223  
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80377b30  A9      : 0x3fcad900  
A10     : 0x3fcad24c  A11     : 0x00000804  A12     : 0x00000001  A13     : 0x00000004  
A14     : 0x0000180c  A15     : 0x3fca6b28  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  

Backtrace: 0x4038070f:0x3fcad920 0x4037de6f:0x3fcad950 0x4037e005:0x3fcad970 0x4037e22c:0x3fcad990 0x40378a9e:0x3fcad9b0 0x40378ae2:0x3fcad9d0 0x40378c09:0x3fcada10 0x4202ee8d:0x3fcada30 0x4202dd75:0x3fcada60 0x4202d791:0x3fcada80 0x420b19cd:0x3fcadaa0 0x420afea5:0x3fcadac0 0x420aff2a:0x3fcadae0 0x420b0ccf:0x3fcadb10 0x420c0c4f:0x3fcadb40 0x420b9fba:0x3fcadb60 0x420ba125:0x3fcade80 0x420cce75:0x3fcadeb0 0x4206fc97:0x3fcadee0 0x4206fcbd:0x3fcadf10 0x4205d0b5:0x3fcadf60 0x4038060e:0x3fcadfa0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-09:
@me-no-dev Maybe need to expose some parameter for rx_task_stack_size in https://github.com/espressif/arduino-esp32/blame/master/libraries/Ethernet/src/ETH.cpp#L168?

me-no-dev, 2024-07-10:
Thanks @lbernstone . Feature added here: https://github.com/espressif/arduino-esp32/pull/10003

closed 2024-07-10 by me-no-dev

tresler, 2024-07-11:
Thanks for helping. Unfortunately now I have after few second new error:

'''
E (198396) w5500.mac: received frame was truncated
'''