espressif/arduino-esp32 #10004: Cannot open source file "fileName.h" (dependency or "Arduino.h") suddently happening after a git.merge

Victor-stll, 2024-07-10:
### Board

ESP32-S3

### Device Description

I use a custom board, but it has been tested previously and worked perfectly fine.

### Hardware Configuration

Just using the USB port to connect with my laptop

### Version

latest development Release Candidate (RC-X)

### IDE Name

VSCode

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

After i tried to merge a branch i worked on on my master branch, my code suddently stop building, and some libraries that were used in the previous branch started being unable to open.
i can't build my code because some libraries aren't found by vscode. Even when i define the includepath in the c_pp_properties.json for each of them. 
Not only the merged project didn't built, i picked up the previous version of my code from the commit just before the merge, and this one wasn't building anympore too
So i tried to come back to a very simple project like the hello_world template from ESP-IDF, to see if it was only coming from my program, but turns out the same kind of problem occurs. i followed the "Arduino as a ESP-IDF component" tutorial on the espressif website (https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html) to be sure i made things correctly but still
[image]
currently the file that cannot be open is "portmacro.h" but it wasn't the same on the previous program (can't remember the name) and it changed several times.

i tried to run both of the code on ESP-IDF v5.2.2, v5.1.2, v5.1.3, v4.4.7 but it didn't worked.

Also i don't know if it's related but i'm putting it here just in case, in the IntelliSense configurations page, one of my include path isn't found even though it's not in the include path section of the c_cpp_properties.json. 
[image]

### Sketch

```cpp
'''
/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include "Arduino.h"
extern "C"
{
// #include <stdio.h>
// #include <inttypes.h>
// #include "sdkconfig.h"
// #include "freertos/FreeRTOS.h"
// #include "freertos/task.h"
// #include "esp_chip_info.h"
// #include "esp_flash.h"
// #include "esp_system.h"
}

extern "C"
{
    void app_main(void);
}

void app_main(void)
{
    printf("Hello world!\n");

   
}
'''
```

### Debug Message

```plain
Here is the message i got after trying to build the code

'''
-- Found Git: C:/Users/victor.stella/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/victor.stella/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/victor.stella/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/victor.stella/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
NOTICE: Skipping optional dependency: espressif/esp-sr
IDF version changed from 5.1.2 to 5.2.2, solving dependencies.
..............................................................HINT: Please check manifest file of the following component(s): arduino
CMake Error at C:/Users/victor.stella/esp/v5.2/esp-idf/tools/cmake/build.cmake:552 (message):
  WARNING: Component "idf" not found

  ERROR: Because project depends on idf (>=5.1,<5.2) which doesn't match any
  versions, version solving failed.

Call Stack (most recent call first):
  C:/Users/victor.stella/esp/v5.2/esp-idf/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/victor.stella/Documents/HUPI/project-debug-tuto/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "c:\Users\victor.stella\.espressif\tools\cmake\3.24.0\bin\cmake.exe '-G=Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B=c:\Users\victor.stella\Documents\HUPI\project-debug-tuto\build', '-S=c:\Users\victor.stella\Documents\HUPI\project-debug-tuto'" terminated with exit code: 1. 
'''
```

### Other Steps to Reproduce

-tried to build the program on several version of ESP-IDF(v4.4.7, v5.1.2, v5.1.3, v5.2.2) non of them worked
-tried to add the includePath suggested by the quick fix (see next screenshot) but didn't worked
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-10:
`IDF version changed from 5.1.2 to 5.2.2, solving dependencies.`
`ERROR: Because project depends on idf (>=5.1,<5.2) which doesn't match any versions, version solving failed.`

You need ESP-IDF v5.1.4

Victor-stll, 2024-07-10:
I installed it and it works for the basic hello world code, but back in my original code it doesn't work anymore. i did run the exact same way and i got this in result, any idea ? i tried to include <gpio_types.h> or <esp32_mock.h> as suggested but i didn't solved the problem

[image]
[image]

-- Found Git: C:/Users/victor.stella/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/victor.stella/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/victor.stella/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/victor.stella/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
NOTICE: Skipping optional dependency: espressif/esp-sr
Processing 23 dependencies:
[1/23] chmorgan/esp-libhelix-mp3 (1.0.3)
[2/23] espressif/arduino-esp32 (3.0.2)
[3/23] espressif/cbor (0.6.0~1)
[4/23] espressif/esp-dsp (1.4.12)
[5/23] espressif/esp-modbus (1.0.15)
[6/23] espressif/esp-zboss-lib (1.4.0)
[7/23] espressif/esp-zigbee-lib (1.4.0)
[8/23] espressif/esp_diag_data_store (1.0.1)
[9/23] espressif/esp_diagnostics (1.1.0)
[10/23] espressif/esp_insights (1.1.0)
[11/23] espressif/esp_modem (1.1.0)
[12/23] espressif/esp_rainmaker (1.2.0)
[13/23] espressif/esp_schedule (1.2.0)
[14/23] espressif/esp_secure_cert_mgr (2.4.1)
[15/23] espressif/jsmn (1.1.0)
[16/23] espressif/json_generator (1.1.2)
[17/23] espressif/json_parser (1.0.3)
[18/23] espressif/libsodium (1.0.20~1)
[19/23] espressif/mdns (1.3.2)
[20/23] espressif/qrcode (0.1.0~2)
[21/23] espressif/rmaker_common (1.4.6)
[22/23] idf (5.1.4)
[23/23] joltwallet/littlefs (1.14.8)
-- DEBUG: Use esp-modbus component folder: C:/Users/victor.stella/Documents/HUPI/project-debug-tuto/managed_components/espressif__esp-modbus.
NOTICE: Skipping optional dependency: espressif/esp-sr
CMake Error at C:/Users/victor.stella/esp/v5.1.4/esp-idf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'arduino'.
Call Stack (most recent call first):
  C:/Users/victor.stella/esp/v5.1.4/esp-idf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  C:/Users/victor.stella/esp/v5.1.4/esp-idf/tools/cmake/build.cmake:595 (__build_expand_requirements)
  C:/Users/victor.stella/esp/v5.1.4/esp-idf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/victor.stella/Documents/HUPI/project-debug-tuto/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "c:\Users\victor.stella\.espressif\tools\cmake\3.24.0\bin\cmake.exe '-G=Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B=c:\Users\victor.stella\Documents\HUPI\project-debug-tuto\build', '-S=c:\Users\victor.stella\Documents\HUPI\project-debug-tuto'" terminated with exit code: 1.

igrr, 2024-07-10:
> Failed to resolve component 'arduino'.

Most likely one of your components has `REQUIRES arduino` (or `PRIV_REQUIRES`) in its CMakeLists.txt.

After you have installed Arduino from the component registry, the component name is going to be `espressif__arduino-esp32`.

Victor-stll, 2024-07-10:
thank you for your help, it seemed to work, but it revealed an other issue
some of the GPIO's i was using wern't found (such as GPIO_NUM_46 or 41) but it was just because the espressif target was esp32 instead of esp32s3. so i tried to change it  but each time i select the esp32s3 target, the following error messag appear :

[Error: non zero exit code 2 Directory 'C:\Users\victor.stella\Documents\HUPI\project-debug-tuto\build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it. Adding "set-target"'s dependency "fullclean" to list of commands with default set of options. Executing action: fullclean at ChildProcess.<anonymous> (\dist\extension.js:2:1461587) at ChildProcess.emit (node:events:514:28) at ChildProcess.emit (node:domain:488:12) at ChildProcess._handle.onexit (node:internal/child_process:294:12)]

i tried to manually delete the build directory, the managed component directory, or sdkconfig whatever nothing worked

-
-
-
-

so i tried to create a new program to set the good target before doing anything, then paste my whole code. got a different error :

Executing task: c:\Users\victor.stella\.espressif\tools\ninja\1.10.2\ninja.exe  

[0/1] Re-running CMake...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3        
Processing 24 dependencies:
[1/24] chmorgan/esp-libhelix-mp3 (1.0.3)
[2/24] espressif/arduino-esp32 (3.0.2)
[3/24] espressif/cbor (0.6.0~1)
[4/24] espressif/esp-dsp (1.4.12)
[5/24] espressif/esp-modbus (1.0.15)
[6/24] espressif/esp-sr (1.7.1)
[7/24] espressif/esp-zboss-lib (1.4.0)
[8/24] espressif/esp-zigbee-lib (1.4.0)
[9/24] espressif/esp_diag_data_store (1.0.1)
[10/24] espressif/esp_diagnostics (1.1.0)
[11/24] espressif/esp_insights (1.1.0)
[12/24] espressif/esp_modem (1.1.0)
[13/24] espressif/esp_rainmaker (1.2.0)
[14/24] espressif/esp_schedule (1.2.0)
[15/24] espressif/esp_secure_cert_mgr (2.4.1)
[16/24] espressif/jsmn (1.1.0)
[17/24] espressif/json_generator (1.1.2)
[18/24] espressif/json_parser (1.0.3) 
[19/24] espressif/libsodium (1.0.20~1)
[20/24] espressif/mdns (1.3.2)
[21/24] espressif/qrcode (0.1.0~2)
[22/24] espressif/rmaker_common (1.4.6)
[23/24] idf (5.1.4)
[24/24] joltwallet/littlefs (1.14.8)
-- DEBUG: Use esp-modbus component folder: C:/Users/victor.stella/Documents/HUPI/project-name-anothertry/managed_components/espressif__esp-modbus.
-- Project sdkconfig file C:/Users/victor.stella/Documents/HUPI/project-name-anothertry/sdkconfig
Compiler supported targets: xtensa-esp32s3-elf

-- App "project-name-anothertry" version: 1
-- Adding linker script C:/Users/victor.stella/Documents/HUPI/project-name-anothertry/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/victor.stella/esp/v5.1.4/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Users/victor.stella/esp/v5.1.4/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Users/victor.stella/esp/v5.1.4/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Users/victor.stella/esp/v5.1.4/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Users/victor.stella/esp/v5.1.4/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Users/victor.stella/esp/v5.1.4/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Users/victor.stella/esp/v5.1.4/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- DEBUG: Use esp-modbus component folder: C:/Users/victor.stella/Documents/HUPI/project-name-anothertry/managed_components/espressif__esp-modbus.
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
ESP Insights Project commit: HEAD-HASH-NOTFOUND
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
ESP RainMaker Project commit: HEAD-HASH-NOTFOUND
CMake Error at managed_components/espressif__arduino-esp32/CMakeLists.txt:303 (message):
  esp32-arduino requires CONFIG_FREERTOS_HZ=1000 (currently 100)

-- Configuring incomplete, errors occurred!
See also "C:/Users/victor.stella/Documents/HUPI/project-name-anothertry/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
C:\Users\victor.stella\.espressif\tools\cmake\3.24.0\bin\cmake.exe --regenerate-during-build -SC:\Users\victor.stella\Documents\HUPI\project-name-anothertry -BC:\Users\victor.stella\Documents\HUPI\project-name-anothertry\build
[... 50 lines cut ...]
  I.E.  for "espressif__mdns" run:

  mv
  C:\Users\victor.stella\Documents\HUPI\project-name-anothertry\managed_components\espressif__mdns
  C:\Users\victor.stella\Documents\HUPI\project-name-anothertry\components\espressif__mdns

  Or, if you want to discard the changes remove the ".component_hash" file
  from the component's directory.

  I.E.  for "espressif__mdns" run:

  rm
  C:\Users\victor.stella\Documents\HUPI\project-name-anothertry\managed_components\espressif__mdns\.component_hash

Call Stack (most recent call first):
  C:/Users/victor.stella/esp/v5.1.4/esp-idf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/victor.stella/Documents/HUPI/project-name-anothertry/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
C:\Users\victor.stella\.espressif\tools\cmake\3.24.0\bin\cmake.exe --regenerate-during-build -SC:\Users\victor.stella\Documents\HUPI\project-name-anothertry -BC:\Users\victor.stella\Documents\HUPI\project-name-anothertry\build
ninja: error: rebuilding 'build.ninja': subcommand failed

 *  The terminal process "c:\Users\victor.stella\.espressif\tools\ninja\1.10.2\ninja.exe" terminated with exit code: 1.