espressif/arduino-esp32 #10006: esp 32 not connecting to Mac M2

damiea, 2024-07-10:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module

### Hardware Configuration

ESP32

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS Sonoma

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

-

### Description

Anyone can help i have downloaded all the driver that are available but still it doesnt detect any new port especially the UART.

only this is shown /dev/cu.Bluetooth-Incoming-Port when i prompt  ls /dev/cu.*

### Sketch

```cpp
none
```

### Debug Message

```plain
Sketch uses 267981 bytes (20%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20184 bytes (6%) of dynamic memory, leaving 307496 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.6
Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-07-10:
Driver for the USB serial chip installed?

damiea, 2024-07-10:
i have tried downloading all of this:

- CH34xVCPDriver
- Mac_OSX_VCP_Driver
- SiLabsUSBDriverDIsk
- CH341SER_MAC

and the usb is connected via USB-hub

Jason2866, 2024-07-10:
driver can conflict with each other. The USB hub can do harm too.

damiea, 2024-07-10:
> driver can conflict with each other. The USB hub can do harm too.

what should i do

Jason2866, 2024-07-10:
Check which USB serial chip your board uses. Deinstall all drivers. This can be tricky.
Install only the driver which is the correct one for your dev board. Do not use a USB Hub.

Jason2866, 2024-07-12:
@robertlipe My M1 Mac does work with all dev boards I own. There are some since I am maintain the customized Tasmota Arduino Platformio Framework and do "some" testing ;-)

damiea, 2024-07-12:
where can i find the chip? is it on the esp32 board itself? im still new in developing IOT projects so im not sure how to fix the problem.

Jason2866, 2024-07-12:
> where can i find the chip? is it on the esp32 board itself?

Yes, on the board itself. You can post an photo here.

This one ` /dev/cu.wchWHATEVER` tooks me a while to get fixed. The issue was to get the old driver fully deinstalled. After having ONLY the needed actual driver installed ALL boards works well to flash

damiea, 2024-07-12:
[image]

> Yes, on the board itself. You can post an photo here.
>
> This one ` /dev/cu.wchWHATEVER` tooks me a while to get fixed. The issue was to get the old driver fully deinstalled. After having ONLY the [...]

damiea, 2024-07-12:
its the CP2102

damiea, 2024-07-12:
do i need to use the direct cable from usb-c  to the micro-usb or can i plug it on a USB Hub.

lbernstone, 2024-07-12:
For troubleshooting, make a direct connection from the computer to the esp32.  Once you know it is working, you can test through the hub, using the other ports, etc.

leezicai, 2024-12-25:
Someone said that the Thunderbolt interface doesn’t support direct connections. Have you tried using a USB hub to adapt it? That might work.

damieaamri, 2024-12-26:
i tried both direct connection and USB HUb. but still nothing happen. i think the board is broken already.