espressif/arduino-esp32 #10008: TCB data getting semi-randomly corrupted

iniw, 2024-07-10:
### Board

Custom board, powered by ESP32-D0WD (revision v1.1)

### Device Description

DLC32 board by MakerBase

### Hardware Configuration

Motors, sensors, etc. (Kind of a CNC machine)

### Version

v2.0.16

### IDE Name

PlatformIO

### Operating System

MacOS

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have two tasks that looks something like this (pseudocode):

```c++
/// task 1
while (true) {
  log_stack(); // this gathers the task name and stack watermark, calls `pcTaskGetTaskName` and `uxTaskGetStackHighWaterMark`
  auto event = receive_event_from_task_two();
  process_event(event);
  send_reply_to_task_two();
}

/// task 2
while (not_finished()) {
  log_stack();
  send_event_to_task_one(event);
  auto reply = wait_for_reply_from_task_one();
  process_reply(reply);
}
```

After a while (something like 30-40 iterations) the TCB of task 1 becomes corrupted, it's stack somehow ends up filled with garbage data and the call to `log_stack()` eventually panics, producing a stacktrace like this:

[image]

Some relevant information:

- `xPortGetFreeHeapSize` reports a stable ~8Kb of available heap space, this number does not fluctuate at any point.

- There are no heap allocations being performed anywhere and there is plenty of stack space left on all task in the system before the corruption happens.

- The two tasks that interact with each other (through FreeRTOS queues) are statically allocated, pinned to core 0 (with `xTaskCreateStaticPinnedToCore`) and have plenty of stack space left.

- The two event queues in question are also statically allocated and so are the elements being passed around between them.

- Here's what `log_stack()` from the pseudocode produces before and after the corruption (fomatted like `printf("[%s] -> %u\n", pcTaskName, usStackHighWaterMark`) (POUR = task 1):

[image]

[image]

- [This](https://esp32.com/viewtopic.php?f=13&t=38559) forum post looks very simillar to the issue i'm having (even the register addresses) but it unfortunately got almost no attention and the OP didn't follow it up with a solution (if there even is one). I have also posted a reply there.

ESP-IDF mirror of this issue: https://github.com/espressif/esp-idf/issues/14161

### Sketch

```cpp
(Shown above)
```

### Debug Message

```plain
(Shown above)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-11:
IDF 4.4 is EOL, so I doubt anyone there will help. There are ways to run newer Arduino on IDF 5.1 and PIO. Why not try that? Also 8KB heap on device that has 300+KB is a bit low and you also did not mention what is the stack size of the threads.

Jason2866, 2024-07-11:
In my experiences all kind of weird issues do happen with a free heap below 10kbytes.
I have a fork of Platformio Framework espressif32 running perfectly well with Arduino 3.0.2 as a component of IDF 5.1.

iniw, 2024-07-11:
> There are ways to run newer Arduino on IDF 5.1 and PIO. Why not try that? 
I am unfortunately stuck to a codebase that uses 4.4, not much I can do about that. 

> Also 8KB heap on device that has 300+KB is a bit low
The low heap space is mostly due to the SSP Bluedroid stack. My code consumes about ~45KB of ram as a single, statically allocated, big block, which contains all the tasks and queues needed for my personal application. Pretty much all the rest is consumed by Bluedroid.

> and you also did not mention what is the stack size of the threads.
Task 1 is 3072 bytes and Task 2 is 4096 bytes. Both have plenty of stack space left, here's what their `log_stack()` produces (`{task name} -> {stack highmark}`) right before the panic happens

```
09:13:20.922 > [TASK1] -> 568
09:13:20.923 > [TASK2] -> 964
09:13:31.424 > [TASK1] -> 568
09:13:31.424 > [TASK2] -> 964
09:13:41.941 > [TASK1] -> 568
09:13:41.941 > [TASK2] -> 964
09:13:52.457 > [TASK1] -> 568
09:13:52.457 > [TASK2] -> 964
```

iniw, 2024-07-11:
> In my experiences all kind of weird issues do happen with a free heap below 10kbytes.

Could you elaborate a bit on this? Why exactly is that a problem?

Jason2866, 2024-07-11:
Why, idk, just experience. Issues I have encountered: Wifi instable, disconnect and connect problems. mqtt server disconnects, random crashes

iniw, 2024-07-11:
Figured out the issue. The problem was that because all task's entire stack space and "personal information" are layed out sequentially in memory, in that big 45KB block I mentioned earlier, and out-of-bounds array access on Task 2 (which sat in memory right before Task 1) was poking into Task 2's memory region, causing the corruption.

closed 2024-07-11 by iniw