espressif/arduino-esp32 #10009: WiFi.status() always reports WL_STOPPED although connected

Nanowires, 2024-07-10:
### Board

ESP32-WROOM-32

### Device Description

-

### Hardware Configuration

-

### Version

v3.0.1

### IDE Name

Arduino IDE

### Operating System

Fedora 39

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Problem:
When connecting to a Wifi, the ESP already receives an IP adress, but WiFi.status() still reports an WL_STOPPED (254) status.

### Sketch

```cpp
#include <WiFi.h>

#define HOME_SSID                       "SSID"
#define PASSWORD                        "PASSWORD"
void setup()
{
// Add your initialization code here

    Serial.begin(115200);
    delay(10);
    WiFi.begin(HOME_SSID, PASSWORD);

    while (WiFi.waitForConnectResult(500) != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
        Serial.println(WiFi.status());
        Serial.println(WiFi.localIP());
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

// The loop function is called in an endless loop
void loop()
{
}
```

### Debug Message

```plain
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q6
  Revision          : 100
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   323328 B ( 315.8 KB)
  Free Bytes        :   289924 B ( 283.1 KB)
  Allocated Bytes   :    26148 B (  25.5 KB)
  Minimum Free Bytes:   283960 B ( 277.3 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jul 10 2024 23:36:58
  Compile Host OS   : linux
  ESP-IDF Version   : v5.1.4-358-gbd2b9390ef-dirty
  Arduino Version   : 3.0.2
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_WROOM_DA
  Arduino Variant   : esp32da
  Arduino FQBN      : esp32:esp32:esp32da:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=verbose,LoopCore=1,EventsCore=1,EraseFlash=none
============ Before Setup End ============
[   755][V][esp32-hal-uart.c:408] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   764][V][esp32-hal-uart.c:497] uartBegin(): UART0 not installed. Starting installation
[   774][V][esp32-hal-uart.c:544] uartBegin(): UART0 initialization done.
[   819][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 9 - WIFI_READY
[   896][I][WiFiGeneric.cpp:726] setDualAntennaConfig(): TX Antenna will be automatically selected
[   897][V][STA.cpp:184] _onStaEvent(): STA Started
[   909][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 11 - STA_START
[   917][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[   975][V][STA.cpp:204] _onStaEvent(): STA Connected: SSID: MartinRouter, BSSID: e8:df:70:0a:bb:49, Channel: 1, Auth: WPA2_PSK
[   986][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 13 - STA_CONNECTED
[   994][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 13 - STA_CONNECTED
[  1010][V][NetworkInterface.cpp:66] _onIpEvent(): sta Got New IP: <REDACTED IP> MASK: 255.255.255.0 GW: <REDACTED IP>
[  1021][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 16 - STA_GOT_IP
[  1029][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 16 - STA_GOT_IP
[  1036][V][STA.cpp:169] _onStaArduinoEvent(): STA IP: <REDACTED IP>, MASK: 255.255.255.0, GW: <REDACTED IP>
.254
<REDACTED: some valid IP>
.254
<REDACTED: some valid IP>
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-10:
Always is a long time.  Do you mean the waitForConnectResult never finishes, or that it is reporting WL_STOPPED for a couple loops after see these events which show the connection?

Nanowires, 2024-07-11:
Thanks for the response.

Yes I know that waitForConnectResult() is blocking, I just wanted to ensure, that waitForConnectResult() and status() are consistent.

To be honest: I don't know exactly what waitForConnectResult is returning, just that it doesn't return WL_CONNECTED. But I guess, that it's also returning the WL_STOPPED like the status() call.
And yes, the waitForConnectResult() never succeeds (at least for the next 10 minutes, after the ESP already received an IP address).

If ti would have succeeded the logs would show something like

```

WiFi connected
IP address: 
x.x.x.x
```
But these logs never show up. I would have to break the loop manually, after checking that localIP() is returning something different than 0.0.0.0, but that is not how to check if the connection is established.
Also when flashing Arduino version 2.0.17, it's working as expected. I.e. waitForConnectResult() is returning WL_CONNECTED after a short duration.

me-no-dev, 2024-07-11:
This is not how `waitForConnectResult()` is used. I'll check to make sure it works fine, but your `while` should check the status instead.

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino#L48-L58

me-no-dev, 2024-07-11:
BTW for me the test sketch you provided works just fine. Try 3.0.2

Nanowires, 2024-07-11:
Thanks for the feedback and the testing.
Indeed the problem lies of course on my side.

I had still included some of my libraries, which shouldn't interfere during the setup() function, but magically they do...
Need to investigate further, but this issue can be closed.

closed 2024-07-11 by Nanowires

Nanowires, 2024-07-11:
Found the issue, for the interested reader:

I've written an abstraction library, which defined it's own object of the WiFiClass.
After changing this to apointer to point to the already existent WiFi object, it's working again, like in 2.0.17