espressif/arduino-esp32 #10010: ledcWrite() has no option to write a LED-duty to a channel, but only to a specific pin

purehunter1, 2024-07-11:
### Related area

Arduino-ESP32-Core 3.0.x

### Hardware specification

ESP32 (all types)

### Is your feature request related to a problem?

Yes, there is a problem, migrating a program written for Arduino-ESP32-Core 2.x
The "old" Arduino-ESP32-Core 2.x provided a command "ledcWrite(LEDchan, ledBrightness);" which allowed to get a synchronized LED output duty signal on all pins attached to a specific channel.
The new 3.0.x core does not provide a similar write command to control a channel.

### Describe the solution you'd like

I would like to have (again) a write command similar to the command in the 2.0.x core, or in the ESP-IDF: ledc_set_duty(ledc_mode_t speed_mode, ledc_channel_t channel, uint32_t duty).

### Describe alternatives you've considered

Write a duty value to one of the pins attached to a specific channel would change the output of all pins attached to that channel.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2024-07-11:
@P-R-O-C-H-Y can you please help with triage?

purehunter1, 2024-07-15:
@P-R-O-C-H-Y yes, it is also allowed in the Arduino environment to add pins to a channel, but there is no possibility to control the channel...