espressif/arduino-esp32 #10017: 4-mic setup resulting in static and quiet audio

nroede, 2024-07-12:
### Board

ESP32 S3 Dev Kit

### Device Description

Plain module on breadboard

### Hardware Configuration

Two pairs of I2S ICS43434 microphones are attached to the board, using GPIO 21 as SCK, 47 as WS, 17 as SD for the first pair, and 16 as SD for the second pair. An SD card is also connected to the board, using GPIO 15 as SD_CMD, 14 as SD_CLK, and 2 as SD_DATA. The microphones are powered by an external 5V power supply, and the SD card is powered by the board's 3V3 pin.

### Version

v3.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have a sketch that initializes the SD card, then initializes both pairs of I2S microphones. It then opens and begins writing to two wav files simultaneously, one for each mic pair, for 10 seconds. The headers are updated and the files closed. This audio can be heard by reading the SD card on a PC and dragging the wavs into Audacity or another similar audio program. It is expected that the wav files would both be 10 seconds, have relatively clear audio, and have approximately the same volume level (since the microphones are all next to each other). However, mic pair 2 is always much too quiet, and about 30% of the time mic pair 1 has significant static noise as well. Even when the data pins are switched, it's always mic pair 1 (so it isn't a hardware issue with the mics). Initializing the mics in the opposite order switches the issues so that mic pair 1 is quiet and mic pair 2 has static.

### Sketch

```cpp
#include "ESP_I2S.h"
#include "FS.h"
#include "SD_MMC.h"

#define ONE_BIT_MODE true

// Define common I2S pins for both sets of microphones
const uint8_t I2S_SCK = 21;
const uint8_t I2S_WS = 47;

// Define I2S pins for first set of microphones
const uint8_t I2S_DIN1 = 17;

// Define I2S pins for second set of microphones
const uint8_t I2S_DIN2 = 16;

// Define SD card pins
const uint8_t SD_CMD = 15;
const uint8_t SD_CLK = 14;
const uint8_t SD_DATA0 = 2;

#define SAMPLE_RATE 16000
#define BITS_PER_SAMPLE I2S_DATA_BIT_WIDTH_32BIT
#define CHANNELS I2S_SLOT_MODE_STEREO
#define BUFFER_SIZE 512  // Reduced buffer size to save stack space
#define RECORD_DURATION 10  // Record duration in seconds

[... 267 lines cut ...]
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     2 : SDMMC_D0
     4 : GPIO
    12 : GPIO
    13 : GPIO
    14 : SDMMC_CLK
    15 : SDMMC_CMD
    16 : I2S_STD_DIN
    17 : I2S_STD_DIN
    19 : USB_DM
    20 : USB_DP
    21 : I2S_STD_BCLK
    43 : UART_TX[0]
    44 : UART_RX[0]
    47 : I2S_STD_WS
============ After Setup End =============
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

nroede, 2024-07-12:
@me-no-dev @P-R-O-C-H-Y

lbernstone, 2024-07-12:
If you have just a single mic on i2s2 (whichever one seems too quiet), does it have the issue, or only when there are 2 mics?

me-no-dev, 2024-07-15:
you can not use the same pins for WS and SCK for two different I2S peripherals. You need separate pins for that

nroede, 2024-07-16:
@me-no-dev Thank you! This was the issue we were having and solved our problem!

closed 2024-07-16 by nroede