espressif/arduino-esp32 #10020: Segment clash error during linking

memecode, 2024-07-12:
### Board

ESP32-S2

### Device Description

I'm trying to build [this code](http://memecode.com/hw/teleproject/esp32s2/nrf_test.zip) and I'm getting [this error log.](http://memecode.com/hw/teleproject/esp32s2/error.log)

I don't even know where to start in addressing this. But it seems like a bug in the linker script.

### Hardware Configuration

n/a as it's a compile/link issue. But I have a nRF24L01 board attached to the SPI pins with gpio17 for control pin and gpio18 for the SPI chip select.

### Version

v3.0.2

### IDE Name

Arduino IDE v2.3.2

### Operating System

Windows 10

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm expecting that the compile / link just works and produces a binary.

### Sketch

```cpp
http://memecode.com/hw/teleproject/esp32s2/nrf_test.zip
```

### Debug Message

```plain
http://memecode.com/hw/teleproject/esp32s2/error.log
```

### Other Steps to Reproduce

If I back out the debug printing code it works again but that doesn't help me fix my issues on the device.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-07-13:
PSRAM do use GPIO 16 and 17. So you can't use for anything else to connect.

me-no-dev, 2024-07-15:
The problem is in the following lines inside `RF24_config.h`:
```cpp
// Avoid spurious warnings
#if 1
#if ! defined( NATIVE ) && defined( ARDUINO )
#undef PROGMEM
#define PROGMEM __attribute__(( section(".progmem.data") ))
#undef PSTR
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
#endif
#endif
```

Remove them and it will compile

closed 2024-07-15 by me-no-dev

memecode, 2024-07-15:
Amazing. Thank you @me-no-dev