espressif/arduino-esp32 #10022: Arduino_GFX_Library not recognized

Gartosz, 2024-07-13:
### Board

ESP32S3 Dev Module

### Device Description

ESP32S3 board with touchscreen as a premade device (ESP328048070C), but it doesn't matter as issue occurs during compilation.

### Hardware Configuration

NA

### Version

v3.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

In v2.0.17 my sketches are working fine, with [Arduino_GFX_Library](https://github.com/moononournation/Arduino_GFX), which is installed from Arduino IDE library explorer. However when I go to v.3.0.0 and higher it just doesn't recognise that lib, but doesn't raise any import errors.

### Sketch

```cpp
NA
```

### Debug Message

```plain
error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?
error: 'Arduino_RGB_Display' has not been declared
```

### Other Steps to Reproduce

I've tried all v3 releases as well as reinstaling the library and restarting. Not sure what else I could do.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Gartosz, 2024-07-13:
I've made a basic test file and it's having the same issues.

> error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?
   10 | Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
      | ^~~~~~~~~~~~~~~~~~~~~
      | Arduino_ESP32S2PAR8Q

> error: 'Arduino_RGB_Display' does not name a type
   17 | Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
      | ^~~~~~~~~~~~~~~~~~~

> exit status 1
Compilation error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?

```
#include <Arduino_GFX_Library.h>

#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin
#define TFT_BL 2
/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

#endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

lbernstone, 2024-07-13:
This is an issue with that library.  AFAICT it has been changed to use the native display panel methods in IDF 5.1.  You should be able to restore the old methods by removing [this line](https://github.com/moononournation/Arduino_GFX/blob/master/src/databus/Arduino_ESP32RGBPanel.h#L4)

Gartosz, 2024-07-13:
Oh, okay. Have you got any idea why would the author prevent using v3 of ESP32 board manager? Were there any similar cases with other libraries? I've tried altering it and it's not compiling (one of the many errors - error: ISO C++ forbids declaration of 'LIST_HEAD' with no type [-fpermissive]). Looks like many things aren't compatible, so I will reach out to [moononournation](https://github.com/moononournation). Can you tell me when or if there will be Overview of major changes in 3.0.0 Release posted? It's been empty for the 1,5 month.

lbernstone, 2024-07-13:
There are major changes between the APIs in v2.X and v3.X.  The authors of libraries are responsible to make them work with new versions.  Framework development happens from the lowest levels upward.
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

closed 2024-07-14 by Gartosz