espressif/arduino-esp32 #10033: SD MMC does not initialize when powered by external battery but works fine while powered by USB through PC.

raprakashvi, 2024-07-15:
### Board

ESP32 S3

### Device Description

ESP32 S3 Wroom-1 N16R8 dev kit. 

### Hardware Configuration

// Define common I2S pins for both sets of microphones
const uint8_t I2S_SCK1 = 21;
const uint8_t I2S_WS1 = 47;

const uint8_t I2S_SCK2 = 5;
const uint8_t I2S_WS2 = 6;

// Define I2S pins for first set of microphones
const uint8_t I2S_DIN1 = 17;

// Define I2S pins for second set of microphones
const uint8_t I2S_DIN2 = 16;

// Define SD card pins
const uint8_t SD_CMD = 40;
const uint8_t SD_CLK = 14;
const uint8_t SD_DATA0 = 2;

// Define LED pin
const uint8_t ledPin = 10;

### Version

v3.0.2

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

For our experiment we want to power our ESP32 with a battery. The setup works fine when we use ESP32 Wroom boards both when connected to PC and when powered by battery but we cannot recreate the same behavior when using ESP32 S3 Wroom-1 N16R8.

Our code works and we can  write data to SD card when connected to the PC using USB port but when we power it by battery, the SD card does not initialize. We are using ESP32 S3 Devkit 1.1 with 16MB Flash an d8 MB PSRAM. We are powering the ESP by an external power supply set at 3.3 V or by a battery source either connected to 5V VIN or to 3.3v pin after using a LDO. We have added an LED blink after SD card initialization step to make sure we know it is working when powered by the battery. It is not working. 

Is it something to do with strapping pins? Is there a difference in behaviour if the board is powered by the USB port vs when powered by battery?

### Sketch

```cpp
#include "ESP_I2S.h"
#include "FS.h"
#include "SD_MMC.h"

#define ONE_BIT_MODE true

// Define common I2S pins for both sets of microphones
const uint8_t I2S_SCK1 = 21;
const uint8_t I2S_WS1 = 47;

const uint8_t I2S_SCK2 = 5;
const uint8_t I2S_WS2 = 6;

// Define I2S pins for first set of microphones
const uint8_t I2S_DIN1 = 17;

// Define I2S pins for second set of microphones
const uint8_t I2S_DIN2 = 16;

// Define SD card pins
const uint8_t SD_CMD = 40;
const uint8_t SD_CLK = 14;
const uint8_t SD_DATA0 = 2;

// Define LED pin
const uint8_t ledPin = 10;

#define SAMPLE_RATE 8000
#define BITS_PER_SAMPLE I2S_DATA_BIT_WIDTH_32BIT
#define CHANNELS I2S_SLOT_MODE_STEREO
#define BUFFER_SIZE 512  // Reduced buffer size to save stack space
#define RECORD_DURATION 10  // Record duration in seconds

I2SClass i2s1;
I2SClass i2s2;
const char *audioFileName1 = "/audio_mics1_240630.wav";
const char *audioFileName2 = "/audio_mics2_240630.wav";

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    delay(10);
  }

  pinMode(15, OUTPUT);
  digitalWrite(15, HIGH);

  delay(50);

  pinMode(ledPin, OUTPUT);
  digitalWrite(ledPin, LOW);

  // Initialize SD card
  pinMode(2, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  pinMode(13, INPUT_PULLUP);
  pinMode(15, INPUT_PULLUP);

  Serial.println("Initializing SD card...");
  if (!SD_MMC.setPins(SD_CLK, SD_CMD, SD_DATA0)) {
    Serial.println("Failed to set SD pins!");
    return;
  }

  if (!SD_MMC.begin("/sdcard", ONE_BIT_MODE)) {
    Serial.println("SD Card initialization failed!");
    return;
  }

  Serial.println("SD card initialized.");

  digitalWrite(ledPin, HIGH);

  // Create or open the files for writing audio data
  File audioFile1 = SD_MMC.open(audioFileName1, FILE_WRITE);
  if (!audioFile1) {
    Serial.println("Failed to open file 1 for writing");
    return;
  }

  File audioFile2 = SD_MMC.open(audioFileName2, FILE_WRITE);
  if (!audioFile2) {
    Serial.println("Failed to open file 2 for writing");
    return;
  }

  // Initialize I2S for first set of microphones
  Serial.println("Initializing I2S bus for first set of mics...");
  i2s1.setPins(I2S_SCK1, I2S_WS1, -1, I2S_DIN1);
  if (!i2s1.begin(I2S_MODE_STD, SAMPLE_RATE, BITS_PER_SAMPLE, CHANNELS)) {
    Serial.println("Failed to initialize I2S for first set of mics");
    return;
  }

  // Initialize I2S for second set of microphones
  Serial.println("Initializing I2S bus for second set of mics...");
  i2s2.setPins(I2S_SCK2, I2S_WS2, -1, I2S_DIN2);
  if (!i2s2.begin(I2S_MODE_STD, SAMPLE_RATE, BITS_PER_SAMPLE, CHANNELS)) {
    Serial.println("Failed to initialize I2S for second set of mics");
    return;
  }

  // Configure I2S RX transformation for both sets of microphones
  if (!i2s1.configureRX(SAMPLE_RATE, BITS_PER_SAMPLE, CHANNELS, I2S_RX_TRANSFORM_32_TO_16)) {
    Serial.println("Failed to configure I2S RX transformation for first set of mics!");
    return;
  }
  
  if (!i2s2.configureRX(SAMPLE_RATE, BITS_PER_SAMPLE, CHANNELS, I2S_RX_TRANSFORM_32_TO_16)) {
    Serial.println("Failed to configure I2S RX transformation for second set of mics!");
    return;
  }

  Serial.println("I2S bus initialized for both sets of mics.");

  // Write WAV headers to files
  writeWavHeader(audioFile1, SAMPLE_RATE, 16, 2);
  writeWavHeader(audioFile2, SAMPLE_RATE, 16, 2);

  // Buffers for audio data
  static char buffer1[BUFFER_SIZE];
  static char buffer2[BUFFER_SIZE];
  char *currentBuffer1 = buffer1;
  char *writeBuffer1 = buffer2;
  static char buffer3[BUFFER_SIZE];
  static char buffer4[BUFFER_SIZE];
  char *currentBuffer2 = buffer3;
  char *writeBuffer2 = buffer4;
  bool useBuffer1 = true;

  // Record audio for the specified duration
  Serial.println("Recording audio...");
  unsigned long startTime = millis();
  while ((millis() - startTime) < (RECORD_DURATION * 1000)) {
    size_t bytesRead1 = i2s1.readBytes(currentBuffer1, BUFFER_SIZE);
    size_t bytesRead2 = i2s2.readBytes(currentBuffer2, BUFFER_SIZE);
    if (bytesRead1 > 0 && bytesRead2 > 0) {
      // Swap buffers
      if (useBuffer1) {
        currentBuffer1 = buffer2;
        writeBuffer1 = buffer1;
        currentBuffer2 = buffer4;
        writeBuffer2 = buffer3;
      } else {
        currentBuffer1 = buffer1;
        writeBuffer1 = buffer2;
        currentBuffer2 = buffer3;
        writeBuffer2 = buffer4;
      }
      useBuffer1 = !useBuffer1;

      // Write the buffers to the SD card
      audioFile1.write((uint8_t *)writeBuffer1, bytesRead1);
      audioFile2.write((uint8_t *)writeBuffer2, bytesRead2);
    } else {
      Serial.println("Error reading from I2S");
    }
  }

  // Update WAV headers with final sizes
  updateWavHeader(audioFile1);
  updateWavHeader(audioFile2);

  // Close the files
  audioFile1.close();
  audioFile2.close();

  Serial.println("Recording stopped.");
}

void loop() {
  // Main loop is handled in the setup for continuous recording
}

void writeWavHeader(File file, uint32_t sampleRate, uint16_t bitsPerSample, uint16_t channels) {
  uint32_t fileSize = 36;       // Placeholder for file size
  uint32_t dataChunkSize = 0;  // Placeholder for data chunk size

  // RIFF chunk
  file.write((const uint8_t *)"RIFF", 4);
  file.write((const uint8_t *)&fileSize, 4);  // Placeholder for file size
  file.write((const uint8_t *)"WAVE", 4);

  // fmt subchunk
  file.write((const uint8_t *)"fmt ", 4);
  uint32_t subchunk1Size = 16;
  file.write((const uint8_t *)&subchunk1Size, 4);
  uint16_t audioFormat = 1;
  file.write((const uint8_t *)&audioFormat, 2);
  file.write((const uint8_t *)&channels, 2);
  file.write((const uint8_t *)&sampleRate, 4);
  uint32_t byteRate = sampleRate * channels * (bitsPerSample / 8);
  file.write((const uint8_t *)&byteRate, 4);
  uint16_t blockAlign = channels * (bitsPerSample / 8);
  file.write((const uint8_t *)&blockAlign, 2);
  file.write((const uint8_t *)&bitsPerSample, 2);

  // data subchunk
  file.write((const uint8_t *)"data", 4);
  file.write((const uint8_t *)&dataChunkSize, 4);  // Placeholder for data chunk size
}

void updateWavHeader(File file) {
  // Get the total file size and data chunk size
  uint32_t fileSize = file.size() - 8;
  uint32_t dataChunkSize = file.size() - 44;

  // Update file size in the header
  file.seek(4);
  file.write((const uint8_t *)&fileSize, 4);

  // Update data chunk size in the header
  file.seek(40);
  file.write((const uint8_t *)&dataChunkSize, 4);
}
```

### Debug Message

```plain
It works when connected to the PC so hard to tell.
```

### Other Steps to Reproduce

Powering the board by battery through 3.3v pin, VIN or even by USB port but powered by battery.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

raprakashvi, 2024-07-15:
@P-R-O-C-H-Y

lbernstone, 2024-07-15:
Uh, your power issue is likely a power issue.  How are you delivering power to the SD card?  It should have a power line directly from the 3v3 rail to VDD, with at least 1uF capacitor close to the power entrance. Put a multimeter on the 3v3 rail while the battery is connected and make sure it is right at 3.3. Batteries are generally able to deliver the necessary current, but may have voltage sags, so you need to make sure you have adequate capacitance on the power line.
Test with the [SDMMC_Test.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino)

raprakashvi, 2024-07-15:
Thanks for the quick feedback. The SD MMC code does work both when powered by battery and PC. This was without the capacitor and with internal resistors pulled up. Attached is the modified SD_MMC Test code which works, but still the previous setup code does not as SD card is not being initialized. We have added the 1 uF capacitor close to the VDD pins of the SD card which is directly connected to 3.3v pin of the board.

`/*
 * pin 1 - D2                |  Micro SD card     |
 * pin 2 - D3                |                   /
 * pin 3 - CMD               |                  |__
 * pin 4 - VDD (3.3V)        |                    |
 * pin 5 - CLK               | 8 7 6 5 4 3 2 1   /
 * pin 6 - VSS (GND)         | ▄ ▄ ▄ ▄ ▄ ▄ ▄ ▄  /
 * pin 7 - D0                | ▀ ▀ █ ▀ █ ▀ ▀ ▀ |
 * pin 8 - D1                |_________________|
 *                             ║ ║ ║ ║ ║ ║ ║ ║
 *                     ╔═══════╝ ║ ║ ║ ║ ║ ║ ╚═════════╗
 *                     ║         ║ ║ ║ ║ ║ ╚══════╗    ║
 *                     ║   ╔═════╝ ║ ║ ║ ╚═════╗  ║    ║
 * Connections for     ║   ║   ╔═══╩═║═║═══╗   ║  ║    ║
 * full-sized          ║   ║   ║   ╔═╝ ║   ║   ║  ║    ║
 * SD card             ║   ║   ║   ║   ║   ║   ║  ║    ║
 * ESP32-S3 DevKit  | 21  47  GND  39 3V3 GND  40 41  42  |
 * ESP32-S3-USB-OTG | 38  37  GND  36 3V3 GND  35 34  33  |
 * ESP32            |  4   2  GND  14 3V3 GND  15 13  12  |
 * Pin name         | D1  D0  VSS CLK VDD VSS CMD D3  D2  |
 * SD pin number    |  8   7   6   5   4   3   2   1   9 /
 *                  |                                  █/
 *                  |__▍___▊___█___█___█___█___█___█___/
 * WARNING: ALL data pins must be pulled up to 3.3V with an external 10k Ohm resistor!
 * Note to ESP32 pin 2 (D0): Add a 1K Ohm pull-up resistor to 3.3V after flashing
 *
 * SD Card | ESP32
 *    D2       12
 *    D3       13
 *    CMD      15
 *    VSS      GND
 *    VDD      3.3V
 *    CLK      14
 *    VSS      GND
 *    D0       2  (add 1K pull up after flashing)
 *    D1       4
 *
 *    For more info see file README.md in this library or on URL:
 *    https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC
 */

#include "FS.h"
#include "SD_MMC.h"

// Default pins for ESP-S3
// Warning: ESP32-S3-WROOM-2 is using most of the default GPIOs (33-37) to interface with on-board OPI flash.
//   If the SD_MMC is initialized with default pins it will result in rebooting loop - please
//   reassign the pins elsewhere using the mentioned command `setPins`.
// Note: ESP32-S3-WROOM-1 does not have GPIO 33 and 34 broken out.
// Note: if it's ok to use default pins, you do not need to call the setPins
int clk = 39;
int cmd = 40;
int d0 = 47;
int d1 = 21;
int d2 = 42;
int d3 = 41;  // GPIO 34 is not broken-out on ESP32-S3-DevKitC-1 v1.1

void listDir(fs::FS &fs, const char *dirname, uint8_t levels) {
  Serial.printf("Listing directory: %s\n", dirname);

  File root = fs.open(dirname);
  if (!root) {
    Serial.println("Failed to open directory");
    return;
  }
  if (!root.isDirectory()) {
    Serial.println("Not a directory");
    return;
[... 182 lines cut ...]
    Serial.println("UNKNOWN");
  }

  uint64_t cardSize = SD_MMC.cardSize() / (1024 * 1024);
  Serial.printf("SD_MMC Card Size: %lluMB\n", cardSize);

  listDir(SD_MMC, "/", 0);
  createDir(SD_MMC, "/mydir");
  listDir(SD_MMC, "/", 0);
  removeDir(SD_MMC, "/mydir");
  listDir(SD_MMC, "/", 2);
  writeFile(SD_MMC, "/hello.txt", "Hello ");
  appendFile(SD_MMC, "/hello.txt", "World!\n");
  readFile(SD_MMC, "/hello.txt");
  deleteFile(SD_MMC, "/foo.txt");
  renameFile(SD_MMC, "/hello.txt", "/foo.txt");
  readFile(SD_MMC, "/foo.txt");
  testFileIO(SD_MMC, "/test.txt");
  Serial.printf("Total space: %lluMB\n", SD_MMC.totalBytes() / (1024 * 1024));
  Serial.printf("Used space: %lluMB\n", SD_MMC.usedBytes() / (1024 * 1024));
}

void loop() {}

`

VojtechBartoska, 2024-07-16:
Hello @raprakashvi, it really looks like a power issue. I will close this issue, if needed you can reopen it.

closed 2024-07-16 by VojtechBartoska