espressif/arduino-esp32 #10048: Calling nvs_flash_deinit() Causes WiFi.begin() to Fail

emintz, 2024-07-18:
### Board

AITRIP  38pin ESP-WROOM-32 ESP32 ESP-32S Development Board (https://www.amazon.com/gp/product/B0B12822SF/ref=ppx_yo_dt_b_search_asin_title?th=1)

### Device Description

This is the bare board as provided by the manufacturer. It is a bare bones system consisting of an ESP32 S2 with builtin WiFi antenna on a board with a voltage regulator, LED, USB interface, and a few other supporting parts. It is a generic, low-end product that I bought because it is inexpensive and works.

### Hardware Configuration

No peripherals are attached. 

### Version

v3.0.3

### IDE Name

Sloeber

### Operating System

Ubuntu 22.04

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I need to retrieve WiFi credentials and other configuration settings from NVRAM. I will run a short configuration sketch to set them, then install the application that will retrieve them. I have disabled "wipe flash on upload".

I am using the nvs library to retrieve the configuration values. WiFi.begin() fails if I invoke nvs_flash_deinit().

### Sketch

```cpp
Here's the entire sketch. Note that the two static char arrays will hold the retrieved credentials in the final version. Note that I have redacted sensitive information.

--------------------------------------------------------------------------------

#include "Arduino.h"
#include "WiFi.h"
#include <cstring>

#include "nvs_flash.h"

static char ssid[64];
static char password[64];

void setup() {
  Serial.begin(115200);
  Serial.print("Simple, brain dead WIFI station starting, compiled on ");
  Serial.print(__DATE__);
  Serial.print(" at ");
  Serial.print(__TIME__);
  Serial.println(".");

  esp_err_t esp_status;

  if (ESP_OK == (esp_status = nvs_flash_init())) {
    Serial.println("Flash memory initialized.");
  } else {
    Serial.printf("Flash initialization failed with status %x.\n", esp_status);
  }
//  WiFi.begin() fails if this code is uncommented
//      |                       |
//      v                       v
//
//  if (ESP_OK == (esp_status = nvs_flash_deinit())) {
//    Serial.println("Flash shut down successfully.");
//  } else {
//    Serial.printf("Flash shut down failed with status %x.\n", esp_status);
//  }
//      ^                       ^
//      |                       |
// End of block.

  memset(ssid, 0, sizeof(ssid));
  memset(password, 0, sizeof(password));

  WiFi.begin("SSID", "*********");
  Serial.print("Waiting for WiFi connection ");
  while (WiFi.status() != wl_status_t::WL_CONNECTED) {
    vTaskDelay(pdMS_TO_TICKS(500));
    Serial.print('.');
  }
  Serial.println(" done!");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

}

// The loop function is called in an endless loop
void loop()
{
//Add your repeated code here
}

--------------------------------------------------------------------------------
Work around: don't shutdown non-volatile storage.
```

### Debug Message

```plain
The sketch runs perfectly when nvs_flash_deinit() is not invoked and produces the following output:

Simple, brain dead WIFI station starting, compiled on Jul 18 2024 at 17:21:33.
Flash memory initialized.
Waiting for WiFi connection . done! 
IP address: 192.168.XX.XX

When the nvs_flash_deinit() is invoked before WiFi.begin(), it produces:

Simple, brain dead WIFI station starting, compiled on Jul 18 2024 at 17:16:11.
Flash memory initialized.
Flash shut down successfully.
E (38) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
E (38) wifi_init: Failed to deinit Wi-Fi (0x3001)
[    42][E][WiFiGeneric.cpp:248] wifiLowLevelInit(): esp_wifi_init 4353
[    58][E][STA.cpp:298] begin(): STA enable failed!
Waiting for WiFi connection ...............................................................

The connection wait hangs until I power off the board.
```

### Other Steps to Reproduce

Sorry, nothing comes to mind.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-19:
NVS is used by WiFi. Why are you deiniting it?

emintz, 2024-07-19:
Thank you for answering so quickly.

The dependency appears not to be documented, so I thought I could save
resources by removing it when I no longer need it. Please consider marking
the issue as a request that the dependency be documented..

OrhanYigitDurmaz, 2024-08-05:
> Thank you for answering so quickly. The dependency appears not to be documented, so I thought I could save resources by removing it when I no longer need it. Please consider marking the issue as a request that the dependency be documented..
> […](#)

it documented actually

closed 2025-03-13 by Parsaabasi