espressif/arduino-esp32 #10050: Issue with Heltec LoRa ESP32 V2 Not Connecting to TTN Server

vijayincise, 2024-07-19:
### Related area

Heltec LoRa ESP32 V2 

### Hardware specification

esp32-s3

### Is your feature request related to a problem?

Hello,

I am experiencing difficulties connecting my Heltec LoRa ESP32 V2 module to The Things Network (TTN) server using the OTAA example code provided by the TTN_esp32 library. I have encountered errors with both the latest and previous versions of the library and ESP32 board manager.

Setup Details:

Heltec LoRa ESP32 V2
TTN_esp32 library version: 0.1.7
ESP32 board manager version: 3.0.2
Code Used:

```
#include <TTN_esp32.h>
#include "TTN_CayenneLPP.h"

const char* devEui = "70B3D57ED005EC18";
const char* appEui = "0000000000000100";
const char* appKey = "C7A02794D8631B7A2E307F766C8BC3F6";

TTN_esp32 ttn;
TTN_CayenneLPP lpp;

void message(const uint8_t* payload, size_t size, uint8_t port, int rssi)
{
    Serial.println("-- MESSAGE");
    Serial.printf("Received %d bytes on port %d (RSSI=%ddB) :", size, port, rssi);
    for (int i = 0; i < size; i++)
    {
        Serial.printf(" %02X", payload[i]);
    }
    Serial.println();
}

void setup()
{
    Serial.begin(115200);
    Serial.println("Starting");
    ttn.begin();
    ttn.onMessage(message);
    ttn.join(devEui, appEui, appKey);
    Serial.print("Joining TTN ");
    while (!ttn.isJoined())
    {
        Serial.print(".");
        delay(500);
    }
    Serial.println("\njoined !");
    ttn.showStatus();
}

void loop()
{
    static float nb = 18.2;
    nb += 0.1;
    lpp.reset();
    lpp.addTemperature(1, nb);
    if (ttn.sendBytes(lpp.getBuffer(), lpp.getSize()))
    {
        Serial.printf("Temp: %f TTN_CayenneLPP: %d %x %02X%02X\n", nb, lpp.getBuffer()[0], lpp.getBuffer()[1],
            lpp.getBuffer()[2], lpp.getBuffer()[3]);
    }
    delay(10000);
}
```
Errors Encountered With TTN_esp32 library version 0.1.7 and ESP32 board manager version 3.0.2:

```
c:\Users\Vijay\Documents\Arduino\libraries\TTN_esp32\src\TTN_BLE_esp32.cpp:4:10: fatal error: esp_int_wdt.h: No such file or directory
    4 | #include <esp_int_wdt.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Compilation error: exit status 1
```
tried another version:

Errors Encountered With TTN_esp32 library version 0.1.6 and ESP32 board manager version 2.0.17:

```
ELF file SHA256: 32253762a57caf58

Rebooting...
Starting
Using stored keys to join
FAILURE 
c:\Users\Vijay\Documents\Arduino\libraries\TTN_esp32\src\lmic\lmic\radio.c:1164
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:

Backtrace: 0x400f82c7:0x3ffbc830 0x400e084a:0x3ffbc850 0x40089538:0x3ffbc870
```
c:\Users\Vijay\Documents\Arduino\libraries\TTN_esp32\src\lmic\lmic\radio.c:1164-> ASSERT( i != 0 ); // this is line i’ve been facing issue.

Steps Taken:

 1.Verified the correct installation of the latest ESP32 board support package.
 2.Tried different versions of the TTN_esp32 library and ESP32 board manager.
 3.Ensured the device is powered properly and connected with antenna .
Despite these efforts, the issue persists. I would appreciate any guidance or suggestions on resolving this problem.

Thank you for your assistance.

### Describe the solution you'd like

Not Connecting to TTN Server

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-07-20:
From TTN_esp32 on github:
> This repository is archived. Please use [EzLoRaWAN](https://github.com/rgot-org/EzLoRaWAN) for now. The new library can use SX127X and SX126X chips.

Jason2866, 2025-03-12:
Closing issue is outdated. Feel free to ask to reopen if the issue still exists in actual Arduino core

closed 2025-03-12 by Jason2866