espressif/arduino-esp32 #10052: git clone failed

YFJiang, 2024-07-20:
### Related area

ESP32

### Hardware specification

ESP32

### Is your feature request related to a problem?

when i try to clone this repo, it reports below errors:" git  clone --progress --recursive -v  https://github.com/espressif/arduino-esp32.git   hardware/espressif/esp32
Cloning into 'hardware/espressif/esp32'...
POST git-upload-pack (185 bytes)
POST git-upload-pack (gzip 3612 to 1828 bytes)
remote: Enumerating objects: 55147, done.
remote: Counting objects: 100% (1242/1242), done.
remote: Compressing objects: 100% (592/592), done.
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
error: 4847 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
"

### Describe the solution you'd like

can do clone

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-07-20:
this must be some local or temporary problem. Repository is working fine

Jason2866, 2024-07-21:
If you are located in China, these are known issues. There is the chinese firewall...

VojtechBartoska, 2024-07-22:
as mentioned above, this problem is more likely on your end.

Can I consider this issue as answered?

everslick, 2024-08-17:
I see this also from time to time. But in any case it is either a wonky internet connection or something at githubs side.

I believe this issue can be closed.