espressif/arduino-esp32 #10055: Cannot build due to "ERROR: Because project depends on idf (>=5.1.0,<5.2.0) which doesn't match"

plystdr, 2024-07-20:
### Board

Custom esp32-s3 board

### Device Description

Custom board.

### Hardware Configuration

Multiple sensors.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO VSCode

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Cannot make platformio build the project. I even tried skip version checking but nothing works...

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[common]
lib_deps = 
	https://github.com/joltwallet/esp_littlefs.git
	ayushsharma82/ElegantOTA@^3.1.2

[env:test-project]
platform = https://github.com/platformio/platform-espressif32.git
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
framework = espidf, arduino
board = test-board
board_build.mcu = esp32s3
upload_protocol = esptool
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = partition_scheme.csv
board_build.filesystem = littlefs
board_upload.offset_address = 0x670000
lib_deps = 
	${common.lib_deps}
extra_scripts = 
	tasks.py
build_flags = 
	-DCORE_DEBUG_LEVEL=5
	-DBOARD_HAS_PSRAM
	-DARDUINO_ESP32S3_DEV
	-DCONFIG_SPIRAM_USE
	-mfix-esp32-psram-cache-issue
	-w
	-Wno-unused-variable
	-Wno-unused-function
	-DELEGANTOTA_USE_ASYNC_WEBSERVER=1
	-DARDUINO_SKIP_IDF_VERSION_CHECK=1

### Sketch

```cpp
Not even start compiling.
```

### Debug Message

```plain
Processing projecttest (platform: https://github.com/platformio/platform-espressif32.git; framework: espidf, arduino; board: testboard)
--------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION:
PLATFORM: Espressif 32 (6.7.0+sha.68ad40f) > Espressif ESP32-S3-N8-R2 (8 MB QD, 2MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.3+sha.690bdb5
 - framework-espidf @ 3.40407.240606 (4.4.7)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
-- Building ESP-IDF components for target esp32s3
Manifest files have changed, solving dependencies.

...........NOTICE: Skipping optional dependency: espressif/cbor

.......NOTICE: Skipping optional dependency: espressif/mdns

....NOTICE: Skipping optional dependency: espressif/jsmn

...........-- Configuring incomplete, errors occurred!
See also "D:/firmware/projecttest/.pio/build/projecttest/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/usertest/.platformio/packages/framework-espidf/tools/cmake/build.cmake:463 (message):
  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  ERROR: Because project depends on idf (>=5.1.0,<5.2.0) which doesn't match
  any versions, version solving failed.

Call Stack (most recent call first):
  C:/Users/usertest/.platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:7 (project)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-21:
cc @Jason2866 

Official Platformio does not support Arduino v3. There are other ways to get this going with PlatformIO supported by the community

Jason2866, 2024-07-21:
@plystdr  You can use the community version which supports https://github.com/pioarduino/platform-espressif32

```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip
board = ...
...
```

VojtechBartoska, 2024-07-22:
PlatfromIO issue, solution is proposed above. Marking as wontfix at our end.

Can I consider this as answered @plystdr? thanks!

closed 2024-07-22 by plystdr

WebDust21, 2024-08-01:
> @plystdr You can use the community version which supports https://github.com/pioarduino/platform-espressif32 See examples folders to see how Arduino as component setup has to be done
> 
> ```
> platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip
> board = ...
> ...
> ```

@Jason2866  You're a hero.
Got myself stuck up a tree with no way down after finding an issue deep in ESP-IDF (using Arduino-ESP32), only to find actually trying to use ESP-IDF a very probable-cause reason that so many programmers are completely bald...

Your repo and suggestions actually WORK, thank you!

Now I just need to figure out how to get a modified repository (based on the proper ESP-IDF version) linked into PlatformIO with my corrections in it to verify that the issue is fixed...

(yes, the corrections DO work...now I'll have to PR the two lines of changed code deeeeeep in the Bluedroid libraries...)

Jason2866, 2024-08-01:
@WebDust21 To use your own IDF version is easy. Fork https://github.com/pioarduino/esp-idf/tree/release/v5.1
Do your changes there. Create a zip including the git submodules. Easiest to do is via this github actions https://github.com/pioarduino/esp-idf/blob/release/v5.1/.github/workflows/release_zips.yml It will create the zip and does a release in your fork. Fork https://github.com/pioarduino/platform-espressif32 and change this entry pointing to your IDF https://github.com/pioarduino/platform-espressif32/blob/18cb216bea0ffb63390d035d76451af9d8c0435d/platform.json#L48
Use your platform with
```
platform = https://github.com/webdust21/platform-espressif32.git#develop
```

CompunixSupport, 2025-06-11:
One possible issue is that it is looking for various tools in  .platformio/tool folder (nonexistent), but installing them to .platformio/packages. I have tried to duplicate them in the created .platformio/tool. Not sure how to correct it. Please advise.

set following in platformio.ini:
platform_packages =
    espressif/toolchain-xtensa-esp32
    platformio/tool-ninja@1.9.0
    platformio/tool-mklittlefs@^1.203.210628
    platformio/tool-cmake@^3.30.2

Upon VScode start it checks for the enviroment and get following error:
Resolving esp32dev dependencies...
Tool Manager: Installing file://C:\Users\DaliborSimacek\.platformio\tools\toolchain-xtensa-esp-elf
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing file://C:\Users\DaliborSimacek\.platformio\tools\tool-esp-rom-elfs
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/xtensa-esp-elf-14.2.0_20241119.zip
Unpacking 0% 10% 20%
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing platformio/tool-mklittlefs @ ^1.203.210628
Unpacking 0% 10% 20%
Tool Manager: tool-mklittlefs@1.203.210628 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esp-rom-elfs-20241011.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
Tool Manager: Installing platformio/tool-ninja @ 1.9.0
Unpacking 0% 10% 20%
Tool Manager: tool-ninja@1.9.0 has been installed!
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing esp32dev (platform: https://github.com/pioarduino/platform-espressif32.git#develop; board: esp32dev; framework: espidf, arduino)
--------------------------------------------------------------------------------
Tool Manager: Installing file://C:\Users\DaliborSimacek\.platformio\tools\toolchain-xtensa-esp-elf
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Wrong version for tool-ninja. Installing needed version...
Tool Manager: Installing file://C:\Users\DaliborSimacek\.platformio\tools\tool-esp-rom-elfs
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/xtensa-esp-elf-14.2.0_20241119.zip
Unpacking 0% 10% 20%
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing platformio/tool-mklittlefs @ ^1.203.210628
Unpacking 0% 10% 20%
Tool Manager: tool-mklittlefs@1.203.210628 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esp-rom-elfs-20241011.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
Tool Manager: Installing platformio/tool-ninja @ 1.9.0
Unpacking 0% 10% 20%
Tool Manager: tool-ninja@1.9.0 has been installed!
Tool Manager: Installing file://C:\Users\DaliborSimacek\.platformio\tools\toolchain-xtensa-esp-elf
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Wrong version for tool-ninja. Installing needed version...
Tool Manager: Installing file://C:\Users\DaliborSimacek\.platformio\tools\tool-esp-rom-elfs
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
Tool Manager: Installing file://C:\Users\DaliborSimacek\.platformio\tools\toolchain-xtensa-esp-elf
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing file://C:\Users\DaliborSimacek\.platformio\tools\tool-esp-rom-elfs
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (54.3.20+develop.sha.17b022c) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.0 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.aed8bdc8dd 
 - framework-espidf @ 3.50401.0 (5.4.1) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.11 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
[... 6 lines cut ...]
  File "C:\Users\DaliborSimacek\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\DaliborSimacek\.platformio\platforms\espressif32\builder\main.py", line 399:
    target_elf = env.BuildProgram()
  File "C:\Users\DaliborSimacek\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\DaliborSimacek\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "C:\Users\DaliborSimacek\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\DaliborSimacek\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 142:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
[... cut ...]
  File "C:\Users\DaliborSimacek\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\DaliborSimacek\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\DaliborSimacek\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 155:
    assert os.path.isdir(TOOLCHAIN_DIR)
========================= [FAILED] Took 40.75 seconds =========================

CompunixSupport, 2025-06-12:
As previously mentioned, same result: FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\DaliborSimacek\\.platformio\\tools\\toolchain-xtensa-esp-elf' looking into non-existing "tools" folder and I cannot find anywhere the setting. It happens after the change to platform = https://github.com/pioarduino/platform-espressif32.git#develop, before it looks for .platformio\packages I could see the settings for.
If I create a simple project with arduino component (only WiFi and Wire components) it works. Then adding library NTPClient will cause the issue from the top of the page. After change to platform = https://github.com/pioarduino/platform-espressif32.git#develop, and deleting the folders as you suggested it starts to look for the .platformio/tools folder, which does not get created.