espressif/arduino-esp32 #10057: Incorrect quartz frequency

iSetebos, 2024-07-21:
### Board

TTGO Lora32-OLED

### Device Description

TTGO Lora32-OLED

### Hardware Configuration

Use serial

### Version

v3.0.3

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

I use Arduino 2.3.2. Board of TTGO Lora32-OLED. Quartz on the board is 26 MHz. The seril is set to 115200, but the data is output at a frequency of 74880. The getXtalFrequencyMhz function outputs 40MHz. The same project works fine in arduino 1.8.5.

### Sketch

```cpp
/*
   Simple Sketch for testing HardwareSerial with different CPU Frequencies
   Changing the CPU Frequency may affect peripherals and Wireless functionality
   In ESP32 Arduino, UART shall work correctly in order to let the user see DGB info
   and other application messages.

   CPU Frequency is usually lowered in sleep modes
   and some other Low Power configurations

*/

int cpufreqs[6] = {240, 160, 80, 40, 20, 10};
#define NUM_CPU_FREQS (sizeof(cpufreqs) / sizeof(int))

void setup() {

  Serial.begin(115200);
  delay(1000);
  Serial.println("\n Starting...\n");
  Serial.flush();

  // initial information
  uint32_t Freq = getCpuFrequencyMhz();
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getXtalFrequencyMhz();
  Serial.print("XTAL Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getApbFrequency();
  Serial.print("APB Freq = ");
  Serial.print(Freq);
  Serial.println(" Hz");
  delay(500);

  // ESP32-C3 and other RISC-V target may not support 240MHz
#ifdef CONFIG_IDF_TARGET_ESP32C3
  uint8_t firstFreq = 1;
#else
  uint8_t firstFreq = 0;
#endif

  // testing HardwareSerial for all possible CPU/APB Frequencies
  for (uint8_t i = firstFreq; i < NUM_CPU_FREQS; i++) {
    Serial.printf("\n------- Trying CPU Freq = %d ---------\n", cpufreqs[i]);
    Serial.flush();  // wait to empty the UART FIFO before changing the CPU Freq.
    setCpuFrequencyMhz(cpufreqs[i]);
    Serial.updateBaudRate(115200);

    Freq = getCpuFrequencyMhz();
    Serial.print("CPU Freq = ");
    Serial.print(Freq);
    Serial.println(" MHz");
    Freq = getXtalFrequencyMhz();
    Serial.print("XTAL Freq = ");
    Serial.print(Freq);
    Serial.println(" MHz");
    Freq = getApbFrequency();
    Serial.print("APB Freq = ");
    Serial.print(Freq);
    Serial.println(" Hz");
    if (i < NUM_CPU_FREQS - 1) {
      Serial.println("Moving to the next frequency after a pause of 2 seconds.");
      delay(2000);
    }
  }
  Serial.println("\n-------------------\n");
  Serial.println("End of testing...");
  Serial.println("\n-------------------\n");
}

void loop() {
  // Nothing here so far
}
```

### Debug Message

```plain
l��b�2No�b#"s$������l���ld$~r$����p��|$�l$l{l`��|$$~rldrlcorlcor�|�oN��ے������d��$bn{$bn{$bnr��ے������l��l$ls$����p��|$�l$l{$`�
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-21:
Which version of the board is this? There seem to be V1, V2 and V2.1 selectable. As an easy way to fix this, you can add a new file in the sketch folder named `build_opt.h` and have it's content be `-DF_XTAL_MHZ=26`. This will solve it for now, once we know the version of the board we can add it for next release to be automatically applied

schloeb, 2024-10-28:
Arduino IDE Vs. 2.3.3   on Oct 27th 2024
Had the same issue, solved with the  buid_opt.h  solution. Thanks!

Parsaabasi, 2025-03-11:
Hi @iSetebos 

Is this still valid?

Jason2866, 2025-03-12:
Closing since not Arduino core code related

closed 2025-03-12 by Jason2866