espressif/arduino-esp32 #10058: Arduino as component not working

adiiot-1316, 2024-07-21:
### Board

ESP32-C6-WROOM-1-N8

### Device Description

ESP32-C6-WROOM-1-N8

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Visual Studio

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am trying to update the sample Hello example to be used as Arduino as component in Visual Studio Code. Steps followed
1.)Compiled Hello example without any error
2.)Added Arduino as Component.
3.)Renamed main.c to main.cpp
4.)Updated path from main.c to main.cpp inn maini/CMakeLists.txt file.
5.)Removed all code from main.cpp and added void setup and void loop function.
6.)Rebuild the project.

### Sketch

```cpp
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include "arduino.h";
void setup(){
    Serial.begin(115200);
}
void loop(){
    Serial.println("Tesitng for some time");
    delay(3000);
}
```

### Debug Message

```plain
*  Executing task: C:\Espressif\tools\ninja\1.10.2\ninja.exe  

[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
NOTICE: Skipping optional dependency: espressif/esp-sr
........NOTICE: Skipping optional dependency: espressif/cbor
.......NOTICE: Skipping optional dependency: espressif/mdns
....NOTICE: Skipping optional dependency: espressif/jsmn
.......................................HINT: Please check manifest file of the following component(s): arduino
CMake Error at C:/Users/Addy-Machine/esp/v4.4.8/esp-idf/tools/cmake/build.cmake:463 (message):
  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  ERROR: Because project depends on idf (>=5.1,<5.2) which doesn't match any
  versions, version solving failed.

Call Stack (most recent call first):
  C:/Users/Addy-Machine/esp/v4.4.8/esp-idf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v5.2.1/workspace/arduino-comp/hello_world/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
C:\Espressif\tools\cmake\3.23.1\bin\cmake.exe --regenerate-during-build -SC:\Espressif\frameworks\esp-idf-v5.2.1\workspace\arduino-comp\hello_world -BC:\Espressif\frameworks\esp-idf-v5.2.1\workspace\arduino-comp\hello_world\build
ninja: error: rebuilding 'build.ninja': subcommand failed

 *  The terminal process "C:\Espressif\tools\ninja\1.10.2\ninja.exe" terminated with exit code: 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-21:
1) Can you build an [example from esp-idf](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/get-started/hello_world)? It seems like you don't have esp-idf available.
2) The include is for "Arduino.h". Case matters

me-no-dev, 2024-07-21:
`ERROR: Because project depends on idf (>=5.1,<5.2) which doesn't match any versions, version solving failed.`

Read the error. You have IDF v4.4.8 and you need IDF v5.1.4

adiiot-1316, 2024-07-25:
@me-no-dev  : Thanks for info. I installed IDF v5.1.4 from espressif Web Downloads. However when  i am configuring through VS Code as  already installed esp-idf, its throwing error.
**Error:** Using existing C:\Espressif\tools\idf-python\3.11.2
"C:\Espressif\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)
Installing ESP-IDF Debug Adapter python packages in C:/Espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe ...

**Detailed Logs**
Using existing C:\Espressif\dist\idf-git-2.39.2-win64.zip
Installing C:\Espressif\tools\idf-git\2.39.2 ...
Warning File C:\Espressif\tools\idf-git\2.39.2\cmd\git.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\cmd\scalar.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\awk.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\b2sum.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\basename.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\basenc.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\cat.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\chattr.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\cmp.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\comm.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\cp.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\cut.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\cygwin-console-helper.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\dash.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\date.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\diff.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\diff3.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\dirname.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\echo.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\env.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\expr.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\false.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\find.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\gencat.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\getfacl.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\getopt.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\gmondump.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\grep.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\head.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\ls.exe
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\bin\lsattr.exe
                                        already exists and was not updated.
[... 175 lines cut ...]
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\share\licenses\libsqlite\LICENSE
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\share\licenses\openssh\LICENCE
                                        already exists and was not updated.
Warning File C:\Espressif\tools\idf-git\2.39.2\usr\share\licenses\openssl\LICENSE
[... cut ...]
                                        already exists and was not updated.
Extracted C:\Espressif\tools\idf-git\2.39.2 ...

Any help would be appreciated please. I have written program into Arduino and now i want to use flash encryption to secure my code, which is not possible in Arduino. 
Hence i have to port it to ESP-idf code to use features like Flash encryption, OTA update, other security features. 

Thanks in  advance for all your support and bearing with me as i am completely new to espressif.

adiiot-1316, 2024-07-25:
> 1. Can you build an [example from esp-idf](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/get-started/hello_world)? It seems like you don't have esp-idf available.
> 2. The include is for "Arduino.h". Case matters

@lbernstone : Thanks for reply. I tried to create a sample hello_world template. And then added Arduino as component. Below is the error. Any idea?
`
 *  Executing task: C:\Espressif\tools\ninja\1.10.2\ninja.exe  

[2/133] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\Addy-Machine\Documents\tmpArduino\hello_world\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.1.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/Addy-Machine/Documents/tmpArduino/hello_world/build/bootloader/bootloader.bin"      
Bootloader binary size 0x5350 bytes. 0x2cb0 bytes (35%) free.
[4/131] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/chip-debug-report.cpp.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/chip-debug-report.cpp.obj 
C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -DESP_PLATFORM -DIDF_VER=\"HEAD-HASH-NOTFOUND\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/build/config -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/variants/esp32c6 -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/cores/esp32 -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/ArduinoOTA/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/AsyncUDP/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/BLE/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/BluetoothSerial/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/DNSServer/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/EEPROM/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/ESP_I2S/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/ESP_NOW/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/ESP_SR/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/ESPmDNS/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/Ethernet/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/FFat/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/FS/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/HTTPClient/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/HTTPUpdate/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/Insights/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/LittleFS/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/NetBIOS/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/Network/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/OpenThread/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/PPP/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/Preferences/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/RainMaker/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/SD_MMC/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/SD/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/SimpleBLE/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/SPIFFS/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/SPI/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/Ticker/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/Update/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/USB/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/WebServer/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/NetworkClientSecure/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/WiFi/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/WiFiProv/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/libraries/Wire/src -IC:/Users/Addy-Machine/Documents/tmpArduino/hello_world/components/arduino/cores/esp32/libb64 -IC:/Espressif/frameworks/esp-idf-v5.1.2/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/include -IC:/Espressif/frameworks/esp-idf [... cut ...] ESP_PARTITION_SUBTYPE_DATA_SPIFFS'?
  198 |             case ESP_PARTITION_SUBTYPE_DATA_LITTLEFS:  chip_report_printf("LITTLEFS"); break;
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  ESP_PARTITION_SUBTYPE_DATA_SPIFFS
[9/131] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/Print.cpp.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Espressif\tools\ninja\1.10.2\ninja.exe" terminated with exit code: 1. 
`

me-no-dev, 2024-07-29:
It is again caused by some version mismatch, but you could easily "fix" this and get going by commenting out [this line](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/chip-debug-report.cpp#L198)

adiiot-1316, 2024-07-29:
@me-no-dev : I commented the line but getting below error now. 

[image]

[image]

me-no-dev, 2024-07-29:
Are you positive this is ESP-IDF v5.1.4 or the latest `release/v5.1` branch of ESP-IDF?

adiiot-1316, 2024-07-29:
Its v5.1.2 . 
Though v5.1.4 is already installed in system but VSCode is not picking up the v5.1.4 . It is showing below error related to python pip

`C:\Espressif\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)`

me-no-dev, 2024-07-29:
Please check with the [IDF VSCode Extension Repo](https://github.com/espressif/vscode-esp-idf-extension) on how to properly set to use IDF 5.1.4

adiiot-1316, 2024-08-02:
@me-no-dev : Thanks a lot man !! Was able to setup IDF 5.1.4 . But still some issue in using Arduino as component. It would be really helpful if we can connect over email to debug this further, which i assume will be easier for you as well to debug real time? Also really appreciate your support and patience.