espressif/arduino-esp32 #10059: Esp - now slave only receives messages when master has serial monitor open

PaulosKapa, 2024-07-21:
### Board

esp32 s2

### Device Description

esp 32 s2 mini

### Hardware Configuration

gpio 2-7 are connected to push buttons

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I followed the code on the esp - now master and slave examples. The master sends data, but only when serial monitor is open. When I don't send the code to the Serial monitor, the message isn't sent I want to send a struct, so I can't use the snprintf command.  The following code works, but only when the serial monitor is open at arduino ide. Otherwise, it doesn't receive any messages. I haven't change anything else from the example code rather than the code i have below

### Sketch

```cpp
////master
Serial.println(String(sensorData.X) + ',' + String(sensorData.Y) + ',' + String(sensorData.Z)+ ',' +String(sensorData.shot) + ',' +  String(sensorData.actions) + ',' +String(sensorData.magId) +','+String(sensorData.unlock));

if (!broadcast_peer.send_message((uint8_t *) &sensorData, sizeof(sensorData))) {
    Serial.println("Failed to broadcast message");

///slave
void onReceive(const uint8_t *data, size_t len, bool broadcast) {
    SensorData sensorData;
   // Serial.printf("Received a message from master " MACSTR " (%s)\n", MAC2STR(addr()), broadcast ? "broadcast" : "unicast");
    memcpy(&sensorData, data, sizeof(sensorData));
    Serial.println(String(sensorData.X) + ',' + String(sensorData.Y) + ',' + String(sensorData.Z)+ ',' +String(sensorData.shot) + ',' +  String(sensorData.actions) + ',' +String(sensorData.magId) +','+String(sensorData.unlock));
  }
```

### Debug Message

```plain
There is no debug message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-21:
[You are looping to see if Serial is established](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP_NOW/examples/ESP_NOW_Broadcast_Slave/ESP_NOW_Broadcast_Slave.ino#L83-L85).  If you are using USB-CDC as your serial port, this will not be true unless you plug something in there.

PaulosKapa, 2024-08-26:
Thank you that was the answer.

closed 2024-08-26 by PaulosKapa