espressif/arduino-esp32 #10068: W5500 Ethernet module does not detect connection

Armagedon13, 2024-07-23:
### Board

NodeMCU-32S

### Device Description

[image]

### Hardware Configuration

```
#ifndef ETH_PHY_TYPE
#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS 15
#define ETH_PHY_IRQ 4
#define ETH_PHY_RST 5
#endif

// SPI pins
#define ETH_SPI_SCK 14
#define ETH_SPI_MISO 12
#define ETH_SPI_MOSI 13
```

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have a weight scale that sends Ethernet UDP CSV data, it is connected directly to the esp32 via Ethernet to receive the data.
ESP32 has a 5V-1A power supply that also powers the W5500 and has a fixed IP.
The weight scale is a KAIO FULL LCD, curiously it has a wiznet W5500 and also has a fixed IP.
The configuration of both works when I connect either to a PC (without DHCP, without internet), from any program that looks at the data coming from the selected IP and port, I can see that the messages from the scale arrive to the PC, but when I connect The scale to the ESP32 starts the Ethernet but don't detect the connection and don't get the ip, but when I connect the ESP32 to the PC, it instantly starts the Ethernet.

### Sketch

```cpp
/*
Transmisor
*/

#ifndef ETH_PHY_TYPE
#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS 15
#define ETH_PHY_IRQ 4
#define ETH_PHY_RST 5
#endif

// SPI pins
#define ETH_SPI_SCK 14
#define ETH_SPI_MISO 12
#define ETH_SPI_MOSI 13

#include <ETH.h>
#include <SPI.h>
#include <NetworkUdp.h>

#include <esp_now.h>
#include <WiFi.h>
#include <esp_wifi.h>

static bool eth_connected = false;

NetworkUDP Udp;                    // create UDP object
unsigned int localUdpPort = 1001;  // Local port number

IPAddress local_ip(192, 168, 0, 140);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);

IPAddress dns1(8, 8, 8, 8);             //optional
IPAddress dns2 = (uint32_t)0x00000000;  //optional

void setup() {
  Serial.begin(115200);
  // Ethernet INIT
  Network.onEvent(onEvent); // don't send it here to much code space

  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);
  ETH.config(local_ip, gateway, subnet, dns1);  // Static IP, leave without this line to get IP via DHCP
  Udp.begin(local_ip, localUdpPort);  // Enable UDP listening to receive data
  delay(5000);

}
void loop() {
  // Ethernet data udp handle
  if (eth_connected) {
    int packetSize = Udp.parsePacket();  // Get the current team header packet length
    if (packetSize) {                    // If data is available
      char incomingData[250];
      int len = Udp.read(incomingData, 250);
      if (len > 0) {
        incomingData[len] = '\0';
      }
      //Send the data over ESP-NOW
      struct_message payload;
      strncpy(payload.data, incomingData, sizeof(payload.data));
      payload.data[sizeof(payload.data) -1] = '\0';

      Serial.println();
      Serial.print("Received: ");
      Serial.println(payload.data);
      Serial.print("From IP: ");
      Serial.println(Udp.remoteIP());
      Serial.print("From Port: ");
      Serial.println(Udp.remotePort());

      esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *)&payload, sizeof(payload));
      if (result == ESP_OK) {
        Serial.println("Envío exitoso");
      } else {
        Serial.println("Error al enviar los datos");
      }
    }
  }
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1288
load:0x40078000,len:13872
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3048
entry 0x40080590
[     4][D][esp32-hal-cpu.c:264] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 1.44
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   332948 B ( 325.1 KB)
  Free Bytes        :   302608 B ( 295.5 KB)
  Allocated Bytes   :    23276 B (  22.7 KB)
  Minimum Free Bytes:   297228 B ( 290.3 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : DIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jul 23 2024 15:14:11
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.4-497-gdc859c1e67-dirty
  Arduino Version   : 3.0.3
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : NODEMCU_32S
  Arduino Variant   : nodemcu-32s
  Arduino FQBN      : esp32:esp32:nodemcu-32s:UploadSpeed=921600,FlashFreq=80,DebugLevel=debug,EraseFlash=all
============ Before Setup End ============
ETH Started
WiFi interface ready
WiFi STA client started
STA MAC: C8:F0:9E:52:78:94
STA CHANNEL 6
ESP-NOW initialized
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   332948 B ( 325.1 KB)
  Free Bytes        :   245024 B ( 239.3 KB)
  Allocated Bytes   :    77396 B (  75.6 KB)
  Minimum Free Bytes:   244900 B ( 239.2 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
     4 : ETH_IRQ[0]
     5 : ETH_RST[0]
    12 : SPI_MASTER_MISO[3]
    13 : SPI_MASTER_MOSI[3]
    14 : SPI_MASTER_SCK[3]
    15 : ETH_CS[0]
    19 : GPIO
    21 : GPIO
    22 : GPIO
    23 : GPIO
============ After Setup End =============
Heartbeat sent to Receiver
Connection to Receiver established
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JAndrassy, 2024-07-23:
did you try a crossed cable?

Armagedon13, 2024-07-23:
yes crossover cable works, thanks for fast response, lan8720 have auto MDI-X?

closed 2024-07-23 by Armagedon13