espressif/arduino-esp32 #10071: WifiClientSecure (arduino-esp32 v. 3.x) connect() fail on some server (e.g. "speechgen.io") - same code working well with 2.x

kaloprojects, 2024-07-24:
### Board

ESP32 DevKit

### Device Description

-

### Hardware Configuration

-

### Version

v3.0.3

### IDE Name

Arduino IDE (1.8.19)

### Operating System

Windows 7

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hope someone can review this potential issue: 
_WifiClientSecure client.connect("some_server_xy", 443) fail_

Since upgrade to 3.x (currently using 3.0.3) my earlier code is no longer working. I spent day in searching for root cause (contacted e.g. speechgen.io support team), but found root cause yesterday (more by accident): seems to be a library 2.x to 3.x issue. _I downgraded for verification to arduino-esp32 2.x (2.0.16) and same code is working well (server connect successful ..i can call their TTS service API)._ 

So can someone maybe check _why_ this code snippet fails with 3.x: (sketch see below) ?

Hope some folks can analyze or fix a potential bug (as my complete [project ](https://github.com/kaloprojects/KALO-ESP32-Voice-Assistant)is on hold now due this issue, also can’t downgrade as mitigation because i need 3.x for lot other features, e.g. new /i2s_std.h). I also tried '<NetworkClientSecure.h>', HTTPClient etc. ..instead of WifiClientSecure .. all with same result: Successful client.connect() with  Arduino-esp32 v. 2.x , but failing with 3.x on server "speechgen.io". Some dedicated server (e.g."api.openai.com") connect succesfully with 3.x and 2.x .. but my urgent needed "speechgen.io" server fails since 3.x 

2 questions (do you @me-no-dev or any other folks have an idea how to solve ?
- Is this a well known issue (some server failing on 3.x  .. and some not) ?
- is there any workaround to connect e.g. _"speechgen.io"_ successfully also with 3.x ? (all idea so much welcome!)

Thx !

### Sketch

```cpp
// Https Connect to Speechgen server "speechgen.io" port 443 
// -> always fails with Arduino-esp32 v. 3.x library (but same code result success with earlier Arduino-esp32 v. 2.x )
// -> other server (e.g. "api.openai.com" are working (also with 3.x)

#include <WiFi.h>           
#include <WiFiClientSecure.h> 
/* .. */
WiFiClientSecure client;  
client.setInsecure();                          
/* .. */  
if ( client.connect("speechgen.io", 443) )   // <- it FAILS here always (with ESP 3.x, only working with 2.x)
{  Serial.println("> Succesfully connected to speeschgen.io port 443"); 
}
else
{ Serial.println("ERROR - WifiClientSecure.connect to SpeechGen.io failed!");    // <-  so we always end here with ESP 3.x
}
```

### Debug Message

```plain
client.connect("speechgen.io", 443) fail
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-24:
Try adding `client.setPlainStart();` before you connect.

kaloprojects, 2024-07-24:
cool .. give me a second ;) .. i'll check

kaloprojects, 2024-07-24:
i tested .. in first moment i was HAPPY ! .. because _now it gets connected_ ! :)
.. **but** been too early pleased .. after sending the header (via POST request) the client.available() gets true (forever) .. but never receiving a response from client, this gets endless loop.:
```
while (client.available())    
 { char c = client.read();
   jsonResponse += String(c);
 }
````      
I tested same with "api.openai.com" (which works with 3.x still), just for verification .. when adding a 'client.setPlainStart()' before .connect it connects (same as it does without) but never sends a response back (available always true).

What is 'client.setPlainStart()' doing ? any idea what i could test next ?
Thx so much btw (much appreciated)!

kaloprojects, 2024-07-24:
ohh, hold on .. what i wrote is wrong. There is a response, let me analyze the server string in detail. I will follow up.

kaloprojects, 2024-07-24:
this happens:
- it connects to server (which is great ;)
- now reading the server response with this code snippet: 
```
// Receiving SpeechGen.IO response 
      /* header is sent .. now we wait response  ...*/
      String json_Response = "";
      while ( json_Response == "" )    
      { while (client_tcp.available())    
        { char c = client_tcp.read();
          Serial.print( c );
          json_Response += String(c);               
        }
      }      
```
and this is the printed response: from "speechgen.io" server now:
```
HTTP/1.1 400 Bad Request
Server: nginx/1.14.1
Date: Wed, 24 Jul 2024 19:58:55 GMT
Content-Type: text/html
Transfer-Encoding: chunked
Connection: close

10f
<html>
<head><title>400 The plain HTTP request was sent to HTTPS port</title></head>
<body bgcolor="white">
<center><h1>400 Bad Request</h1></center>
<center>The plain HTTP request was sent to HTTPS port</center>
<hr><center>nginx/1.14.1</center>
</body>
</html>0

⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮
```
and client.available() is true forever, that's why it stays in endless loop and printing endless ⸮

kaloprojects, 2024-07-25:
just 2 screenshots .. to compare 2.0.16 and 3.0.3 (might help maybe)
[image]
[image]

kaloprojects, 2024-07-25:
just 2 screenshots .. to compare 2.0.16 and 3.0.3 (might help maybe)

Speechgen with 2.0.16 - working well:
[image]

Speechgen with 3.0.3 - Not working:
[image]

kaloprojects, 2024-07-27:
Does someone have an idea why it fails on 3.x .. in case of a well know issue:
Is there any workaround to get it running ?  Thx !

lbernstone, 2024-07-27:
The server is asking for a cipher renegotiation, and that is failing.  It works fine if TLS 1.3 is enabled in IDF 5.1, but I have no idea why it succeeds in IDF 4.4.  This issue comes from upstream, so you can open an issue at https://github.com/espressif/esp-idf/issues and tell them you have a problem with the [https_request example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_request) when you change to the URL/server you want.

kaloprojects, 2024-07-29:
oje, i am afraid i do not have the skills to understand completely. What do you mean with 'it works fine if TLS 1.3 is enabled in IDF 5.1' ? Anything I missed on my side ? 

I can enter a ticket there, thanks, but honestly i don't know what in details i have to post there (I am using Arduino IDE). The only input i can deliver there is: Connection to "speechgen.io" fails with latest 3.0 (which is unfortunatly a showstopper meanwhile for my project as i need the 3.x for the new I2S_std.h) .. and still working with 2.x. I Will this help them to find root cause ? _I do not have any of their https examples in my Arduino sketch_ .. afraid they can't help :| (without someone from your side supporting me) :|. Fyi: I also asked the speechgen.io support, they also have no idea.

kaloprojects, 2024-08-04:
I just posted the issue here: https://github.com/espressif/esp-idf/issues/14298
Hope we get some feedback/ideas.

kaloprojects, 2024-08-07:
@lbernstone : we got feedback from the Espressif team :) .. see here:

_Thanks for the issue. This is due to the fact that the `MBEDTLS_HARDWARE_AES` is turned on and the required cipher is not supported by the hardware.
This bug has already been fixed with https://github.com/espressif/esp-idf/commit/2c85399da0eb16dc7edc3afcf8be250a7b6bcc44

This has been kept disabled by default on older branches to avoid silently increasing the flash footprint for users. Please enable the fix using `Component Config -> mbedTLS -> Enable support for non-AES ciphers in GCM operation`.  After that the handshake should succeed._

kaloprojects, 2024-08-07:
does this info help to include it in the next arduino-esp32 version ... or is there anything i have to add in my code ? Sorry in case that's a dummy question (i am not experienced in this ;)

lucasssvaz, 2024-08-07:
@me-no-dev I will

kaloprojects, 2024-08-12:
thank you folks for all your help ! :)
Just for my understanding: This means it should work with next arduino-esp32 (3.1.0) in few weeks, did i get it correctly ?