espressif/arduino-esp32 #10077: Unable build with ESP32-S3 on ESP-IDF

KCYoong, 2024-07-26:
### Board

Adafruit Feather ESP32-S3 TFT

### Device Description

ESP32-S3-MINI-1

### Hardware Configuration

N/A

### Version

v3.0.3

### IDE Name

Espressif IDE with ESP-IDF 5.1.4

### Operating System

Window 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Unable build with example Hello World code with Arduino-ESP32 library on ESP32-S3, but other target of ESP32 are able to build.

### Sketch

```cpp
#include "Arduino.h"

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("Hello world!");
  delay(1000);
}
```

### Debug Message

```plain
[1528/1530] Linking CXX executable app-template.elfFAILED: app-template.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 "-Wl,--Map=C:/Users/kcyoong96124/Desktop/P-Light/Firware Development/ESP-IDF/Hello_World-Arduino/build/app-template.map" -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\app-template.elf.rsp -o app-template.elf  && cd ."
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find Development/ESP-IDF/Hello_World-Arduino/managed_components/espressif__esp-sr/lib/esp32s3: No such file or directory
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find Development/ESP-IDF/Hello_World-Arduino/managed_components/espressif__esp-sr/esp-tts/esp_tts_chinese/esp32s3: No such file or directory
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find -ldl_lib: No such file or directory
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find -lfst: No such file or directory
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find -lc_speech_features: No such file or directory
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find -lesp_tts_chinese: No such file or directory
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find -lvoice_set_xiaole: No such file or directory
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-07-26:
Works for me. Sounds like perhaps you were unable to download the esp-sr component.
Remove the build directory. Remove managed_components/espressif__esp-sr/ . Run `idf.py set-target esp32s3`. Run `idf.py build`.

KCYoong, 2024-07-26:
> Works for me. Sounds like perhaps you were unable to download the esp-sr component. Remove the build directory. Remove managed_components/espressif__esp-sr/ . Run `idf.py set-target esp32s3`. Run `idf.py build`.

Still not work for me, it occurs same error.

lbernstone, 2024-07-26:
You'll have to post the full log.

KCYoong, 2024-07-26:
[log.zip](https://github.com/user-attachments/files/16386179/log.zip)
Log file from build > log

lucasssvaz, 2024-07-26:
Also make sure you have the submodules initialized and in the proper version.

KCYoong, 2024-07-29:
> Also make sure you have the submodules initialized and in the proper version.

Sorry for late reply, how do I know submodules initialized in proper version or not ?

SuGlider, 2024-07-30:
@KCYoong - I think that it may be an issue related to long path length and Windows.
Please try making the folder names shorter.

Example: `Development/ESP-IDF/Hello_World-Arduino/` ---> `Dev/IDF/HelloWorld/`

lucasssvaz, 2024-07-30:
> > Also make sure you have the submodules initialized and in the proper version.
> 
> Sorry for late reply, how do I know submodules initialized in proper version or not ?

`git submodule update --init --recursive`

KCYoong, 2024-07-31:
> @KCYoong - I think that it may be an issue related to long path length and Windows. Please try making the folder names shorter.
> 
> Example: `Development/ESP-IDF/Hello_World-Arduino/` ---> `Dev/IDF/HelloWorld/`

Still unable to build, i try make folder name shorter.
Folder Name: Desktop\Firmware\IDF\HelloWorld

Log file attached
[log.zip](https://github.com/user-attachments/files/16435335/log.zip)

KCYoong, 2024-08-01:
> git submodule update --init --recursive
I have run this command and build again but still not working.
[log.zip](https://github.com/user-attachments/files/16448415/log.zip)

SuGlider, 2024-08-01:
>I use the offline installer to install IDF, is it will cause submodules initialized in not proper version ?

Could you please execute `idf.py --version` and report the result back?

KCYoong, 2024-08-01:
> > I use the offline installer to install IDF, is it will cause submodules initialized in not proper version ?
> 
> Could you please execute `idf.py --version` and report the result back?

C:\Espressif\frameworks\esp-idf-v5.1.4>idf.py --version
ESP-IDF v5.1.4-dirty

me-no-dev, 2024-08-01:
You also need to run `./install.sh` from within the IDF folder to pull all tools

KCYoong, 2024-08-01:
> You also need to run `./install.sh` from within the IDF folder to pull all tools

After run install.sh, and git submodule update --init --recursive, i try re-build again but it still unable to build.
[log.zip](https://github.com/user-attachments/files/16453842/log.zip)

SuGlider, 2024-08-01:
@KCYoong - I get no issues...
Those are  the all steps (from ground up) necessary in order to make it work:

--------
- IDF 5.1.4 environment installation:
1) From Windows 11, install IDF 5.1.4 using OffLine installer found in [https://dl.espressif.com/dl/esp-idf/](https://dl.espressif.com/dl/esp-idf/)
2) Run `esp-idf-tools-setup-offline-5.1.4.exe` - Select all option, including PowerShell, all Supported SoC
3) Once IDF is installed, run `ESP-IDF 5.1 CMD`, which was installed into Windows.
3) Run `idf.py --version` to confirm installation and IDF version.
--------
- Create the HelloWorld Arduino as IDF Component basic example:
1) From the `ESP-IDF 5.1 CMD` prompt, execute:
`cd %USERPROFILE%\Desktop`
`mkdir Firmware\IDF`
`cd mkdir Firmware\IDF`
`git clone --depth=1 https://github.com/espressif/arduino-esp32.git arduino`
`cd arduino`
`git sparse-checkout set idf_component_examples/hello_world`
`cd idf_component_examples`
`move hello_world ..\..\HelloWorld`
`cd ../..`
`rmdir /q/s arduino`
`cd HelloWorld`
--------
- Prepare the project components declaration
1) execute this in the CMD: `notepad main\idf_component.yml`
edit the file by commenting out the last two lines, by adding a `#` in the begining of those lines:
2) **Save the file and exit `notepad` application.**

```
## IDF Component Manager Manifest File
dependencies:
  espressif/arduino-esp32:
    version: "*"
#    override_path: "../../../"
#    pre_release: true
```
--------
-Build the HelloWorld example (Arduino as Component - code in the folder `main`
1) From the `ESP-IDF 5.1 CMD` prompt, execute:
`idf.py set-target esp32s3`
`idf.py -p COM7 flash monitor`   ---> replace COM7 with the ESP32-S3 UART Port from your Windows environment
COMxx port shall be marker as `Silicon Labs CP210x USB to UART Bridge (COMxx)` when looking into `Device Manager` application.
check the output after flashing the firmware. It worked fine here.
-------
**Please try those steps.**

KCYoong, 2024-08-02:
> @KCYoong - I get no issues... Those are the all steps (from ground up) necessary in order to make it work:
> 
> * IDF 5.1.4 environment installation:
> 
> 1. From Windows 11, install IDF 5.1.4 using OffLine installer found in https://dl.espressif.com/dl/esp-idf/
> 2. Run `esp-idf-tools-setup-offline-5.1.4.exe` - Select all option, including PowerShell, all Supported SoC
> 3. Once IDF is installed, run `ESP-IDF 5.1 CMD`, which was installed into Windows.
> 4. Run `idf.py --version` to confirm installation and IDF version.
> 
> * Create the HelloWorld Arduino as IDF Component basic example:
> 
> 1. From the `ESP-IDF 5.1 CMD` prompt, execute:
>    `cd %USERPROFILE%\Desktop`
>    `mkdir Firmware\IDF`
>    `cd mkdir Firmware\IDF`
>    `git clone --depth=1 https://github.com/espressif/arduino-esp32.git arduino`
>    `cd arduino`
>    `git sparse-checkout set idf_component_examples/hello_world`
>    `cd idf_component_examples`
>    `move hello_world ..\..\HelloWorld`
>    `cd ../..`
>    `rmdir /q/s arduino`
>    `cd HelloWorld`
> 
> * Prepare the project components declaration
> 
> 1. execute this in the CMD: `notepad main\idf_component.yml`
>    edit the file by commenting out the last two lines, by adding a `#` in the begining of those lines:
> 2. **Save the file and exit `notepad` application.**
> 
> ```
> ## IDF Component Manager Manifest File
> dependencies:
>   espressif/arduino-esp32:
>     version: "*"
> #    override_path: "../../../"
> #    pre_release: true
> ```
> 
> -Build the HelloWorld example (Arduino as Component - code in the folder `main`
> 
> 1. From the `ESP-IDF 5.1 CMD` prompt, execute:
>    `idf.py set-target esp32s3`
>    `idf.py -p COM7 flash monitor`   ---> replace COM7 with the ESP32-S3 UART Port from your Windows environment
>    COMxx port shall be marker as `Silicon Labs CP210x USB to UART Bridge (COMxx)` when looking into `Device Manager` application.
>    check the output after flashing the firmware. It worked fine here.
> 
> **Please try those steps.**

**After I follow those steps, it is able to build and flash.**
Build Log:
[build log.txt](https://github.com/user-attachments/files/16464985/build.log.txt)

But, it is unable to build, if I put the Arduino in components folder for esp32s3.
`├── CMakeLists.txt
├── components
│   ├── Arduino
├── main
│   ├── CMakeLists.txt
│   └── main.cpp
└── README.md `

lbernstone, 2024-08-02:
Are there archive libraries (.a files) in those directories that the linker is complaining about?  There should be a bunch of libs there.

KCYoong, 2024-08-02:
> Are there archive libraries (.a files) in those directories that the linker is complaining about? There should be a bunch of libs there.

Only arduino-esp32 library for build the HelloWorld
[image]

Jason2866, 2024-08-02:
Can't reproduce either. Building Arduino as a component of IDF is imho easier with Platformio. The community version `pioarduino` does support this build variant.
See https://github.com/pioarduino/platform-espressif32
Working examples are here https://github.com/pioarduino/platform-espressif32/tree/main/examples

SuGlider, 2024-08-02:
@KCYoong - `sdkconfig` will tell what is the SoC Target for building. You can open it and check the value for 
`CONFIG_IDF_TARGET="esp32s3"`
`CONFIG_IDF_TARGET_ESP32S3=y`
`CONFIG_ARDUINO_VARIANT="esp32s3"` 
... and many other references to esp32s3 

After building for ESP32, in order to change to another SoC (such as ESP32-S3), it is necessary to delete sdkconfig and delete `build` folder before start building a new SoC.
After deleting it, please run `idf.py set-target esp32s3` as it will download and resolve all necessary dependencies for the ESP32-S3.
Only after that, you can run `idf.py build`

lbernstone, 2024-08-02:
> > Are there archive libraries (.a files) in those directories that the linker is complaining about? There should be a bunch of libs there.
> 
> Only arduino-esp32 library for build the HelloWorld 

`c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find Development/ESP-IDF/Hello_World-Arduino/managed_components/espressif__esp-sr/lib/esp32s3: No such file or directory`
Look in that directory.  I have the following:
```
total 13004
-rw-r--r-- 1 larryb larryb  598288 May  6 21:49 libc_speech_features.a
-rw-r--r-- 1 larryb larryb 3550770 May  6 21:49 libdl_lib.a
-rw-r--r-- 1 larryb larryb  296108 May  6 21:49 libesp_audio_front_end.a
-rw-r--r-- 1 larryb larryb 3482100 May  6 21:49 libesp_audio_processor.a
-rw-r--r-- 1 larryb larryb 1942754 May  6 21:49 libflite_g2p.a
-rw-r--r-- 1 larryb larryb  460286 May  6 21:49 libfst.a
-rw-r--r-- 1 larryb larryb  185556 May  6 21:49 libhufzip.a
-rw-r--r-- 1 larryb larryb 1853570 May  6 21:49 libmultinet.a
-rw-r--r-- 1 larryb larryb  267776 May  6 21:49 libnsnet.a
-rw-r--r-- 1 larryb larryb  653424 May  6 21:49 libwakenet.a
```

KCYoong, 2024-08-05:
> @KCYoong - I have changed the same setup that I mentioned in [#10077 (comment)](https://github.com/espressif/arduino-esp32/issues/10077#issuecomment-2263207958) in order to use `arduino` cloned into `components` folder.
> 
> 1. deleted `build` folder and `sdkconfig` file
> 2. deleted `main\idf_component.yml` file to force using `arduino` cloned version
> 3. created `components` folder and cloned arduino with `git clone --recursive https://github.com/espressif/arduino-esp32.git arduino` into `components` folder
> 4. executed `idf.py set-target esp32s3` from the project folder
> 5. executed `idf.py -p com4 flash monitor` : `com4` if my esp32s3 UART port
> 
> It has built, flashed and executed correctly.
After I try your steps, it is still unable to build.
[sdkconfig.txt](https://github.com/user-attachments/files/16489566/sdkconfig.txt)
[build log 5-8-2024.txt](https://github.com/user-attachments/files/16489567/build.log.5-8-2024.txt)

> The only differente I see is that your project has `Kconfig.projbuild` file in the `main` folder. What is that for?
It is occurs when build the project.
[Kconfig.txt](https://github.com/user-attachments/files/16489579/Kconfig.txt)

KCYoong, 2024-08-05:
> `c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find [...]

I found these in my directory also.
[image]

SuGlider, 2024-08-06:
@KCYoong - I was able to reproduce the same issue as reported in your log file [build log 5-8-2024.txt](https://github.com/user-attachments/files/16489567/build.log.5-8-2024.txt)

I solved it by changing the folder name of `Firmware Development` to just `dev`
`C:\Users\kcyoong96124\Desktop\P-Light\dev\IDF5\HelloWorld_Arduino` will fix it.

This seems to be the problem with long folder path that I initially described.

KCYoong, 2024-08-06:
> @KCYoong - I was able to reproduce the same issue as reported in your log file [build log 5-8-2024.txt](https://github.com/user-attachments/files/16489567/build.log.5-8-2024.txt)
> 
> I solved it by changing the folder name of `Firmware Development` to just `dev` `C:\Users\kcyoong96124\Desktop\P-Light\dev\IDF5\HelloWorld_Arduino` will fix it.
> 
> This seems to be the problem with Windows long folder path name that I initially described.

You are right, I change the folder name to "C:\Users\kcyoong96124\Desktop\P-Light\dev\idf\HelloWorld_Arduino", it is able to build and flash.

Build Log: 
[build log 6-8-2024.txt](https://github.com/user-attachments/files/16506525/build.log.6-8-2024.txt)

Jason2866, 2024-08-06:
@lapshin
Can this complicated folder path be changed?
```
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe
```
The real path to `ld.exe` is shorter ;-)

Lapshin, 2024-08-06:
@Jason2866 , Won't change. This is a common approach to how GCC looks for internals.

Long paths do not seem to be a problem here but spaces.

BTW, in the [release esp-13.2.0_20240530](https://github.com/espressif/crosstool-NG/releases/tag/esp-13.2.0_20240530 ) longpaths and 8dot3 enabled for windows.

Jason2866, 2024-08-06:
@Lapshin Thx for the info. So the issue is solved when Arduino is moving to IDF 5.3

SuGlider, 2024-08-06:
Great, I'll close this issue and change the description to include "windows".

closed 2024-08-06 by SuGlider

SuGlider, 2024-08-06:
>Long paths do not seem to be a problem here but spaces.

In order to confirm the issue here: 
The problem is using spaces in the folfer name. 
"Firmware Develpment" had a space (ASCII 0x20) in its name.
replacing it using the same name, but with no " ", solves the issue.
"Firmware_Development" works fine.

KCYoong, 2024-08-07:
Thank you very much to solve my problem.