espressif/arduino-esp32 #10078: Missing reference to main

FStefanni, 2024-07-26:
### Board

esp32:esp32:esp32

### Device Description

esp32:esp32:esp32

### Hardware Configuration

I am using a lot of peripherals, like two I2C (with attached a oled display, inclinometer, etc), some GPIO, etc.

### Version

v3.0.3

### IDE Name

arduino-cli

### Operating System

Linux Debian Testing

### Flash frequency

Not important

### PSRAM enabled

yes

### Upload speed

Not important

### Description

Hi,

I compile by running this command:
```bash
arduino-cli compile -v --output-dir /tmp --warnings all --build-property build.partitions=min_spiffs --build-property upload.maximum_size=1966080 --fqbn esp32:esp32:esp32 MySketch.ino
```

If I try to compile a simple sketch, it compiles just fine.
But if I try to compile a very complex sketch, I get the following error:

```bash
/home/kappa/.arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kappa/.arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/crt0.o:(.literal+0x0): undefined reference to `main'
collect2: error: ld returned 1 exit status
```

Not clear to me why this happens...
Version 2.0.17 of Arduino-ESP32 works fine, all 3.x.x versions have this issue.

Regards

### Sketch

```cpp
Not able to have a shareable sketch
```

### Debug Message

```plain
/home/kappa/.arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kappa/.arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/crt0.o:(.literal+0x0): undefined reference to `main'
collect2: error: ld returned 1 exit status
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-29:
You need to simplify that complex sketch to something that we can compile to reproduce. The issue submitted as such is not clear.

FStefanni, 2024-07-29:
Hi,

I have finally understood the issue: I was missing the flags `-MMD -c`, since I use the C/C++ `extra_flags`.

Imho, I am not sure if it was a good idea to use the `extra_flags` in v3 to pass flags which are not extra but required for compiling (in fact, people use `extra_flags` to add custom options...).
Nevertheless, at the moment, adding the new mandatory flags, solve the issue.

Thank you for the help.
Regards

closed 2024-07-29 by FStefanni