espressif/arduino-esp32 #10079: SPI buses on ESP32-S2 are mislabeled in arduino-esp32

egnor, 2024-07-26:
### Board

(any)

### Device Description

(any ESP32-S2 device)

### Hardware Configuration

(any configuration that uses SPI)

### Version

latest master (checkout manually)

### IDE Name

(any)

### Operating System

(any)

### Flash frequency

(any)

### PSRAM enabled

yes

### Upload speed

(any)

### Description

[At the hardware level](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf#spi), the ESP32-S2 has four SPI buses: SPI0, SPI1, SPI2, SPI3. SPI0 and SPI1 are (mostly) reserved for flash I/O, SPI2 and SPI3 are available for user use.

In arduino-esp32, [ESP32-S2 SPI buses are labeled as follows](https://github.com/espressif/arduino-esp32/blob/0fa4aa632c6dcb7736a291a33fb9f0a25614d6d0/cores/esp32/esp32-hal-spi.h#L34):
```
#elif CONFIG_IDF_TARGET_ESP32S2
#define FSPI 1  //SPI 1 bus. ESP32S2: for external memory only (can use the same data lines but different SS)
#define HSPI 2  //SPI 2 bus. ESP32S2: external memory or device  - it can be matrixed to any pins
#define SPI2 2  // Another name for ESP32S2 SPI 2
#define SPI3 3  //SPI 3 bus. ESP32S2: device only - it can be matrixed to any pins
...
```

These constants are used as indexes into [`_spi_bus_array[]` in the HAL](https://github.com/espressif/arduino-esp32/blob/0fa4aa632c6dcb7736a291a33fb9f0a25614d6d0/cores/esp32/esp32-hal-spi.c#L156):
```
#if CONFIG_IDF_TARGET_ESP32S2
  {(volatile spi_dev_t *)(DR_REG_SPI1_BASE), NULL, 0, -1, -1, -1, -1},
  {(volatile spi_dev_t *)(DR_REG_SPI2_BASE), NULL, 1, -1, -1, -1, -1},
  {(volatile spi_dev_t *)(DR_REG_SPI3_BASE), NULL, 2, -1, -1, -1, -1}
...
```

So this means that
- `FSPI` is actually SPI2, not SPI1 as commented
- `HSPI` is actually SPI3, not SPI2 as commented
- `SPI2` is actually SPI3
- `SPI3` is off the end of the array

Basically everything is off by one. [The default is to use `FSPI`](https://github.com/espressif/arduino-esp32/blob/0fa4aa632c6dcb7736a291a33fb9f0a25614d6d0/libraries/SPI/src/SPI.cpp#L332C1-L337C7):
```
#if CONFIG_IDF_TARGET_ESP32
SPIClass SPI(VSPI);
#else
SPIClass SPI(FSPI);
#endif
```

So that will actually work, but the situation is confusing and mislabeled, particularly if anyone tries to use the `SPI2` or `SPI3` constants. (And why aren't the numeric values lined up with ESP data sheets anyway?)

### Sketch

```cpp
(any code that uses SPI buses)
```

### Debug Message

```plain
(n/a)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

egnor, 2024-07-26:
(I'd be happy to submit a PR to fix the comments, if that would make sense. Changing the constants, even the `SPI2` and `SPI3` ones, might break existing code. I could also propose adding new constants and/or objects to make things clearer.)

P-R-O-C-H-Y, 2025-06-17:
Hi @egnor, The SPI bus labeling in arduino-esp32 prioritizes Arduino ecosystem consistency across all ESP32 variants (ESP32, ESP32-S2, ESP32-S3, etc.) rather than matching datasheet numbering, allowing code to be portable between different chips without modification. We intentionally avoid breaking existing applications that have relied on the FSPI/HSPI naming scheme for years, as changing this would break thousands of Arduino projects and libraries. The SPI2 and SPI3 constants you mentioned work correctly as alternative names and provide the functionality you need, even though the primary naming convention uses the FSPI/HSPI abstraction layer.

egnor, 2025-06-17:
@P-R-O-C-H-Y but but but
- the comments are factually incorrect
- `SPI3` is _off the end of the array_ and will never work

closed 2025-06-24 by me-no-dev