espressif/arduino-esp32 #10082: Change the amount of time it scans for a wifi signals

brandonfake69, 2024-07-27:
### Related area

Wifi

### Hardware specification

ESP32

### Is your feature request related to a problem?

I am following the example from https://github.com/espressif/arduino-esp32/tree/0fa4aa632c6dcb7736a291a33fb9f0a25614d6d0/libraries/WiFi/examples/WiFiScan 

It seems there is not a way to change the amount of time the ESP32 scans for a signal. For this, I am missing a LOT of access points I can find via other means. I would like to scan for 5 minutes but this seems to scan for about 5 seconds then return the results. How to fix?

### Describe the solution you'd like

Scan for a longer time period.

### Describe alternatives you've considered

Flipper zero

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

brandonfake69, 2024-07-27:
Setting this seems to make no difference in the amount of time it scans:
```
  int n = WiFi.scanNetworks(false, true, false, 30000000);
```

lbernstone, 2024-07-27:
From [esp-idf doco](https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32s3/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_scan_startPK18wifi_scan_config_tb)

The values of maximum active scan time and passive scan time per channel are limited to 1500 milliseconds. Values above 1500ms may cause station to disconnect from AP and are not recommended.

Just a note.  This is not a tutorial site.  It is a forum for errors in the code here.  Please ask general questions at https://esp32.com

P-R-O-C-H-Y, 2024-07-29:
Hi @brandonfake69, as Ibernstone wrote, there is this note in docs of maximum scan time per channel.
Can you test the scan with: `int n = WiFi.scanNetworks(false, true, false, 1500);` if you get better scan results?

PR #10083 (2024-07-31): feat(wifi): Allow setting minimum time for wifi scan
## Description of Change
This PR add a function to WiFiScan class for setting the minimum active scan time.
This allows complete timing customisation for Wifi scan.

`void setScanActiveMinTime(uint32_t ms);`

The default value is 100 to keep compatibility with previous code.

```
WiFi scan timing parameters explained:

min=0, max=0: scan dwells on each channel for 120 ms.
min>0, max=0: scan dwells on each channel for 120 ms.
min=0, max>0: scan dwells on each channel for max ms.
min>0, max>0: the minimum time the scan dwells on each channel i [... cut]
closed 2024-07-31 by me-no-dev