espressif/arduino-esp32 #10084: Issue with lwip_hook_ip6_input in Arduino IDE 2.3.2

kunal-nezai, 2024-07-29:
### Board

ESP32 Dev Module

### Device Description

NA

### Hardware Configuration

NA

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

### Description of Issue

When using Arduino IDE 1.8.19, the code compiles and functions as expected. However, when using Arduino IDE 2.3.2, I encounter the following error:

```
c:/users/admin/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/lib\liblwip.a(ip6.c.obj):(.literal.ip6_input+0x0): undefined reference to `lwip_hook_ip6_input'
c:/users/admin/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/lib\liblwip.a(ip6.c.obj): in function `ip6_input':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/ip6.c:547: undefined reference to `lwip_hook_ip6_input'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

### Sketch

```cpp
NA
```

### Debug Message

```plain
c:/users/admin/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/lib\liblwip.a(ip6.c.obj):(.literal.ip6_input+0x0): undefined reference to `lwip_hook_ip6_input'
c:/users/admin/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/lib\liblwip.a(ip6.c.obj): in function `ip6_input':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/ip6.c:547: undefined reference to `lwip_hook_ip6_input'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

### Other Steps to Reproduce

NA

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-07-29:
please provide minimal sketch to reproduce. This is not expected

kunal-nezai, 2024-07-29:
I am unsure which part of the code is causing this issue. Due to the company's NDA policy, I cannot share any code.

For reference, I can mention that the library was written for ESP-IDF v5.2.2. I used the same header files when making the library compatible with the Arduino IDE.

Jason2866, 2024-07-29:
@kunal-nezai How should anyone help, without a piece of an example! code?
No one says you should provide code which is under NDA.

me-no-dev, 2024-07-29:
@kunal-nezai you are using some IDF code in the place of Arduino code. Maybe custom WiFi handling or something of the sort to get this message. You can copy the function from [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/Network/src/NetworkInterface.cpp#L37-L47) to your codebase and get to compile.

kunal-nezai, 2024-07-30:
@me-no-dev  Hello!

Today, while working, I noticed that the older version of Arduino IDE (1.8.19) uses version 3.0.2 of the ESP32 board, whereas the new Arduino IDE (2.3.2) uses version 3.0.3 of the ESP32 board.

To test my suspicion, I installed version 3.0.2 of the ESP32 board in the new Arduino IDE, and it is now functioning properly.

I suspect there is something in version 3.0.3 that is causing the issue.

me-no-dev, 2024-07-30:
@kunal-nezai it's the fact that you do not use Arduino fully, but instead manage WiFi through the IDF API only. If your network stack was using Arduino's APIs, this would not have happened. Adding the code above will make it work for the future, Else you will be stuck on 3.0.2

me-no-dev, 2024-07-30:
Specifically the reason is in 
https://github.com/espressif/arduino-esp32/pull/10038/files
https://github.com/espressif/esp32-arduino-lib-builder/pull/194/files

i-am-shodan, 2024-07-31:
I have the same issue. Here is a test case

```
#include "esp_netif.h"

#define STATIC_IP_ADDR "10.0.0.2"
#define DEFAULT_GATEWAY "10.0.0.1"
#define DEFAULT_NETMASK "255.255.255.0"

void setup() {
  ip4_addr_t ip;
  ip4_addr_t gw;
  ip4_addr_t netmask;
  ip4addr_aton(STATIC_IP_ADDR, &ip);
  ip4addr_aton((const char *)DEFAULT_GATEWAY, &gw);
  ip4addr_aton((const char *)DEFAULT_NETMASK, &netmask);

  esp_netif_ip_info_t ip_info;
  memset(&ip_info, 0, sizeof(esp_netif_ip_info_t));
  ip_info.ip.addr = ip.addr;
  ip_info.gw.addr = gw.addr;
  ip_info.netmask.addr = netmask.addr;

  uint8_t lwip_addr[6] = { 0x02, 0x02, 0x11, 0x22, 0x33, 0x02 };

  esp_netif_inherent_config_t base_cfg = {
    .flags = (esp_netif_flags_t)(ESP_NETIF_DHCP_SERVER | ESP_NETIF_FLAG_EVENT_IP_MODIFIED | ESP_NETIF_FLAG_AUTOUP),
    .ip_info = &ip_info,
    .get_ip_event = IP_EVENT_ETH_GOT_IP,
    .lost_ip_event = IP_EVENT_ETH_LOST_IP,
    .if_key = "usb_eth",
    .if_desc = "usb ncm config device",
    .route_prio = 10
  };

  esp_netif_driver_ifconfig_t driver_cfg = {
    .handle = (void *)1,
    .transmit = NULL, // just for this example
    .driver_free_rx_buffer = NULL // just for this example
  };

  esp_netif_config_t cfg = {
    .base = &base_cfg,
    .driver = &driver_cfg,
    .stack = ESP_NETIF_NETSTACK_DEFAULT_ETH,  // USB-NCM is an Ethernet netif from lwip perspective, we already have IO definitions for that:
  };
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Using `.stack = ESP_NETIF_NETSTACK_DEFAULT_ETH` or any of the other stacks results in a linker error

`/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/ip6.c:547: undefined reference to `lwip_hook_ip6_input'`

The same code compiles fine on a non-Arduino 3 SDK.

Interestingly this code compiles fine with platformio

Jason2866, 2024-07-31:
> Interestingly this code compiles fine with platformio

I doubt that. Probably **not** using the same Arduino core code base.

i-am-shodan, 2024-07-31:
```
platform = https://github.com/platformio/platform-espressif32.git
platform_packages = 
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.3
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
```

Jason2866, 2024-07-31:
As said not using same code base. This `https://github.com/espressif/arduino-esp32.git#3.0.3` uses Arduino commit of release Core 3.0.3
The change in Arduino was done *after*

i-am-shodan, 2024-07-31:
```
#include "lwip/netif.h"

extern "C" int lwip_hook_ip6_input(struct pbuf *p, struct netif *inp) __attribute__((weak));
extern "C" int lwip_hook_ip6_input(struct pbuf *p, struct netif *inp) {
  if (ip6_addr_isany_val(inp->ip6_addr[0].u_addr.ip6)) {
    // We don't have an LL address -> eat this packet here, so it won't get accepted on input netif
    pbuf_free(p);
    return 1;
  }
  return 0;
}
```

For my future self this might be a fix

me-no-dev, 2024-07-31:
@i-am-shodan do it together with the `#if` like I linked above. Different fix will be coming and we might switch this setting. No need to compile code that will not be used

PureTek-Innovations, 2024-08-22:
Thanks @i-am-shodan, I added your #if code to my project, I use this core version:

```
platform = https://github.com/platformio/platform-espressif32.git#develop
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.4
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.4/esp32-arduino-libs-3.0.4.zip
```
And it compiles, not tried to use it yet but I expect that it will be fine.

jonhp, 2024-09-26:
@i-am-shodan thanks for this work around. is there an integrated fix on the horizon?

atanisoft, 2025-01-26:
@me-no-dev any updates on this? I've reproduced this on v3.1.1 which can be seen in this run: https://github.com/bakerstu/openmrn/actions/runs/12976415979/job/36188645918

me-no-dev, 2025-03-07:
@atanisoft fix coming in 3.2.0-RC2: https://github.com/espressif/esp32-arduino-lib-builder/pull/286

closed 2025-03-07 by me-no-dev