espressif/arduino-esp32 #10085: esp32c3 always rebooting

frrhhhann, 2024-07-29:
### Board

esp32c3

### Device Description

i recently made digital dash for display car ecu data using xiao seed esp32c3, but the esp32 keep rebooting, the code i'm using is from this github https://github.com/73Volvo/Speedypal/tree/main but the original code is for xiao seed esp32s3 with 0.96 oled ssd1306, i custom the code for using xiao esp32c3 only delete led indicator because xiao esp32c3 don't have built in led, and change ssd1306 library to sh1106g. it works fine it can upload the code and display startup logo but it keep rebooting

### Hardware Configuration

xiaoo esp32c3, D4 D5 for i2c connect to sh1106g 1.3 inch oled, D7 D6 for RX  TX comunication to ecu

### Version

latest master (checkout manually)

### IDE Name

arduino ide

### Operating System

windows 10

### Flash frequency

160Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

i finished project but my xiao esp32c3 keep rebooting in serial monitor

### Sketch

```cpp
#include "Arduino.h"
#include "SpeedyPal.h"
#include "Wire.h"

// *****************************************************
// HERE ARE SOME VARIABLES YOU MIGHT WANT TO ADJUST
// *****************************************************

// Set to "true" record a binary MLG log. Set to "false" to record an ASCII MSL log
bool binaryLog = true;

// is there a second screen connected?
bool multiScreen = false;

// frames per second for sampling and display
uint8_t fps = 20;

// vehicle info
uint16_t flowRate = 329; // cc/min
uint8_t cyl = 6; // 4 cylinders
uint16_t weightKg = 1332; // weight in kg
float frontalArea = 1.749; // frontal area in square meters
float dragCoefficient =  0.41; // drag coefficient

// constants for arduino pins
const int8_t ButtonPinL = D1;  // the number of the left button pin
const int8_t ButtonPinR = D2;  // the number of the right button pin
[... 409 lines cut ...]
20:47:40.604 -> 3fc99ec0: 0x3fc97e80 0x706f6f6c 0x6b736154 0x4ea13700 0x003777b4 0x00000000 0x3fc99e70 0x00000001
20:47:40.604 -> 3fc99ee0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc93694 0x3fc936fc 0x3fc93764 0x00000000
20:47:40.604 -> 3fc99f00: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4203eb6e 0x00000000 0x00000000
20:47:40.604 -> 3fc99f20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:47:40.604 -> 3fc99f40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:47:40.604 -> 3fc99f60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:47:40.604 -> 3fc99f80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:47:40.604 -> 3fc99fa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:47:40.604 -> 3fc99fc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:47:40.604 -> 3fc99fe0: 0x4a000000 0xbaad5678 0x00000060 0xabba1234 0x00000054 0x00000000 0x3fc99ff4 0x00000000
20:47:40.604 -> 3fc9a000: 0x00000000 0x00000000 0x3fc9a00c 0xffffffff 0x3fc9a00c 0x3fc9a00c 0x00000000 0x3fc9a020
20:47:40.604 -> 3fc9a020: 0xffffffff 0x3fc9a020 0x3fc9a020 0x00000001 0x00000001 0x00000000 0x2000ffff 0x00000000
20:47:40.604 -> 3fc9a040: 0xb33fffff 0x00000000 0xbaad5678 0x0000008c 0xabba1234 0x00000080 0xff998ce1 0x6acd1a67
20:47:40.604 -> 3fc9a060: 0x5b84c8a1 0x235b4490 0x12260402 0x2550473b 0x7ebbb90a 0xfb91467c 0x29acf3bf 0xc0b7c92f
20:47:40.604 -> 3fc9a080: 0x6b2868c1 0x22a5f5c4 0x0015f514 0x929df6a8 0x1d03f66f 0x49ddc30f 0xba3f34c2 0x6e43b986
20:47:40.604 -> 3fc9a0a0: 0x50220bc2 0x6891f736 0x2c421a88 0x3093b434 0x6b0f6cfe 0xccfa77ca 0x9e8ba2b8 0xb6d75e1b
20:47:40.604 -> 3fc9a0c0: 0x020d284e 0xfaa74984 0x51e36c7c 0xd2422ed5 0x81e968b0 0x8bea17a9 0xbaad5678 0x0000008c
20:47:40.604 -> 3fc9a0e0: 0xabba1234 0x00000080 0x00000040 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:47:40.604 -> 3fc9a100: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:47:40.604 -> 3fc9a120: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x07000000 0x04040404
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

frrhhhann, 2024-07-29:
[custom code speedypal.zip](https://github.com/user-attachments/files/16414404/custom.code.speedypal.zip)
here is full code that i custom

SuGlider, 2024-07-29:
> @frrhhhann - what are those pins:
> 
> ```c++
> // constants for arduino pins
> const int8_t ButtonPinL = D1;  // the number of the left button pin
> const int8_t ButtonPinR = D2;  // the number of the right button pin
> const int8_t chipSelect = D3; // D3 for Seeed ESP32S3, 21 for Seeed Sense  // the number of Chip Select pin for the SD card
> const int8_t TXpin = D6;  // the number of the TX pin
> const int8_t RXpin = D7;  // the number of the RX pin
> ```
> 
> Those are not listed in https://github.com/espressif/arduino-esp32/blob/master/variants/esp32c3/pins_arduino.h

Found within `xiao seed esp32c3` board. 
https://github.com/espressif/arduino-esp32/blob/master/variants/XIAO_ESP32C3/pins_arduino.h

frrhhhann, 2024-07-29:
> Found within `xiao seed esp32c3` board. https://github.com/espressif/arduino-esp32/blob/master/variants/XIAO_ESP32C3/pins_arduino.h

i change the pin but still have rebooting problem

SuGlider, 2024-07-29:
My suggestion is to run this sketch using `Core Debug Level: "Verbose"`.
Add this line to the code:

``` cpp
void setup()
{
  pinMode(ButtonPinR, INPUT_PULLUP);
  pinMode(ButtonPinL, INPUT_PULLUP);

  Wire.begin();
  Serial.begin(115200); // usb connection on Arduino Nano
  Serial.setDebugOutput(true);  // <=== this line will force all Debug Output to go to USB CDC port
```

frrhhhann, 2024-07-29:
> My suggestion is to run this sketch using `Core Debug Level: "Verbose"`. Add this line to the code:
> 
> ```c++
> void setup()
> {
>   pinMode(ButtonPinR, INPUT_PULLUP);
>   pinMode(ButtonPinL, INPUT_PULLUP);
> 
>   Wire.begin();
>   Serial.begin(115200); // usb connection on Arduino Nano
>   Serial.setDebugOutput(true);  // <=== this line will force all Debug Output to go to USB CDC port
> ```

do i need to asign pin to button R and L? and TX Rx pin

frrhhhann, 2024-07-29:
> My suggestion is to run this sketch using `Core Debug Level: "Verbose"`. Add this line to the code:
>
> ```c++
> void setup()
> {
>   pinMode(ButtonPinR, INPUT_PULLUP);
>   pinMode(ButtonPinL, [...]
i add this to the code but still no luck
```c++
Serial.setDebugOutput(true);  // <=== this line will force all Debug Output to go to USB CDC port
```

me-no-dev, 2024-07-29:
Did you set the core debug level to Verbose in the board menu?

SuGlider, 2024-07-29:
>20:47:40.604 -> Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

@frrhhhann - this is a memory fault error. Bad pointer.

I have found it... in `splashStart()`. 
`frame` value goes beyond the `Speeduino_Logo_Anim_Array[frame]` array boundary.

Fix it by `if (frame >= \* ">" only is the bug *\ Speeduino_Logo_Anim_LEN) frame = 0;`

SuGlider, 2024-07-29:
No Arduino Core issue. This is user code failure.

frrhhhann, 2024-07-29:
> @frrhhhann - this is a memory fault error. Bad pointer.
>
> I have found it... in `splashStart()`. `frame` value goes beyond the `Speeduino_Logo_Anim_Array[frame]` array boundary.
>
> Fix it by `if (frame [...]

where do i need to replace the code?
```c++
// '01', 64x64px
const unsigned char Speeduino_Logo_Anim_01 [] PROGMEM = {
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
```

SuGlider, 2024-07-29:
> where do i need to replace the code?

In GraphicFunctions.ino Line #109
``` cpp
void splashStart(){
  int16_t centerX = (SCREEN_WIDTH - 64) / 2;
  int16_t centerY = (SCREEN_HEIGHT - 64) / 2;
  int16_t width = 64;
  int16_t height = 64;
  int16_t x1, y1;
  uint16_t w, h;
  int logoX; // offset for logo display based on screen orientation
  String noECU = "LOADING";
  int frame = 0;

  if (SCREEN_ROTATION == 0 || SCREEN_ROTATION == 2) logoX = 32;
  if (SCREEN_ROTATION == 1 || SCREEN_ROTATION == 3) logoX = 0;

  Serial.println("displaying logo");

  // Display initial boot animation
  for ( int i = 0; i < Speeduino_Logo_Anim_LEN ; i ++) {
  selectedOLED->clearDisplay();
  selectedOLED->drawBitmap(logoX, 0, Speeduino_Logo_Anim_Array[i], 64, 64, SH110X_WHITE);
  endPage();
  delay(50);
  }

  // Cycle boot animation until ECU is detected
  while (!ecuConnected) {
  selectedOLED->clearDisplay();
  selectedOLED->drawBitmap(logoX, 0, Speeduino_Logo_Anim_Array[frame], 64, 64, SH110X_WHITE);
  frame ++;
  if (frame >= Speeduino_Logo_Anim_LEN) frame = 0;  // <<<<<<<<<============================ HERE

  selectedOLED->getTextBounds(noECU,0,0,&x1, &y1, &w, &h);
  if (SCREEN_ROTATION == 0 || SCREEN_ROTATION == 2) selectedOLED->setCursor(SCREEN_WIDTH-w-4, SCREEN_HEIGHT-h-4);
  if (SCREEN_ROTATION == 1 || SCREEN_ROTATION == 3) selectedOLED->setCursor((SCREEN_WIDTH-w)/2, SCREEN_HEIGHT-h-4);
  selectedOLED->setTextSize(1);
  selectedOLED->setTextColor(SH110X_WHITE);
  selectedOLED->print("LOADING");

  endPage();
  delay(50);

  // is the Speeduino connected? Also, read the ECU's information into a buffer called "buffer"
  if (ecuWait < 5) ecuWait = 5; // don't wait less than 5ms for Speeduino to respond
  ecuConnected = requestData(ecuWait);

  }

}

```

closed 2024-07-29 by SuGlider

frrhhhann, 2024-07-30:
when esp32 is not connected via rx tx pin d6 and d7 is works perfectly without rebooting, but when i connect to ecu via rx tx it keep rebooting

SuGlider, 2024-07-30:
We don't have the hardware that you are using. So far it has been an application issue, not an Arduino Core code problem.

Please debug it and find out if there is any Arduino Core failure.
When you find any Arduino Core issue, create a minimum sketch that demonstrates it and post a new issue.
The issue must not contain or use any external library.

Good luck!