espressif/arduino-esp32 #10086: esp 32 wroom MODULE current reading

ViruKM, 2024-07-30:
### Related area

AFTER DOWNLOADING ny codes like I2C, WIFI ANY CODE 

### Hardware specification

Support ESP32 WROOM module wifi

### Is your feature request related to a problem?

after downloading the any codes unable to reading current in uA

### Describe the solution you'd like

any codes examples from ESP 32 Arduino ide examples also same issue

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lucasssvaz, 2024-07-30:
Could you please explain your problem better and provide a minimal sketch to reproduce the issue ? How are you trying to measure the current ? Starting from 3.0.0 the HAL sensor is not supported anymore due to changes in IDF

ViruKM, 2024-07-30:
#include "driver/periph_ctrl.h"
#include "driver/spi_master.h"
#include "driver/adc.h"
#include "esp_sleep.h"
#include "driver/gpio.h"

#define SPI_SCK   18
#define SPI_MISO  19
#define SPI_MOSI  23
#define SPI_CS    5

#define ADC_PIN   36

void setup() {
  Serial.begin(9600);
  delay(100);
  Serial.println("Starting setup...");

  periph_module_enable(PERIPH_SPI_MODULE);

  gpio_reset_pin((gpio_num_t)SPI_SCK);
  gpio_set_direction((gpio_num_t)SPI_SCK, GPIO_MODE_OUTPUT);
  gpio_set_level((gpio_num_t)SPI_SCK, 0);

  adc1_config_width(ADC_WIDTH_BIT_12);
  adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_DB_11);

  esp_sleep_enable_timer_wakeup(10 * 10000);

  Serial.println("SPI clock enabled. No other peripherals are initialized.")
;
}

void loop() {
  float voltage = readVoltage();
  Serial.print("Voltage: ");
  Serial.print(voltage);
  Serial.println(" V");

  if (voltage < 3.0) {
    enterDeepSleep(); //diff power modes
  } else {
    enterLightSleep();
  }

  delay(100);
}

float readVoltage() {
  int adcValue = adc1_get_raw(ADC1_CHANNEL_0);
  float voltage = adcValue * (3.6 / 4095.0);

  voltage *= 2;

  return voltage;
}

void enterLightSleep() {
  Serial.println("Entering light sleep mode...");
  esp_light_sleep_start();
  Serial.println("Waking up from light sleep mode...");
}

void enterDeepSleep() {
  Serial.println("Entering deep sleep mode...");
  esp_deep_sleep_start();
}

void enterActiveMode() {
  Serial.println("Entering active mode...");
}

above code downloaded in ESP 32 wifi WROOM module using Arduino IDE 2.3.2
after uploading, measuring current using multimeter in uA.
it shows Overload,
can you provide proper guide

Thanks
Viru KM

lucasssvaz, 2024-07-31:
The multimeter measurement is going out of scale. Please put the multimeter in mA mode.