espressif/arduino-esp32 #10114: I2C protocol not working

ankushiot, 2024-08-05:
### Board

ESP32 C6 WROOM 1

### Device Description

Unable to detect I2C sensor connected. 

### Hardware Configuration

No only BME 680 was connected. SDA Pin 6, SCL Pin 7

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Unable to detect I2C sensor even I2C scanned doesn't shows anything.

### Sketch

```cpp
Ada Fruit bme680test.ino
```

### Debug Message

```plain
Sensor working fine on Arduino Nano
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Thieuzor, 2025-01-14:
Hi! I've got the same issue with the following sketch:
`#include <Wire.h>
 
void setup() {
  Wire.begin(21,22,100000);
  Serial.begin(115200);
  Serial.println("\nI2C Scanner");
}
 
void loop() {
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
      nDevices++;
    }
    else if (error==4) {
      Serial.print("Unknow error at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  }
  else {
    Serial.println("done\n");
  }
  delay(5000);          
}`

Only on 3.1.x it's not working anymore. I've tested it successfully on previous versions.

me-no-dev, 2025-01-14:
Can you try instead of `Wire.begin(21,22,100000);` to have 
```cpp
Wire.setPins(21, 22);
Wire.begin();
```

me-no-dev, 2025-01-14:
The following code works for me on 3.1.1
```arduino
#include "Wire.h"

void setup() {
  Serial.begin(115200);
  Wire.setPins(6, 7);
  Wire.begin();
}

void loop() {
  byte error, address;
  int nDevices = 0;

  delay(5000);

  Serial.println("Scanning for I2C devices ...");
  for (address = 0x01; address < 0x7f; address++) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.printf("I2C device found at address 0x%02X\n", address);
      nDevices++;
    } else if (error != 2) {
      Serial.printf("Error %d at address 0x%02X\n", error, address);
    }
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found");
  }
}
```

Parsaabasi, 2025-03-11:
Hi @ankushiot 
Do you still need assistance with this?

closed 2025-03-12 by Jason2866