espressif/arduino-esp32 #10115: Guru Meditation Error: Core 1 panic'ed with Espressif board package > 2.0.14

Patsjemoe, 2024-08-05:
### Board

ESP32S3 DEVKitC-1

### Device Description

none 

### Hardware Configuration

ILI9486  3.5"  TFT 

### Version

v3.0.4

### IDE Name

Arduino 2.3.2

### Operating System

windows10

### Flash frequency

QIO80mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Guru Meditation Error: Core 1 panic'ed, problem happening on Esp32S3, not on esp32wroom32
problem is  pointing to SPI write command(in TFT_eSPI::writecommand(unsigned char))  with board package >2.0.14......downgrading to 2.0.14 solved the problem, so my question is to correct the later /latest version 3.0.4

### Sketch

```cpp
irrelevant, problem is board package related, when selecting esp32s3
```

### Debug Message

```plain
ESP Exception Decoder
Sketch: ui FQBN: esp32:esp32:esp32s3

PC : 0x420267fa PS : 0x00060430 A0 : 0x820269d8 A1 : 0x3fcebd20
A2 : 0x3fc969a8 A3 : 0x00000000 A4 : 0x60004000 A5 : 0x00000010
A6 : 0x000000ff A7 : 0xff000000 A8 : 0x08000000 A9 : 0x3fcebcf0
A10 : 0x3fcaa72c A11 : 0x00000001 A12 : 0x02625a00 A13 : 0xffffffff
A14 : 0x00000031 A15 : 0x00000000 SAR : 0x00000006 EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000010 LBEG : 0x4202ae24 LEND : 0x4202ae6b LCOUNT : 0x00000003

Backtrace: 0x420267f7:0x3fcebd20 0x420269d5:0x3fcebd50 0x42026bf4:0x3fcebd80 0x420020bb:0x3fcebda0 0x4202be06:0x3fcebe10 0x4037ed22:0x3fcebe30

PC: 0x420267fa: is in TFT_eSPI::writecommand(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
EXCVADDR: 0x00000010

Decoding stack results
0x420267f7: is in TFT_eSPI::writecommand(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
0x420269d5: is in TFT_eSPI::init(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:692).
0x42026bf4: is in TFT_eSPI::begin(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:603).
0x420020bb: setup() at I:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\ui\ui.ino:82
0x4202be06: loopTask(void*) at C:\Users\ludok\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\cores\esp32\main.cpp:59
0x4037ed22: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa\port.c:162
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-08-05:
@lucasssvaz can you please check and help triage this issue? Thanks

me-no-dev, 2024-08-05:
Issue is in the `TFT_eSPI` library that is not yet compatible with Arduino v3. Please @Patsjemoe go file an issue there (or chime in one already opened by someone else)

Patsjemoe, 2024-08-06:
Hi,
In the datasheet, I found a change in SPI_MOSI_DLEN_REG to SPI_MS_DLEN_REG,
but this change was already implemented in the TFT_eSPI.
Arduino V3  must have made more changes, *specifically related to the SPI*
command section. Can you confirm what changes have been made, so a search
for the problem will be easier .
thanks in advance
best regards
Ludo

Patsjemoe, 2024-08-06:
Hi,
I also tried V2.017 which shows the same error, and read in some links that V2.015 and 2.016 also have the problem.
Can you confirm where I can find details on the changes made from V2.014 to V2.015 specifically related to SPI.
thanks 
regards
Ludo

JimDrewGH, 2024-08-07:
This has been a long standing issue - since v2.0.15 was released.  Perhaps this hack will help to understand the issue:

https://github.com/Bodmer/TFT_eSPI/issues/3384#issuecomment-2211470238

Patsjemoe, 2024-08-07:
> Bodmer/TFT_eSPI#3384 (comment)

Hi Jim,
I think I found a solution to the problem, and posted this in TFT_eSPI.
But since it is started with 2.0.15, something in that board library must
have changed.
In ESP32 technical reference manual, you can find  FSPI, VSPI, HSPI, but
these terms do not show up in the ESP32-S3
version, there it is SPI1, SPI2, SPI3, so I suspect that in the setup,
changes related to these terms must have occured.
regards
Ludo

Jason2866, 2024-08-07:
The issue is in IDF code. The Macros for `REG_SPI_BASE` are wrong there.

JimDrewGH, 2024-08-11:
It might be handy if someone can just post exactly what changes are needed to the IDF files so we can manually fix this issue, and also let the Espressif team know.

Jason2866, 2024-08-11:
@JimDrewGH Did you follow the link mentioned the post before yours?
Everything mentioned there what's wrong and needs to be changed in **IDF**

lucasssvaz, 2024-08-29:
Related to https://github.com/espressif/arduino-esp32/issues/10254

VojtechBartoska, 2024-10-21:
@lucasssvaz do you know if this was fixed already in ESP-IDF?

Jason2866, 2024-10-21:
Checked the S3 entry https://github.com/espressif/esp-idf/blob/632e0c2a9fc7c754db4135dabb67f7fc6aa9fb87/components/soc/esp32s3/include/soc/soc.h#L37
still not correct. A working solution for the S3 would be for example `#define REG_SPI_BASE(i)         (((i)==2) ? (DR_REG_SPI2_BASE) : (DR_REG_SPI0_BASE - ((i) * 0x1000)))  // GPSPI2 and GPSPI3`

Parsaabasi, 2026-10-14:
Hello, 

Closing this issue because it refers to an older version. Please update to the latest version and check if the issue persists. If you still need help, feel free to reopen this issue or create a new one. Thanks!

closed 2026-10-14 by Parsaabasi