espressif/arduino-esp32 #10116: "error: 'WiFi' was not declared in this scope"

maxcraig112, 2024-08-05:
### Board

ESP32-H2

### Device Description

Device on plain breadboard connected to laptop via USB C, running arduino-ide

### Hardware Configuration

Board selected: ESP32H2 Dev Module
No pins connected
COM11

### Version

v3.0.4

### IDE Name

ARDUINO IDE

### Operating System

Windows 11

### Flash frequency

90MHZ

### PSRAM enabled

yes

### Upload speed

921600

### Description

Trying to set up Wifi of any form with the ESP32-H2, and am coming across the following error, suggesting that the wifi module is not properly installed for this version and ESP

### Sketch

```cpp
#include <iostream>
#include <WiFi.h>

const char* ssid = "home";
const char* password = "***";

void setup(){

    WiFi.begin(ssid, password);
    Serial.println("\nConnecting");

    while(WiFi.status() != WL_CONNECTED){
        std::cout << "." << std::endl;
        delay(100);
    }

    std::cout << "\nConnected to the WiFi network" << std::endl;
    std::cout << "Local ESP32 IP: " << std::endl;
    std::cout << WiFi.localIP() << std::endl;
}

void loop(){}
```

### Debug Message

```plain
C:\.arduinoIDE-unsaved202475-41180-19c449o.0egbj\sketch_aug5a\sketch_aug5a.ino: In function 'void setup()':
C:\.arduinoIDE-unsaved202475-41180-19c449o.0egbj\sketch_aug5a\sketch_aug5a.ino:9:5: error: 'WiFi' was not declared in this scope
    9 |     WiFi.mode(WIFI_STA); //Optional
      |     ^~~~
C:\.arduinoIDE-unsaved202475-41180-19c449o.0egbj\sketch_aug5a\sketch_aug5a.ino:9:15: error: 'WIFI_STA' was not declared in this scope
    9 |     WiFi.mode(WIFI_STA); //Optional
      |               ^~~~~~~~
C:\.arduinoIDE-unsaved202475-41180-19c449o.0egbj\sketch_aug5a\sketch_aug5a.ino:13:28: error: 'WL_CONNECTED' was not declared in this scope
   13 |     while(WiFi.status() != WL_CONNECTED){
      |                            ^~~~~~~~~~~~

exit status 1

Compilation error: 'WiFi' was not declared in this scope
```

### Other Steps to Reproduce

Attempting to verify or upload the code returns the error.

Reverting to previous versions, such as 2.0.14 stops this issue from happening, but ESP32-H2 is not supported in this version. Even the earliest version that has ESP32-H2 3.0.0 has this error

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-08-05:
Hello @maxcraig112, the ESP32-H2 does not have Wi-Fi. The chip has IEEE 802.15.4 for Thread, Zigbee, Matter + Bluetooth 5 (LE).

OrhanYigitDurmaz, 2024-08-05:
you guys should add checks for these too just like the usb ones

P-R-O-C-H-Y, 2024-11-14:
I am closing this issue as solved. The esp32-h2 does not have WiFi and the code is guarded.

OrhanYigitDurmaz, 2024-11-14:
> I am closing this issue as solved. The esp32-h2 does not have WiFi and the code is guarded.

I meant showing an error saying that wifi is not avaliable on esp32-h2