espressif/arduino-esp32 #10121: ESP_LOGE and log_e broken with USB-CDC on ESP32-S2 (and others?) if #include <Arduino.h> [[includes diagnosis!]]

egnor, 2024-08-06:
### Board

any board with ESP32-S2 and USB-CDC, possibly other cores

### Device Description

any board which uses on-chip USB as its primary serial port (i.e. running with `-DARDUINO_USB_CDC_ON_BOOT=1`)

### Hardware Configuration

n/a

### Version

latest master (checkout manually)

### IDE Name

any

### Operating System

any

### Flash frequency

any

### PSRAM enabled

yes

### Upload speed

any

### Description

On the ESP32-S2 (and probably some other cores), if using USB-CDC as the primary serial console, `Serial.printf` text will show up but `log_*` and `ESP_LOG*` reports will not. This severely hampers debugging as much of the system uses those macros to report important diagnostics.

I believe this is why:
1. [`esp32_hal_log.h`](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-log.h) is included indirectly by `Arduino.h`
2. In that file, (by default) [`ESP_LOG*` is replaced with `log_*`](https://github.com/espressif/arduino-esp32/blob/70786dc5fa51601f525496d0f92a220c917b4ad9/cores/esp32/esp32-hal-log.h#L331), and [`log_*` calls `log_printf(...)`](https://github.com/espressif/arduino-esp32/blob/70786dc5fa51601f525496d0f92a220c917b4ad9/cores/esp32/esp32-hal-log.h#L246)
3. Surprisingly(?), [`log_printf(...)` is defined in `esp32_hal_uart.c`](https://github.com/espressif/arduino-esp32/blob/70786dc5fa51601f525496d0f92a220c917b4ad9/cores/esp32/esp32-hal-uart.c#L925); it calls `log_printfv` which includes this code:
```
#if CONFIG_IDF_TARGET_ESP32C3 || ((CONFIG_IDF_TARGET_ESP32H2 || CONFIG_IDF_TARGET_ESP32C6) && ARDUINO_USB_CDC_ON_BOOT)
  vsnprintf(temp, len + 1, format, arg);
  ets_printf("%s", temp);
#else
  int wlen = vsnprintf(temp, len + 1, format, arg);
  for (int i = 0; i < wlen; i++) {
    ets_write_char_uart(temp[i]);
  }
#endif
```

4. As you can see, on any chip which isn't the -C3, -H2, or -C6, it will always pass the log message to `ets_write_char_uart`, which (per its name) writes directly to the UART, even if the console is otherwise configured to use USB-CDC.

### Sketch

```cpp
#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE  // Optional, doesn't help 

#include <Arduino.h>
#include <esp_log.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);  // Optional, doesn't help
  esp_log_level_set("*", ESP_LOG_VERBOSE);  // Optional, doesn't help
}

void loop() {
  Serial.println("This is a Serial.println");
  log_e("This is a log_e error");
  ESP_LOGE("mainloop", "This is an ESP_LOGE error");
  delay(250);
}
```

### Debug Message

```plain
You get lots of

This is a Serial.println
This is a Serial.println
This is a Serial.println
This is a Serial.println
...
```

...but never see `This is a log_e error` or `This is an ESP_LOGE error`, even though these should absolutely show up.
```

### Other Steps to Reproduce

This happens for me on an ESP32-S2 board. I am using this FQBN: `esp32:esp32:esp32s2:CDCOnBoot=cdc,UploadMode=cdc,PSRAM=enabled,DebugLevel=debug`

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-08-06:
This issue affecct ESP32-S2 and ESP32-S3 when using TinyUSB OTG port.

SuGlider, 2024-08-06:
@egnor - `Serial.setDebugOutput(true);` is necessary, otherwise there is no LOG output.
I'll post a PR to fix this issue.

PR #10123 (2024-08-13): fix(USB): OTG S2 and S3 debug print
## Description of Change
ESP32-S2 and S3 using USB CDC OTG (TinyUSB) won't print any log information. This commit fixes that.

## Tests scenarios
Using ESP32-S2 and S3 with `USB CDC On Boot` enabled and in the case of S3, USB Mode shall be `USB OTG (TinyUSB)`.
Set log level to Verbose.

```cpp
#include <Arduino.h>
#include <esp_log.h>
#include <chip-debug-report.h> 

void setup() {
  neopixelWrite(RGB_BUILTIN, 32, 0, 0); // RED until Serial Monitor is open
  Serial.begin();
  Serial.setDebugOutput(true);

  while (!Serial) delay(100);
  printBeforeSetupInfo( [... cut]
closed 2024-08-13 by me-no-dev